dresden-elektronik/deconz-rest-plugin #5006: `--dev` feature regression past 2.05.77

Red-M, 2021-06-13:
## Describe the bug
deCONZ no longer connects to the raspbee.
I'd like to update my deCONZ package due to the age of the package for new features and security reasons.

## Steps to reproduce the behavior
Any deCONZ version above `2.05.77` fails to connect to the device over an SSH tunnelled virtual serial device

## Expected behavior
No regression from `2.05.78` onward.

## Screenshots
[image]
[image]

## Environment
deCONZ/Phoscon versions as per above

 - Host system: x86 VM
 - Running method: Rpi 0W with a RaspBee II (I also have a RaspBee that I can test with as well)
 - Firmware version: 26690700
 - deCONZ version: `2.05.77` tested on `2.11.05-debian-buster-stable` and `2.12.00-ubuntu-xenial-beta`
 - Device: RaspBee I/RaspBee II
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs
`2.05.77` has:
```15:52:21:724 COM: --dev: /home/redm/ttyAMA0 (RaspBee)```
Where as any version onward does not have this line.
I am running deCONZ via `/usr/bin/deCONZ --dev=/home/redm/ttyAMA0 --upnp=0` on any version I test.

## Additional context
While the serial redirect may seem odd, it worked in the past and was the entire reason I wanted a deCONZ set up.
The reason for doing it this way is that it makes integration with my VM cluster much easier and deCONZ in its entirety is backed up nightly and I don't have to worry about the rpi dying or the SD card in the pi dying.

manup, 2021-07-02:
Is `/home/redm/ttyAMA0` a symlink?

Red-M, 2021-07-02:
its technically a serial device presented via remserial in this case.

manup, 2021-07-02:
Can you please provide the output of:

`ls -l /home/redm/ttyAMA0`

The early log output of starting deCONZ as follows would also help, there should be some "COM: ..." lines printed.

```
deCONZ --dbg-info=1 --dev=/home/redm/ttyAMA0
```

Red-M, 2021-07-02:
That COM never shows up on later versions.
```
$ ls -l /home/redm/ttyAMA0
lrwxrwxrwx 1 redm redm 10 Jul  2 21:01 /home/redm/ttyAMA0 -> /dev/pts/0
```

manup, 2021-07-02:
Sorry my bad wrong debug level, should be 2:

```
deCONZ --dbg-info=2 --dev=/home/redm/ttyAMA0
```

>`$ ls -l /home/redm/ttyAMA0`
>`lrwxrwxrwx 1 redm redm 10 Jul  2 21:01 /home/redm/ttyAMA0 -> /dev/pts/0`

Hm interesting so the real device is `/dev/pts/0`.

Red-M, 2021-07-02:
Yeah I know the line you're asking for, it doesn't show up on newer versions.

manup, 2021-07-02:
Even with debug level 2?

`COM: ...` should always be printed for all serial com devices which are supported by Qt, there is no filter at this point where it is printed.

Red-M, 2021-07-02:
```
$ /usr/bin/deCONZ -platform minimal --http-port=8125 --auto-connect=1 --d
ev=/home/redm/ttyAMA0 --dbg-info=2 | grep COM
libpng warning: iCCP: known incorrect sRGB profile
23:21:38:638 CTRL. 3.27.223:21:38:704 COM: --dev: /home/redm/ttyAMA0 (RaspBee)
23:21:42:556 COM: --dev: /home/redm/ttyAMA0 (RaspBee)
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
^C
$ /usr/bin/deCONZ -platform minimal --http-port=8125 --auto-connect=1 --dev=/home/redm/ttyAMA0 --dbg-info=2 | grep COM
libpng warning: iCCP: known incorrect sRGB profile
This plugin does not support propagateSizeHints()
This plugin does not support propagateSizeHints()
```

Red-M, 2021-07-02:
thanks, pts devices are actually for pseudo terminal slave devices so its understandable that Qt might remove them for serial devices.

manup, 2021-07-06:
A fix (hopefully) is now in the latest alpha version.

http://deconz.dresden-elektronik.de/ubuntu/alpha/deconz-2.12.01-ubuntu-bionic-alpha~git.5e5991.c9b404_amd64.deb

Red-M, 2021-07-06:
This works!
Sadly though there were too many timeouts so I had to switch from my pi0w to a pi3-somthing with an ethernet cable (it doesn't look as nice) but its working now.

manup, 2021-07-06:
Not sure if it can be configured for the tunnel, but for TCP connections the `TCP_NODELAY` option might help with timeouts.

Red-M, 2021-07-07:
I just added the feature to Red-M/RedSSH@895be695f037da669d87c84e4270926ab8587f76 which turned `TCP_NODELAY` on for the SSH session and the tunnel. Even on ethernet this is way faster than before and the zigbee network is much more responsive than before.

Red-M, 2021-07-12:
I've been running this for about a week now without any major issues, so if this will be brought into the next stable, I'd consider this resolved.

manup, 2021-07-12:
Cool, thanks for the feedback, it will also be part of the upcoming v2.12.2-beta.

manup, 2021-07-13:
Fixed in https://github.com/dresden-elektronik/deconz-rest-plugin/releases/tag/v2.12.2-beta

closed 2021-07-13 by manup