dresden-elektronik/deconz-rest-plugin #5014: Legrand devices are blinking / switching

samsam-rolon, 2021-06-14:
## Describe the bug
Randomly, all my legrand stuff is "bliking" :
- Lights are quickly switch off/on
- micro-module too
- wall plug are quickly off/on
- shuter cover relay are quickly action (I hear a "click" from relay)

Looks like not related to an update

## Steps to reproduce the behavior
No Idea : it's random

## Expected behavior
Nothing

## Screenshots
NC

## Environment
Full Legrand Environment (router and wireless switch),, 2 ikea bulbs, 1 Xiami button, 1 Xiaomi Temp sensors with 1 CC2530+CC2591 extender for garage for SonOff Sensors and Orvibo Actioner

    Host system: (NAS Synology with Docker)
    Running method: (Marthoc Docker container )
    Firmware version: (26680700)
    deCONZ version: (2.10.04 )
    Device: (ConBee II)
    Do you use an USB extension cable: yes, USB 3.0 Hub with external power supply
    Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
Nothing ... as far as I see
[_domotique_deconz_1_logs.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6647210/_domotique_deconz_1_logs.txt)

## Additional context
NC

Smanar, 2021-06-14:
It s something new ?
It s not when deconz send periodicaly the permit join = 0 ? (But it will be every 30 mn)
All devices in same time ?
It s visible on device log on your third app too ? (to check if the state change is visible on zigbee side)
How it s random ? every days, weeks, months ?

samsam-rolon, 2021-06-14:
> It s something new ?

Few weeks / month ago but no I have no timestamp ... My GF notice me but at first, I didn't saw that in HA logs first

> All devices in same time ?

No, it's one device, at one moment. Not always the same device. On day, it's a wall plug (i know it because a camera is connected on it), an other it's the bathroom or kitchen. By night, I can sometimes ear relay for rollershutter switch.

> It s visible on device log on your third app too ? (to check if the state change is visible on zigbee side)

No, that why I didn't take care at the begining, but now I can "see" it (thanks to remote work)

> How it s random ? every days, weeks, months ?

Completly random but I can notice 1 event (what ever the device) per day

> It s not when deconz send periodicaly the permit join = 0 ? (But it will be every 30 mn)

Oh really ? Since when deCONZ do that ?
Something intresting on this part is : during pairing process of legrand devices, they are "blinking" to acknowledge the pairing. With legrand gateway and with deconz as far as I remember !

Smanar, 2021-06-14:
So for all device and 1 time every 24 h.
I think it s from the device (according to inclusion time), else you will have it at same time for all device, deconz don't use different daily periodic event for me.

I will ask pipiche to see if he have noticied that too on the zigate.

Mimiix, 2021-06-15:
06:00:02:030 0x00124B0022318070 error APSDE-DATA.confirm: 0xE1 on task
06:09:06:633 0x000474000099C9BF error APSDE-DATA.confirm: 0xE1 on task
06:36:43:860 0x00124B0022318070 error APSDE-DATA.confirm: 0xE9 on task

There's some issues there aswell(interference). Should check that too.

Smanar, 2021-06-15:
0x00124B0XXXXXXX Are TI Mac adress ?

Pipiche have said me same, try to disable CCXXXX to make tests, this stuff can cause issues.

samsam-rolon, 2021-06-16:
> So for all device and 1 time every 24 h.
> I think it s from the device (according to inclusion time), else you will have it at same time for all device, deconz don't use different daily periodic event for me.

It's one or multiple device, almost everday yes ! I didn't notice that last year so looks like related to a recent update ?

> I will ask pipiche to see if he have noticied that too on the zigate.

Thanks !

> 
> 
> 06:00:02:030 0x00124B0022318070 error APSDE-DATA.confirm: 0xE1 on task
> 06:09:06:633 0x000474000099C9BF error APSDE-DATA.confirm: 0xE1 on task
> 06:36:43:860 0x00124B0022318070 error APSDE-DATA.confirm: 0xE9 on task
> 
> There's some issues there aswell(interference). Should check that too.

0x00124B0022318070  : It's my Garage Door action which is borderline "out of range" of my zigbee network
0x000474000099C9BF : Rolershutter switch Legrand. IDK why there is some errors, this device is very well placed in my network, far from Wifi AP. Maybe too near an existing device ? All rollershutter a placed just over an wired wall switch

CCXXXX ? I have only one active : CC2530_CC2591 for extending range to my garage door. It's active since more than one year without any issue. 

I have tested CC2652RB_router with zstack firmware 3.0 but it's unplug (not improving range such as CC2530_CC2591 )

Mimiix, 2021-06-16:
@samsam-rolon 0xE1 is mostly done on the coordinator.

Smanar, 2021-06-16:
And as you have it on all devices, I don't think it s from a specific router ....
Nothing on log, so probably device based, but what can provoke that ?

I know it exist on some "car charger" to be sure the relay is not "frozen", but totaly stupid on a plug ....

I will ask to Legrand if they have a tips ...

samsam-rolon, 2021-06-17:
Hello
I have catch logs at the moment or a device "blink"
I turn on a light at 10:41:38 and off at 10:45:51 and between this laps of time, I got a blinking.
Incriminated device is 0x00047400008b0567
[image]

```logs
today at 10:41:00 10:40:59:933 ZCL attribute report 0x00047400000BA4F7 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:00 10:41:00:153 ZCL attribute report 0x000474000099C9BF for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:00 10:41:00:373 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290900, zclSeq: 98
today at 10:41:00 10:41:00:374 ZCL attribute report 0x00047400000BA4C3 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:01 10:41:01:052 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
today at 10:41:01 10:41:01:891 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001000, zclSeq: 138
today at 10:41:02 10:41:01:892 ZCL attribute report 0x00047400000BA4F7 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:02 10:41:02:376 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
today at 10:41:03 10:41:03:623 ZCL attribute report 0x00047400008B0567 for cluster: 0x0008, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:04 10:41:04:846 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290000, zclSeq: 58
today at 10:41:04 10:41:04:847 ZCL attribute report 0x00047400000BA4F5 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:05 10:41:04:956 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290000, zclSeq: 36
today at 10:41:05 10:41:04:957 ZCL attribute report 0x00047400000BA50F for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:05 10:41:05:631 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290000, zclSeq: 133
today at 10:41:05 10:41:05:632 ZCL attribute report 0x00047400000BA4F6 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:06 10:41:06:840 max transmit errors for node 0x00047400008AF4A3, last seen by neighbors 477134 s
today at 10:41:06 10:41:06:848 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001000, zclSeq: 59
today at 10:41:06 10:41:06:848 ZCL attribute report 0x00047400000BA4F5 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:07 10:41:07:358 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
today at 10:41:07 10:41:07:586 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290B00, zclSeq: 99
today at 10:41:07 10:41:07:586 ZCL attribute report 0x00047400000BA4C3 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:07 10:41:07:880 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
today at 10:41:08 10:41:07:965 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001000, zclSeq: 37
today at 10:41:08 10:41:07:965 ZCL attribute report 0x00047400000BA50F for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:09 10:41:09:068 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
today at 10:41:09 10:41:09:470 [INFO] - No button map for: WarningDevice, unicast to: 0x0000, endpoint: 0x01, cluster: IAS_ZONE (0x0500), command: STATUS_CHANGE (0x00), payload: 100000640000, zclSeq: 86
today at 10:41:12 10:41:12:575 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: ONOFF (0x0006), command: ATTRIBUTE_REPORT (0x0A), payload: 00001000, zclSeq: 134
today at 10:41:12 10:41:12:576 ZCL attribute report 0x00047400000BA4F6 for cluster: 0x0006, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:13 10:41:13:557 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290900, zclSeq: 100
today at 10:41:13 10:41:13:558 ZCL attribute report 0x00047400000BA4C3 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:41:14 10:41:14:950 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
today at 10:41:16 10:41:16:008 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
today at 10:41:16 10:41:16:963 max transmit errors for node 0x00047400008AF4A3, last seen by neighbors 477144 s
today at 10:41:17 10:41:17:114 ZCL attribute report 0x00124B001FC6428B for cluster: 0x0006, ep: 0x08, frame control: 0x00, mfcode: 0x0000 
[... cut ...]
today at 10:45:58 10:45:58:696 ZCL attribute report 0x00047400000BA4F7 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:45:58 10:45:58:854 ZCL attribute report 0x000474000099C9BF for cluster: 0x0102, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
today at 10:45:59 10:45:59:603 [INFO] - No button map for: Connected outlet, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0B04, command: 0x0A, payload: 0B05290C00, zclSeq: 130
today at 10:45:59 10:45:59:604 ZCL attribute report 0x00047400000BA4C3 for cluster: 0x0B04, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
```

SwoopX, 2021-06-17:
The device in question announce itself to the network.

Smanar, 2021-06-17:
Ha yep right, it s like it have leave and rejoin the network.
This action update the "lastannounced" value in JSON too ?

samsam-rolon, 2021-06-18:
And so ? What does it mean ? :-/

Smanar, 2021-06-18:
Can be nothing, wireless device make that every time you use them (only for Legrand)
Somes years ago, I was able to reproduce the issue, when sending set level value = 255 instead of 254 maximum, but not possible ATM.

I have no answer from netatmo yet, but can you check if you have the "lastannounced" value updated in the JSON ? can be a good way to check if a device have the issue or not.

samsam-rolon, 2021-06-30:
It's very strange ...

Smanar, 2021-06-30:
Still no answer from netatmo guys, Idk if It will be better to contact him direclty by email.

But have you check the "lastannounced" values in JSON to check if it s the same time than during your issue ?

samsam-rolon, 2021-07-12:
Hello
Sorry for delay ... Time to catch the event and got logs ...

Yeah, lastannounced looks like be updated when light blink !

Smanar, 2021-07-12:
>Yeah, lastannounced looks like be updated when light blink !

So it s a good way to find buggy device or if someone else have the same issue.

No answer from the Legrand guy, I will try to contact him direclty by mail.

samsam-rolon, 2021-07-13:
Okay ! On my Side, no update regarding releasing Legrand Firmware ...

closed 2021-08-12 by github-actions[bot]

samsam-rolon, 2021-08-15:
Stay away bot !

Smanar, 2021-08-15:
Still no idea on my side, and my contact at Legrand is not realy talkative ATM.

Smanar, 2021-08-28:
For me, It's not so easy, remember the profalux issue, was just the "permit join security" that provoke a short cycle on device.
It have worked before, perhaps there is a new setting or a working mode not liked by device

For pipiche it s provoked by CCXXXX devices.

Perhaps try to look on the mesh can help, if all bugged devices are in the same zone, or use the same router, ...

samsam-rolon, 2021-10-12:
Hello
Sorry, I forgot to awnser !
I've remove my CCxxxx device except my extender CC2530 + CC2591 router.
I don't know why they affect my network ...

Smanar, 2021-10-12:
And it s better after that ?