dresden-elektronik/deconz-rest-plugin #5016: Loading backup failed

chris58008, 2021-06-14:
## Describe the bug

When trying to restore a backup from a raspbee (2.05.64 / 4/22/2019)  (Firmware 26330500)
to a fresh install on windows (2.12.00 / 5/9/2021) (Firmware 26580700)

I get the error message 

Loading backup failed
! 

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

[image]

## Environment

 - Host system: (PC)
 - Running method: (Windows 10 21H1)
 - Firmware version: (26580700)
 - deCONZ version: 2.12.00 )
 - Device: (ConBee II)
 - Do you use an USB extension cable: (no) 
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

SwoopX, 2021-06-14:
Duplicate #4962

SwoopX, 2021-06-14:
Duplicate #4962

closed 2021-06-14 by SwoopX