dresden-elektronik/deconz-rest-plugin #5020: Xiaomi sensor not working with decons 6.9.0

hofvanta, 2021-06-15:
## Describe the bug
The sensor are decovered and seen (not greyed out) in Phoscon, but mentioned status and or activity of the Xiaomi sensor are not visible, also Deconz see the sensor (door/PIR) as not reachable after upgrade from 6.8.0 to 6.9.0. This is the case for all sensors. Resetting the sensor (three short presses) or reconnected didnt solve the problem.
Rolling back was fixing the problem again. In HA they had a clear status.

## Steps to reproduce the behavior
Upgrade to 6.9.0.

## Expected behavior
Sensor shown as not reachable in Deconz.

## Screenshots

## Environment
Conbee-II
Deconz 6.9.0 and HA core 2021.6.3 and Supervisor 2021.05.4.
Last working HA core 2021.6.3 & Supervisor 2021.05.4 with Deconz 6.8.0.
Running on Pi3B

## deCONZ Logs
No special logging available. Please let me know how to see them and Im happy to check.

## Additional context

Mimiix, 2021-06-15:
Canyou please add the deconz version instead of the addon version?

Also, can we get some logs? Refer to the HA Documentation on how to get them.

hofvanta, 2021-06-15:
The version is 2.10.04, firmware 26660700 in case of 6.8.0. After upgrade to 6.9.0 again, 2.11.05 the problem is back again. How Im able to get the logs for you? With the logger function?

Mimiix, 2021-06-15:
For logs, check the documentation:
https://github.com/home-assistant/addons/blob/master/deconz/DOCS.md#advanced-debug-output-control

I'd klike to have error 2, aps 2 and info 2.

For xiaomi devices, this also coudl be checked: 
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

hofvanta, 2021-06-16:
It seems you fixed the problem. The second link did the trick. Why it happend after an upgrade?

Mimiix, 2021-06-17:
No clue. @SwoopX might be able to answer that.

closed 2021-07-16 by github-actions[bot]