dresden-elektronik/deconz-rest-plugin #5021: Legrand Netatmo - 4 Scenes Wireless Remote Command with battery (077709L, 600091, 600191, 067755, 067756, 067757)

ftitou, 2021-06-15:
Hi,

I succeed to pair the device with deCONZ but not with Phoscon. I guess the Device Request shall be completed first, right?

Thanks!

## Device

- Product name: 4 Scenes Wireless Remote Command with battery
- Manufacturer: Legrand
- Model identifier: Pocket Remote
- Device type : 
  - Remote

[image]

## Screenshots

### Basic
[image]

### Identify
[image]

### Poll Control
[image]

### Power Configuration
[image]

### Node Info
[image]

### Scenes (greyed)
[image]
[image]

### Identify (greyed)
[image]

### Basic (greyed)
[image]

### OTAU
[image]
[image]

Smanar, 2021-06-15:
Ha yes, totaly skipped it.
Do you have a linux machine with full OS to test code modifications ?

ftitou, 2021-06-16:
Hello, thanks a lot for your answer. I am currently using Jeedom on RPI4, I am not sure it is a full OS but if requested I can install one.

Smanar, 2021-06-16:
I think it s a complete OS, but the compilation is realy CPU intensive, if you have a small rapsberry or another OS I think it will be better, and with a desktop if possible, to use the GUI.

So first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch legrand_scene https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have a small raspberry,it can freeze during compilation, my pi 3 do, so remove "-j2"

After that you will be able to include it, but need to fill the json file.
If this part is working, on logs (deconz/help/debug view if you have the GUI) with the flag "info", you will see an error message everytime you will use the switch.
I need the error message by button pls ?

ftitou, 2021-06-17:
I installed a freesh new OS on my RPI4 according to : https://phoscon.de/en/conbee2/sdcard using Raspbian Buster Desktop => Stable image with desktop based on Debian Buster.

Using Putty, I followed your instruction but I got an error message after:
`qmake && make -j2`

(same error with or without -j2)
[image]

So I am currently installing a VM on Win10 to run Ubuntu and will try again. I hope it will help.

Smanar, 2021-06-17:
I m restesting on my side, it work, it s probably because you have not the last version of dev-package
You update the OS or just try 

`sudo apt install deconz-dev`

You need to have the version 2.12.00

ftitou, 2021-06-18:
Hi,

You are right I was not using 2.12.00 but the latest stable issue which is older.  I failed to install a VM despite a long time to work on it so I skipped it. But the good news is I succeed to install Deconz 2.12.0 on RPI4 with a GUI.

I successfully passed the following code without any error this time :-)

> ```
> sudo apt install deconz-dev
> git clone --branch legrand_scene https://github.com/Smanar/deconz-rest-plugin.git
> cd deconz-rest-plugin
> qmake && make -j2
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
> ```

> After that you will be able to include it, but need to fill the json file.

I did not know what to modify on JSON file then I skipped this part, may I ask you some help for that?

I tried to go further, then I  : 
1.  Launched Deconz
2. Launched Phoscon app
3.  Activated deconz/help/debug view with flag "info"
4.  Launched a pairing process in Phoscon which looks successfull meaning that I got a green LED on Legrand device. But it was the case as well before my "Device Request" on Github. Phoscon did not inform me that the pairing was sucessfull. 
5. I click on button 1 to 4, it generates the following line in debug view so I am not sure it will helps : 
`16:47:04:440 FP indication 0x0104 / 0x0005 (0x00047400008340F2 / 0x07BB)
16:47:04:441                       ...     (0x00047400008340F2 / 0xF504)
16:47:05:666 FP indication 0x0104 / 0x0005 (0x00047400008340F2 / 0x07BB)
16:47:05:667                       ...     (0x00047400008340F2 / 0xF504)
16:47:06:911 FP indication 0x0104 / 0x0005 (0x00047400008340F2 / 0x07BB)
16:47:06:912                       ...     (0x00047400008340F2 / 0xF504)
16:47:07:963 FP indication 0x0104 / 0x0005 (0x00047400008340F2 / 0x07BB)
16:47:07:964                       ...     (0x00047400008340F2 / 0xF504)`

Complete log file : 
[Logdeconz.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6678717/Logdeconz.txt)

> read again the basic attribute on deconz

I got this : 
[image]

Did I missed something? Maybe the JSON part?

Smanar, 2021-06-18:
Don't worry, the json is the last part, and the easier, this one can be done just with a text editor.

The device seem reconisable in the api, but he have left the network just at last step ^^. It happen with Legrand.
The inclusion was not finished on your log, easy to see, on deconz you still have the network id as name > 0xBD99.

If you don't want to repeat this step, can use this method

-    set phoscon in permit join (add new sensor)
-    read again the basic attribute on deconz (need probably to wake it up)

When he will be correclty included, you will see the message in log, the node nme will change in deconz. And when you will use a button you will see a message like

> [INFO] - No button map for:.........
> [INFO] - Button ............

ftitou, 2021-06-18:
In order to give you the basic attribute on deconz I have already did the following : 
1. set phoscon in permit join (add new sensor)
2. read again the basic attribute on deconz (need probably to wake it up)

So I do not see what I should do differently this time. Trying again to pair the device until Phoscon discover it?

Smanar, 2021-06-18:
Ha ?
You don't need to put phoscon in permit join to get basic attribute for capture ^^.

You need to have in log 
>SensorNode XXXX added

or have a special name instead of the id network (0xXXXX) to be sure the device is correclty included

You can try to re-include it, but without deleting the old one, will be faster, was just to finish it.

On your log, we can see you have reached the last step

>16:46:40:727 [4.2] get basic cluster attr 0x0005 for 0x00047400008340f2

But the device have made a rejoin (and get a new network id)

> 16:46:46:086 0xF504 nwk changed to 0x07BB
> 16:46:46:088 device announce 0x00047400008340F2 (0x07BB) mac capabilities 0x80

But can be a problem from code too, from that I m reading the scene cluster 0x0005 is enought to trigger a ZHASwitch creation, but I can be wrong.

ftitou, 2021-06-18:
I havevtried so many times. So my procedure is as follows : 
1/ delete node in deconz
2/ open phoscon
3/ push 8sec legrand reset button until it turns red, then 3 green blink then red for 1sec
4/ switch/add new switch / other
5/ push 8sec legrand reset button until it turns red. When i release the button it starts blinking green several time and i push the button each seconds until both led are green meaning i guess pairing is successfull
6/ phoscon does not confirmed it worked, after some time i got "failed to connect"
7/ in deconz i see the module name 0x something, not a real name
8/ i read the basic several times, the first time i saw info changed as you see in previous screenshot but i never succeed to have something else new even clicking on device button to wake it up. I never succed to have sensor node xxxx added in the log.
9/ repeat all these steps except the first one

Am I doing something wrong?

Smanar, 2021-06-19:
No all is perfect, perhaps the device go to sleep too fast (but you are using action to prevent it going to sleep).
I can add some line in code to follow the inclusion, but I will explain what I m looking in log

> 16:46:37:007 [1] get node descriptor for 0x00047400008340f2
> 16:46:37:709 [2] get active endpoints for 0x00047400008340f2
> 16:46:38:694 [3] get simple descriptor 0x01 for 0x00047400008340f2
> 16:46:39:739 [4.1] Get manufacturer code
> 16:46:39:740 [4.2] get basic cluster attr 0x0004 for 0x00047400008340f2
> 16:46:40:727 [4.1] Get model ID
> 16:46:40:727 [4.2] get basic cluster attr 0x0005 for 0x00047400008340f2

ATM on your previous logs it s stopping here, You need to have after

> [4.1] Get date code

or

> [4.1] Get sw build id

But according to timer, the device have made announce 5 s before, perhaps too much.
You don't have more logs during the second inclusion ? or if using the trick (don't need step 3 and 5, so easier, just need to use the device to make it awake)

I ask that, because I don't see something blocking in the code at this step.

Smanar, 2021-06-19:
Ok have found a problem.
>Model identifier: Pocket Remote

It s not "Remote" but "remote", have corrected on the code, except on whitelist, so no way for the device will works .....

ftitou, 2021-06-20:
Thanks a lot for your help. Things are going better now with your code update.

I follow the previous procedure : 

> 1/ delete node in deconz
> 2/ open phoscon
> 3/ push 8sec legrand reset button until it turns red, then 3 green blink then red for 1sec
> 4/ in poscon : switch/add new switch / other => permit to join
> 5/ push 8sec legrand reset button until it turns red. When i release the button it starts blinking green several time and i push the reset button each seconds until both led are green meaning i guess pairing is successfull

As previously, phoscon does not confirmed it worked but in Deconz I saw a real ID node name (Pocket remote 2 instead of 0xXXXX).

It gives the following info : 

[image]

[image]

[log_with_all_steps - 1st trial.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6681764/log_with_all_steps.-.1st.trial.txt)

As mentioned in this log : 
Pressing button 1 gives : 
`04:26:09:543 [INFO] - No button map for: Pocket remote, broadcast to: 0xFFED, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EDFF010000, zclSeq: 7`

Pressing button 2 gives : 
`04:26:11:134 [INFO] - No button map for: Pocket remote, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: ECFF010000, zclSeq: 8`

Pressing button 3 gives : 
`04:26:12:131 [INFO] - No button map for: Pocket remote, broadcast to: 0xFFEB, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EBFF010000, zclSeq: 9`

Pressing button 4 gives : 
`04:26:13:231 [INFO] - No button map for: Pocket remote, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), command: RECALL_SCENE (0x05), payload: EAFF010000, zclSeq: 10`

Then I tried to do long-press instead of short-press as above to see if there is a difference but not. Then I guess these button do not support this functionality (but they are not supposed to). I just wanted to try it.

Then I wanted to test your recommandation to not perform step 3 and 5 to speed the process. Then I close Phoscon and Deconz windows and launched Deconz again. I was suprised to see the ID Node Name came back to 0xXXXX instead of "Pocket Remote 2". It's strange, right? Looks like Deconz did not save the device name.
[image]
 
So I tried to do again the procedure without step 3 and 5 as you mentionned. I never succeed to have something else than "Failed to Connect" in Phoscon. Here is the log :  

[log_without step 3 and 5.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6681770/log_without.step.3.and.5.txt)

So I did the full procedure again (with 3 and 5). It gives the following for info if you need to see a 2nd inclusion trial.
[log_with_all_steps - 2nd trial.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6681772/log_with_all_steps.-.2nd.trial.txt)

Smanar, 2021-06-20:
Lol, thx a lot, support is easier with guys like you, a pleasure.

You are not the first one with a failling procedure without step 3 and 5, but I don't understand, it was worked before ...
Perhaps will be better if asking node descriptor too, but will search on my side, don't loose time with that, your device is fully included, so you will not need to remove it now.

Have updated the code a last time, to manage scene, Legrand use a special method.

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Now the Json part. So you know how switches are working on deconz ? All switches have a field in state/buttonevent, with a values X00Y, the X value is the button number and the Y is the action. Some third app manage direclty this value so you have nothing to do, IDK wich one you use ?

Now on your machine pls check the json file in /usr/share/deCONZ/devices/button_maps.json with a simple text editor, and try thoses changes https://github.com/Smanar/deconz-rest-plugin/commit/0411a8a95e04551737ac0d221af11a6e5bd0e3c1#diff-a3adce467fe93b9c612c198c187dad34f5284d6c9adf66aa28127f0f619979b5 (only the first part, the second one if the modification to support Legrand scene in x++  code)

Don't need to re-include the device, but you need to restart deconz.

ftitou, 2021-06-20:
So on my rpi4 with only deconz installed, i updated again your code, json file updated. I did not see any difference. When launching deconz, device id were lost (0xXXXX instead of pocket remote). So I start a new inclusion again. Still no successfull message in phoscon.

It is normal or should i change to go now to my jeedom installation instead and try? (Actuallly i started to do that but face some trouble installing your code on it, but i will investigate a bit more tonight)

Smanar, 2021-06-20:
The compilation is "host dependent" so you need to compile on the same host you use deconz (and you can't have 2 instances running, so if jeedom is using the headless version, you can do nothing more on this host)

But it have worked previously, have checked again the last modification, it impact only the scene cluster management, no change on inclusion part.

>[INFO] - No button map for: Pocket remote

This one mean, the model id "Pocket remote' is not found on the json. Are you sure there is not a typo ?

If I m wrong you need to have

>[INFO] - No button handler for:

Can have issue because the device use broadcast request instead of unicast, but can be native and not configurable.

Smanar, 2021-06-20:
Have updated the code again, trying to prevent the code be locked on
```

04:24:33:290 [4.1] Get sw build id
04:24:33:291 [4.2] get basic cluster attr 0x4000 for 0x00047400008340f2
04:24:42:554 [4.1] Get sw build id
04:24:42:554 [4.2] get basic cluster attr 0x4000 for 0x00047400008340f2
04:24:51:746 [4.1] Get sw build id
04:24:51:747 [4.2] get basic cluster attr 0x4000 for 0x00047400008340f2
04:25:10:105 [4.1] Get sw build id
04:25:10:106 [4.2] get basic cluster attr 0x4000 for 0x00047400008340f2
```

ftitou, 2021-06-22:
Hi, sorry for having taking time to answer. I faced many issue with Jeedom and I had finally to reinstall it from scratch... Whatever to summarize the current status is:
- Each time I close and relaunch deconz, the device ID goes back to 0xXXXX.
- You asked me if my third app (jeedom) is able to see the button action

As a refresh to follow all your recommandation I install a brand new system based on a new SD card based on https://phoscon.de/en/conbee2/sdcard without jeedom. I still have another SD with my jeedom that I was not using on this thread discussion but now I will.

Part A : System with only deconz installed

 On this system, I updated your last code according to the following procedure. Unfortunately the result is still that at each time I close and relaunch deconz, the device ID goes back to 0xXXXX. I checked json file which looks like OK.

Procedure I followed to upadate the code
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

JSON file screenshot, it seems to be OK : 
[image]

Part B : Jeedom System
I followed this procedure : 
```
First I stopped jeedom Demon
sudo apt install deconz-dev
git clone --branch legrand_scene https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
sudo apt show deconz => confirmed I am using deconz 2.12.01
```
JSON file updated : 
[image]

I succeed to pair the device using Jedoom GUI.  When I push a button no reaction in Jeedom (read command remain empty whereas on other device it displays a value)

Smanar, 2021-06-22:
Ok so first, good new the json is working
```
10:35:00:164 [INFO] - Button 4002 - Pocket remote, broadcast to: 0xFFED, endpoint: 0x01, cluster: SCENES (0x0005), action: Scene 4, payload: EDFF010000, zclSeq: 3
10:35:01:468 [INFO] - Button 3002 - Pocket remote, broadcast to: 0xFFEC, endpoint: 0x01, cluster: SCENES (0x0005), action: Scene 3, payload: ECFF010000, zclSeq: 4
10:35:01:475 create group 65516 for sensor 2
10:35:03:028 [INFO] - Button 2002 - Pocket remote, broadcast to: 0xFFEB, endpoint: 0x01, cluster: SCENES (0x0005), action: Scene 2, payload: EBFF010000, zclSeq: 5
10:35:04:062 [INFO] - Button 1002 - Pocket remote, broadcast to: 0xFFEA, endpoint: 0x01, cluster: SCENES (0x0005), action: Scene 1, payload: EAFF010000, zclSeq: 6
```

Your procedure in jeedom is good as you are able to include it.

But now need to check why you have no update value, and why the device is badly saved.
I think I have somthing to explore
>10:35:03:038 delete old group 65516 of sensor Pocket remote 2
>10:30:50:601 SensorNode 2: Switch 2 added

Same ID (2) but not same name ...

Just by curiosity, when the inclusion succed (and have a name instead of the 0xXXXX) are you able to check the json in the API ?

If I m right you will not be able to find it.

ftitou, 2021-06-22:
> Just by curiosity, when the inclusion succed (and have a name instead of the 0xXXXX) are you able to check the json in the API ?

Could you please tell me how I can check that please?

Smanar, 2021-06-22:
Can use http://phoscon.de/pwabeta with help/api information or in jeedom "données brutes"

ftitou, 2021-06-22:
...

[image]
[image]

Smanar, 2021-06-22:
Hu ?
lol, how jeedom can reconize it just with "[]", but it have reconized it as switch, so I think you have type="ZHASwitch", but probably no "state/buttonevent".
Tu es sur que c'est les "informations brutes" de l'appareil ?

Pizzas are waiting for me,will check after if I can see something that can explain the reset at reload, I think it s because the device have only the cluster 0x0005.

Smanar, 2021-06-22:
Ok, so I haven't found something special ....
Can you update the code ?
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I have updated the code for the uniqueID use the scene cluster, but I don't see how it can have an impact, was more to make test.
Can you take logs using "info_l2" too ? (in combinaison with "info" like that, I will see websocket communication.

ftitou, 2021-06-22:
OK I will update the code tomorrow. 

Tonight I tried again  to read "information brute" in jeedom. As I was sure I select the good one I added another kind of Legrand device to see the difference. The problem was the demon was not activated + I had to pair several time the device to have something working completely as described below.

So the result with the 4 scene device : 
[image]

As it is now better I checked again if Jeedom was able to read button. It is now working : for button Scene 4 I got : 
[image]
4002 => Scene 1
3002 => Scene 2
2002 => Scene 3
1002 => Scene 4

As soon I restart Jeedom, the values are no more read :-( To make it work again I have to include the device again.

I will check tomorrow again with my system with only deconz installed if he is able to save the device id instead of 0xXXXX. Then I will update your code and try again.

ftitou, 2021-06-23:
In addition to my previous message : 
In order to compare the behaviour with 4 scene device with another device from legrand (wireless shutter), I performed the test with both.

**Before your SW code update (after closing and launching again deconz):** 
NB : Remote Switch = Wireless shutter device
[image]
=> 4scene device is not saved

[log_06_23_before SW update.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6699513/log_06_23_before.SW.update.txt)
Before 110047 => wireless shutter inclusion
After  110047 => 4 scene inclusion

**After your SW code update (after closing and launching again deconz):**
[image]
=> now it looks like the device is correctly saved. Great, thanks!

However the device is still not recognized by Phoscon.
Even after your SW code update (but as well before), I get this for 4 scene device (actually after 3 minute I got the error message)
[image]

Whereas with the wireless shutter I got this:
[image]

[log_06_23_after SW update.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6699514/log_06_23_after.SW.update.txt)
Before 060645 => wireless shutter inclusion
After  060744 => 4 scene inclusion

And just to be sure that button were correctly read after deconz restart : 
[log_06_23_after SW update_after deconz reboot.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6699516/log_06_23_after.SW.update_after.deconz.reboot.txt)
=> looks fine now

Smanar, 2021-06-23:
>However the device is still not recognized by Phoscon.

It's normal, not all devices are supported by deconz, but you can use jeedom ?
And I think it s for that too, phoscon don't display a message, because I have take a look on logs, websocket send exactly the same request after inclusion

>06:06:23:973 Websocket 192.168.8.1:39520 send message: {"e":"added","id":"2","r":"sensors","sensor":{"config":{"battery":null,"group":"9","on":true,"reachable":true},"ep":1,"etag":"a1d805829bccdcc6fdf4175f5e07fafc","id":"2","lastseen":"2021-06-23T06:06Z","manufacturername":"Legrand","mode":1,"modelid":"Shutters central remote switch","name":"Shutters central remote switch 2","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"00:04:74:00:00:b0:1a:74-01-0102"},"t":"event","uniqueid":"00:04:74:00:00:b0:1a:74-01-0102"} (ret = -1095166864)

>06:07:10:520 Websocket 192.168.8.1:39520 send message: {"e":"added","id":"3","r":"sensors","sensor":{"config":{"battery":null,"group":"10","on":true,"reachable":true},"ep":1,"etag":"586df0f106ab3be3c6c1c736c269ce29","id":"3","lastseen":"2021-06-23T06:07Z","manufacturername":"Legrand","mode":1,"modelid":"Pocket remote","name":"Pocket remote 3","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"00:04:74:00:00:83:40:f2-01-0005"},"t":"event","uniqueid":"00:04:74:00:00:83:40:f2-01-0005"} (ret = -1095166864)

Will ask to discord by curiosity.

This device have an issue too, the "group" management. Technicaly it don't use a group where you can put another device like the other remote, but 4, one by button, and the code is not ready for that.

But more worrying, you still have that in loop after inclusion finished ?
```
06:07:52:034 discard double entry in binding queue (size: 3) for for 0x00047400008340F2, cluster 0x0005
06:07:52:034 discard double entry in binding queue (size: 3) for for 0x00047400008340F2, cluster 0x0005
```

Or it s just looping during the inclusion ?

ftitou, 2021-06-25:
> Or it s just looping during the inclusion ?

Seems to be only during inclusion I guess according to this log :
[log_06_23.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6715009/log_06_23.txt)

I confirm using Jeedom, the 4 scene device is recognized and that Jeedom is able to read the button. However after restarting Jeedom, Jeedom is no more able to read the button (value remains stucked at the last value read before restarting). Same bahaviour for Wireless Shutter device.  Either maybe the way I installed the beta deconz SW on jeedom broke something (I don't know why but I read it is tricky to install a deconz version manually on jeedom). Or is there a possibility that something in the code prevent reading the button after jeedom restart? 

To make it work again I need to perform an inclusion again.

Smanar, 2021-06-25:
>I don't know why but I read it is tricky to install a deconz version manually on jeedom

Yeah I know, but it s easier than on HA ^^.
It s because jeedom try to help users, and because of that, deconz is too much "integrated". On classic install you just need to use "sudo apt install deconz"

But your issue is something new.
I think when you say "restarting Jeedom" it mean jeedom + deconz ?
You have this issue since my last changes or since the start ?

If I m right you can enable deconz log in jeedom (you can set the log level too, deamon logs), can you take a look if you can see something ? or you can use too the phoscon beta link http://phoscon.de/pwabeta in help / api information
There is a tool to see websocket notification in same time, can be used to see if the problem is from deconz or jeedom.

ftitou, 2021-06-25:
> I think when you say "restarting Jeedom" it mean jeedom + deconz ?

Yes full RPI reboot.

> You have this issue since my last changes or since the start (for the Wireless Shutter device, this one is working) ?

I'd rather not to say because I did not test this specifically. If you need I can install jeedom from scratch to have a look on it.

> If I m right you can enable deconz log in jeedom (you can set the log level too, deamon logs),

Ok for deconz but for deamon, do you know which one it is?
[image]

>  you can use too the phoscon beta link http://phoscon.de/pwabeta in help / api information
> There is a tool to see websocket notification in same time, can be used to see if the problem is from deconz or jeedom.

Below is what I got but I am not sure it was you were expecting... If not tell me.

[Websocket.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6718551/Websocket.txt)
[API.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6718553/API.txt)

Smanar, 2021-06-25:
>I'd rather not to say because I did not test this specifically. If you need I can install jeedom from scratch to have a look on it.

Too much work, I hope we can solve it with an other way

For log, I thought about 
[image]
in "log et surveillance"

And for websocket
[image]

This log websocket events, it's the way used by deconz to send changes to third app.
If you see something in it and not in jeedom > jeedom issue
If you see nothing in it > deconz issue.

ftitou, 2021-06-26:
Here are the log : 
[deconz - after reboot.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6719653/deconz.-.after.reboot.txt)
[deconzServer - after reboot.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6719654/deconzServer.-.after.reboot.txt)

Maybe not exactly the right time but it gives an idea of what I did : 
[2021-06-26 08:04:35] => starting inclusion from scratch
[2021-06-26 08:05:54][DEBUG] => I pushed on device button, works fine, Jeedom can read it
> [2021-06-26 08:08:17][DEBUG] : {"00212EFFFF06AB5E":{"e":"scene-called","gid":"65516","r":"scenes","scid":"1","t":"event"}}
> [2021-06-26 08:08:17][DEBUG] : {"00212EFFFF06AB5E":{"e":"changed","id":"2","r":"sensors","state":{"**buttonevent":3002**,"lastupdated":"2021-06-26T06:08:17.902"},"t":"event","uniqueid":"00:04:74:00:00:83:40:f2-01"}}

[2021-06-26 08:08:56][DEBUG]  => restart jeedom (Réglages => Système => Redemarrer)
[2021-06-26 08:10:33.966][DEBUG] : => when pushing on device button I don't get the same result as I got at 08:05. Jeedom is no more able to read button

> [2021-06-26 08:11:01.380][DEBUG] : Received message from gateway 127.0.0.1 : {"e":"scene-called","gid":"65516","r":"scenes","scid":"1","t":"event"}
> [2021-06-26 08:11:01.381][DEBUG] : Send to jeedom :  {'00212EFFFF06AB5E': {'e': 'scene-called', 'gid': '65516', 'r': 'scenes', 'scid': '1', 't': 'event'}}
> [2021-06-26 08:11:01.385][DEBUG] : Starting new HTTP connection (1): 127.0.0.1:80

Now regarding the API information, after new inclusion on jeedom: 

>                     {
>     "08:31:20:804": {
>         "e": "scene-called",
>         "gid": "65517",
>         "r": "scenes",
>         "scid": "1",
>         "t": "event"
>     },
>     "08:31:20:849": {
>         "config": {
>             "battery": 80,
>             "group": "65517",
>             "on": true,
>             "reachable": true
>         },
>         "e": "changed",
>         "id": "2",
>         "r": "sensors",
>         "t": "event",
>         "uniqueid": "00:04:74:00:00:83:40:f2-01"
>     },
>     "08:31:20:850": {
>         "e": "changed",
>         "id": "2",
>         "r": "sensors",
>         "state": {
>             "buttonevent": 4002,
>             "lastupdated": "2021-06-26T06:31:20.889"
>         },
>         "t": "event",
>         "uniqueid": "00:04:74:00:00:83:40:f2-01"
>     },
>     "08:31:32:643": {
>         "e": "scene-called",
>         "gid": "65516",
>         "r": "scenes",
>         "scid": "1",
>         "t": "event"
>     },
>     "08:31:32:692": {
>         "config": {
>             "battery": 80,
>             "group": "65516",
>             "on": true,
>             "reachable": true
>         },
>         "e": "changed",
>         "id": "2",
>         "r": "sensors",
>         "t": "event",
>         "uniqueid": "00:04:74:00:00:83:40:f2-01"
>     },
>     "08:31:32:693": {
>         "e": "changed",
>         "id": "2",
>         "r": "sensors",
>         "state": {
>             "buttonevent": 3002,
>             "lastupdated": "2021-06-26T06:31:32.728"
>         },
>         "t": "event",
>         "uniqueid": "00:04:74:00:00:83:40:f2-01"
>     }
> }

                    
And now after restarting jeedom : 

>                     {
>     "08:28:48:981": {
>         "attr": {
>             "id": "1",
>             "lastannounced": null,
>             "lastseen": "2021-06-26T06:28Z",
>             "manufacturername": "dresden elektronik",
>             "modelid": "ConBee II",
>             "name": "Configuration tool 1",
>             "swversion": "0x266b0700",
>             "type": "Configuration tool",
>             "uniqueid": "00:21:2e:ff:ff:06:ab:5e-01"
>         },
>         "e": "changed",
>         "id": "1",
>         "r": "lights",
>         "t": "event",
>         "uniqueid": "00:21:2e:ff:ff:06:ab:5e-01"
>     },
>     "08:28:57:2": {
>         "e": "scene-called",
>         "gid": "65517",
>         "r": "scenes",
>         "scid": "1",
>         "t": "event"
>     },
>     "08:29:02:581": {
>         "e": "scene-called",
>         "gid": "65516",
>         "r": "scenes",
>         "scid": "1",
>         "t": "event"
>     }
> }
>          

Looks like the API is no more sending the button action (below) explaining why Jeedom is no more able to read the value, right?
> "state": {
>         "buttonevent": 4002,           
   

> BTW, I haven't asked, do you have other device that works, or you have only those 2 on your machine for tests ? If all have issues, or if you have only thoses 2 devices, just restart the deamon.
I am only using those 2 for test, I am new on Jeedom and first trying to have everything working before starting to create scenarios. I tried to restart deamon after reboot but jeedom can still not read the value.

Smanar, 2021-06-26:
>Looks like the API is no more sending the button action (below) explaining why Jeedom is no more able to read the value, right?

right. so it s deconz issue.

>only the 4 scene device has the issue

Ok, I m reassured  ^^. So it s just this new device that have a "magic" working mode, I will take a look again on it today.

ftitou, 2021-06-26:
Legrand seems to not ease the developpers job :-( 
I put the deconz log at the very beginning of my previous message. Do you need something else?

Smanar, 2021-06-26:
Houps, sorry too much logs I m lost.
No , Legrand are zigbee compliant for me, they have some specificity, but nothing "out of rules".
This device is a little special, it have only the scene cluster and is made to work only with group, but it s not the harder one.

On previous log you have 
```
06:28:16:417 APS-DATA.indication from child 0xD627
06:28:16:418 Websocket 192.168.8.1:47422 send message: {"e":"scene-called","gid":"65517","r":"scenes","scid":"1","t":"event"} (ret = -1093054280)
06:28:16:420 [INFO] - Button 4002 - Pocket remote, broadcast to: 0xFFED, endpoint: 0x01, cluster: SCENES (0x0005), action: Scene 4, payload: EDFF010000, zclSeq: 3
```

On last one after reboot

```
06:10:49:469 APS-DATA.indication srcAddr: 0x642f, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0005, lqi: 255, rssi: -54
06:10:49:469 	asdu: 110305edff010000
06:10:49:469 APS-DATA.indication from child 0x642F
06:10:49:469 Websocket 192.168.1.12:52662 send message: {"e":"scene-called","gid":"65517","r":"scenes","scid":"1","t":"event"} (ret = -1096806472)
```

There is some difference because the second one hve more information "taked with info_l2", but will check with that.

BTW, when you have made your tries before, without jeedom, and after a reset of deconz, when the device was still reconized, button was working ?

ftitou, 2021-06-27:
> BTW, when you have made your tries before, without jeedom, and after a reset of deconz, when the device was still reconized, button was working ?

I did not checked, so let's do it!

The procedure to update the code on both system is the following : 

> cd deconz-rest-plugin
> git pull
> qmake && make -j2
> sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins

**With the system without jeedom :** 
1. I updated the code but it was already up to date
2.  With the system without jeedom, I do have some button action info in the INFO and INFO_L2 logs even after reboot :

I have something like : 
> [INFO] - Button 4002 - Pocket remote, broadcast to: 0xFFED, endpoint: 0x01, cluster: SCENES (0x0005), action: Scene 4, payload: EDFF010000, zclSeq: 3

3. I tried as well with http://phoscon.de/pwabeta, button seems to be correctly read, see log below : 

[1.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6721554/1.txt)

**With the system with jeedom :** 

1. I updated the code and it told me there was few modification, so maybe I forgot to do it last time.
2. Do an inclusion again
3. I do have the same behaviour, jeedom can read the info, after restart cannot read it.
4. Using http://phoscon.de/pwabeta, I see the button action before jeedom restart but not after, so same behaviour
5. I did it again from 1st step to ensure I did not miss something same result, except I have been told the code was already up to date of course.

I do not understand why the log I got from http://phoscon.de/pwabeta did not give the same result between system without and the one with jeedom. Maybe the way I am updating in Jeedom is not the right one.

Smanar, 2021-06-27:
>Maybe the way I am updating in Jeedom is not the right one.

It s possible, or jeedom use a different path, or perhaps a right issue, but I think in this situation you will have an error message.
On my side it work even I don't close deconz, it will just crash it and need to restart it.

Can check file information with
```
pi@raspberrypi:~/deCONZ/deconz-rest-plugin $ ls -lh /usr/share/deCONZ/plugins/libde_rest_plugin.so
-rwxrwxrwx 1 root root 2,6M juin  27 15:53 /usr/share/deCONZ/plugins/libde_rest_plugin.so
```

ftitou, 2021-06-27:
I get the following : 

-rwxr-xr-x 1 root root 2,5M juin  27 13:17 /usr/share/deCONZ/plugins/libde_rest_plugin.so

Smanar, 2021-06-27:
Ha yes, I m sure the update have failed ^^

The last code add the cluster (0x0005) to the uniqueid
>"uniqueid":"00:04:74:00:00:83:40:f2-01-0005

But on your tests with jeedom you have only "00:04:74:00:00:83:40:f2-01"

ftitou, 2021-06-28:
> Ha yes, I m sure the update have failed ^^

In a way, this is a good news. 

> Perhaps something persistant in the database ?

I tried to update again the code with no difference so I decided to relaunch a new jeedom install from scratch. I am still on it facing many issues during install process to be solved first (conbee not fully recognised by jeedom). I will do everything again and let you know

Smanar, 2021-06-28:
Your dev machine is not the same one than the production one ?
I can perhaps help you for jeedom reconization ?

But there is something strange, you are using new code on your jeedo machinbe, else the device will not be reconized, but not the last one, else you will have the cluster in the uniqueID.
Can try to delete the device using the GUI with playing with backup/restore deconz database, will be faster than re-installing jeedom ?

You can too connect your jeedom to your dev machine (where deconz is installed), jeedom support the "distant deconz installation"

Smanar, 2021-06-29:
Lol, thx for your patience too ^^.
Have made the PR https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5070
The issue will be closed when the PR will be validated, but it will be first in a beta version I think, there is 1 stable by month.

ftitou, 2021-06-29:
Sounds great!

Smanar, 2021-07-01:
Hello, sorry, but we are talking about the device on the PR page.
The working mode is a little "disturbing", in fact on normal switch, during the inclusion, deconz create a "magic group" and use it to make bind  (used for exemple for the autonomous working mode, without the gateway). but your device create his own groups, fixed, and 4, one by buttons.

I think this device don't need bind (but I can 't be wrong), so a fast solution will be to disable totaly the group creation for this device.
To test my idea, you just need to comment this line in binding.cpp (around line 3990)

```
    else if (sensor->modelId() == QLatin1String("Remote switch") ||
         sensor->modelId() == QLatin1String("Double gangs remote switch") ||
	 sensor->modelId() == QLatin1String("Shutters central remote switch") ||
         sensor->modelId() == QLatin1String("Remote toggle switch") ||
         //sensor->modelId() == QLatin1String("Pocket remote") ||
         sensor->modelId() == QLatin1String("Remote motion sensor"))
    {
        //Make group but without uniqueid
    }
```

Then just re-include the device, it will be now without the "config/goup" field and without the error
>10:35:03:038 delete old group 65516 of sensor Pocket remote 2

But not sure it will still working, without binding.

Smanar, 2021-07-03:
@ftitou still on the palce ?

Smanar, 2021-07-05:
Manup have updated the code for the problem we had (not detected bacause empty values for Swbuild and date code)
Have updated the branch if you can re test the code ?

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

But I haven't modified if to test without making group (because the PR is already in checking list)
Will be nice if you can make the modification on the previous post, if you realy have problem say it, I will make it on the PR branch.

ftitou, 2021-07-08:
Hello,
Sorry I am back, as I am not using my email adress I am not notified when new message appears on this thread. Anyway, I am pleased to help. You asked two different things one in your message 1 week ago and another one 3 days ago. Which one you need me to do and what do you need as a result? A log file and/or just to know if it is working or not?

Smanar, 2021-07-08:
Sorry, it s my bad, was sure the PR was good, but after some reflexion have see some possible issue, we realy need to block group feature for this device.

You have recompiled the code, after comment the line ? Because it sem it have not worked
>10:21:02:285 Websocket 192.168.8.1:37290 send message: {"e":"added","id":"2","r":"sensors","sensor":{"config":{"battery":null,"group":"13","on":true,"reachable":true},"ep":1,"etag":"ff33f9691376122c64c6815d213b0034","id":"2","lastseen":"2021-07-08T10:21Z","manufacturername":"Legrand","mode":1,"modelid":"Pocket remote","name":"Pocket remote 2","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"00:04:74:00:00:83:40:f2-01-0005"},"t":"event","uniqueid":"00:04:74:00:00:83:40:f2-01-0005"} (ret = -1097264016)

Still a group in the json, and when you press the button

```
10:22:04:211 delete old group 65516 of sensor Pocket remote 2
10:22:04:212 reanimate group 65515 for sensor Pocket remote 2
```

this device create lot of useless group in the API.

ftitou, 2021-07-08:
> You have recompiled the code, after comment the line ? Because it seem it have not worked

No :-) Could you please tell me what is the procedure after commenting the line?

Smanar, 2021-07-09:
Same
```
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Just instead of updating your file with my branch (with git pull), you use your modified file.

BTW, later if you want to re use git pull, you can delete the folder and restart from scratch or use 
`git reset --hard`

ftitou, 2021-07-10:
OK thanks, then I compiled the code and I got this : 
[aftercompilation.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6794893/aftercompilation.txt)

As the log still contains"delete old group" and "reanimate group", I was wondering if I did the job correctly. So I checked again that I successfully comment the right line, that compilation was OK and that the file copied in /usr/share/deCONZ/plugins has been updated. Seems that I did the job correctly.

BTW since I would say Manup modification, I notice that sometimes the pairing process does not successfully finished, device is recognized but with an name like 0xXXXX instead of "Pocket Remote". Just need to launch pairing process again to make it work. Not a big deal, but I'd rather you to know. In that case I got this : 
[pairingfailed.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6794896/pairingfailed.txt)

Smanar, 2021-07-10:
No it seem you are right, no more group in json

>11:03:52:428 Websocket 192.168.8.1:45610 send message: {"e":"added","id":"3","r":"sensors","sensor":{"config":{"battery":null,"on":true,"reachable":true},"ep":1,"etag":"ff90dcb0f81a72a759cf58452fdecb18","id":"3","lastseen":"2021-07-08T11:03Z","manufacturername":"Legrand","mode":1,"modelid":"Pocket remote","name":"Pocket remote 3","state":{"buttonevent":null,"lastupdated":"none"},"type":"ZHASwitch","uniqueid":"00:04:74:00:00:83:40:f2-01-0005"},"t":"event","uniqueid":"00:04:74:00:00:83:40:f2-01-0005"} (ret = -1097075600)

Have too
>11:04:01:851 skip check bindings for client clusters (no group)

But you are right, after some command the code create group again ...

For the inclusion issue, all is working up the last step, but just at this moment the device make a (re)join, can be a classic issue, Legrand are not easy to include

> 10:48:32:799 [4.2] get basic cluster attr 0x0004 for 0x00047400008340f2
> 10:48:35:380 unknown node 0x00047400008340F2 (0xB78B), lqi: 255
> 10:48:35:380 0x248F nwk changed to 0xB78B
> 10:48:35:383 device announce 0x00047400008340F2 (0xB78B) mac capabilities 0x80

So have checked all tyhe code for code part that can add the "group" in "config" and have added a blacklist somewhere and 1 more debug line at another place.

Have made the change on the branch so command will be

```
cd deconz-rest-plugin
git reset --hard
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If I m right this code will discard your change on the code.

ftitou, 2021-07-11:
So I did this : 
```
cd deconz-rest-plugin
git branch
git reset --hard
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Looks like I am in legrend_scene, then OK. I checked that for example bindings.cpp has well been updated => OK.
[image]

This is what I got : 
[1.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6797269/1.txt)

I notice that it is difficult to get the battery symbol now. I succeed only one time after several pairing.
[image]

Smanar, 2021-07-11:
Nice, ok so have found the guilty

Can you make a last try 

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

For the battery, there is something specific to Legrand, it s a long story, but reporting is not working, so the request is send when you use the device, so you just need to wait, and after some use, the battery will be reported naturally.

ftitou, 2021-07-12:
> For the battery, there is something specific to Legrand, it s a long story, but reporting is not working, so the request is send when you use the device, so you just need to wait, and after some use, the battery will be reported naturally.
> You don't need to have it during the pairing, will come later.

Ok thanks, that's amazing how much you know about Legrand devices!!! Congrats

So this is why I get now, it sounds good I guess :
[1.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6798867/1.txt)

Smanar, 2021-07-12:
>So this is what I get now, it sounds good I guess :

Yes finally ^^. Will clean the PR, but I hope it will be fine now.

>that's amazing how much you know about Legrand devices

Lol, not realy, in fact pipiche help me a lot for Legrand stuff, and I have made the inclusion for almost all in deconz, it s just for that I known well thoses devices.
Configured by deconz, battery swich can sleep looonggg time, without reporting, so I m asking for battery level when the device is awake, when you use it.
It s not the better method, but it work fine for the moment, realy nice for the battery.

And if you restart deconz, it can't have battery level for sensor, because it s sensor themselves that choose when sending this information, so it s same for them, deconz will update the icon when it have the information, can be 1 hour after.

ftitou, 2021-07-17:
Many thanks!

closed 2021-08-02 by manup