dresden-elektronik/deconz-rest-plugin #5028: Connect Aqara wall switches as switch

boehser-enkel, 2021-06-17:
## Feature request type

## Description
Good Morning,

right now you have to connect aqara wall switches as light. This doesn't give the ability to use it as stateless programmable switches (homekit).
Can you add them as switches?

greetz
boehser enkel

## Considered alternatives
Connect the switch as light or as switch

## Additional context

Jagohu, 2025-04-21:
I think this is done, isn't it? Almost all my Aqara wall switches are connected as a switch. I can't say I didn't like them better as a light, but in the end it doesn't make a huge difference and I'm used to it.

ebaauw, 2025-04-21:
Hm, an ancient issue from 2021 which is still open?  Wonder how the bot missed that?  Or maybe the issue was from before the bot?

Anyways, please understand the difference between wired vs wireless switches:
- Wired switches (or relays) are used to smartify dumb devices.  They drive an output wire, connected to dumb device's power.  They behave very much like a smart on/off light.  deCONZ exposed these as a `/lights` resource.  Homebride deCONZ exposes these to HomeKit as _Switch_ or, optionally, as _Lightbulb_ or _Fan_;
- Wireless switches (or buttons) are used to control smart devices.  They provide events on button actions, which are used to trigger automations.  Or they control bound Zigbee devices directly.  deCONZ exposes these as a `ZHASwitch` `/sensors` resource.  Homebridge deCONZ exposes each button to HomeKit as _Stateless Programmable Switch_.

Many devices, including Aqara, provide both wired and wireless switch functionality.  They provide outputs for connecting the dumb device and inputs for connecting a physical switch.  The wireless switch feature typically needs to be configured, depending on the type of physical switch connected: rocker vs push button (momentary).  Some devices also provide a configuration setting to couple or decouple the input to/from the output.  In coupled mode, the physical switch controls the dumb device directly (very 20th century); in decoupled mode the two features work independently.

Both features need to be defined in the DDF, as separate sub devices.  I'm not sure whether that's been done for all Aqara devices.  Support for the inputs of the Aqara dual relay module T2 was added only recently, see https://github.com/dresden-elektronik/deconz-rest-plugin/pull/8163.

manup, 2025-05-17:
Thanks for the description 

Yep looks like the bot doesn't touch issues pre-dating the bot. I'm doing cleanups now for each release to piece by piece close old issues. There are still some Aqara devices running on legacy code. And some with DDFs but still dead C++ code around. I have my boxes with older devices and will keep removing dead code for them, introduce new DDFs if needed.

Closing this issue as the `/lights` (which sometimes should be thought as `/actuators`) and `/sensors` provide the relevant data.

closed 2025-05-17 by manup