dresden-elektronik/deconz-rest-plugin #5030: Pairing Problem Ikea GUNNARP LED Panel with ConBee II and Phoscon

Techtobi83, 2021-06-17:
## Describe the question or issue you are having

Dear ladies and gentlemen, 

after using deConz & Phoscon for years without any problems (using lots of ikea bulbs, including FLOALT Panels) i am having problem to pair the new IKEA Panel called "GUNNARP" (seems to be the same like "Floalt" but can be used in wet rooms). 

The reset procedure of the panel (6x on/of) works fine; panel is blinking after that.

But i am not able to pair it via phoscon.  Have tried it several times, directly (<40cm) near the ConBee II Stick, with several situations (reset during searching-procedure / reset before searching-procedure / ... ) 
Is it possible, that this LED-Panel is still not supported by deConz / Phoscon? 

Thanks for help in advance. 

## Screenshots

## Environment

 - Host system: Raspberry Pi IV 
 - Running method: Raspbian with GUI 
 - deCONZ version:  v2.12.0-beta, Qt 5.7.1, Firmware 0x266b0700
 - Device: ConBee II USB Stick
 - Do you use an USB extension cable: NO
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

## Additional context

Techtobi83, 2021-06-18:
After hours of testing & google i was able to pair the first of two Gunnarp-Panels using the "old" Phoscon Webinterface. 
Under "Settings" --> Open Network for 10 minutes --> wait 

During this time i did the ikea-like-reset of the panel and waited. Some minutes later the new device was shown in the deCONZ GUI. During the pairing the Panel was not (!) near the ConBee II stick. It seems, this does not needed for pairing this type. 

On the normal Phoscon Webinterface there was no way to pair this device.

Smanar, 2021-06-18:
Strange ...
This feature use the API same on phoscon than on the old web app.

Techtobi83, 2021-06-18:
As explained, i have to integrate 2 identical LED-Panels. 

The second panel makes me frustrating! It connects to the network and is shown as a visual node in the deCONZ GUI, but it is not shown in the Phoscon-Webapp and so i am not able to use it. 
I deleted the node several times. It always is found new during the searching procedure..

Smanar, 2021-06-18:
>It connects to the network and is shown as a visual node in the deCONZ GUI

Can you shwo the capture ? if a device is badly included, it can be visible, and it s possible to finish it only using the GUI.

closed 2021-07-17 by github-actions[bot]