dresden-elektronik/deconz-rest-plugin #5032: GCFFlasher download not available for Ubuntu

lienbacher, 2021-06-18:
## Describe the question or issue you are having
I am trying to follow [this ](https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Update-deCONZ-manually#update-in-ubuntu) guide to flash my conbee-2 to the latest firmware and I fail to find gcfflasher for ubuntu. According to the guide I'm supposed to find a download at https://www.dresden-elektronik.de/download/software.html which eventually leads to http://deconz.dresden-elektronik.de/ubuntu/, but nowhere I can find the suggested deb file containing the flash tool. I am not running deconz.

## Screenshots
not applicable

## Environment
ubuntu 20.04.2 on amd64

## deCONZ Logs
not applicable 

## Additional context
not applicable

Smanar, 2021-06-18:
Hello, you have deconz installed ?
Can you just try this command
`GCFFlasher_internal -l`

lienbacher, 2021-06-18:
Yes of course I could install deconz on another machine. But I prefer the way to raise an issue and notify the developer that documentation is outdated and thus wrong. I also would prefer to just get the installer for ubuntu as per the docs instead of installing software that I don't need on a second machine (yes I know I can delete it after updating the stick but that is not the point).

Mimiix, 2021-06-18:
Pinged @manup  :)

manup, 2021-06-19:
Do I miss something? The linked guide / wiki page for Ubuntu refers to `GCFFlasher_internal` which is part of deCONZ.

The standalone GCFFlasher will be resurrected in future again, but there is not ETA yet.

closed 2021-07-19 by github-actions[bot]