dresden-elektronik/deconz-rest-plugin #5034: Immax Magnetic Window and Door Sensor

Thoky81, 2021-06-18:
## Device

- Product name: Magnetic Window and Door Sensor
- Manufacturer: Immax
- Model identifier: DoorWindow-Sensor-ZB3.0
- Device type :
  - Sensor

## Screenshots

### Basic

[image]

### Power Configuration

[image]

### Identify

[image]

### Pool Control

[image]

### IAS Zone

[image]

### Diagnostics

[image]

### OTAU

[image]

### Note

I found this device also here: https://zigbee.blakadder.com/Immax_07045L.html

SwoopX, 2021-06-20:
The attributes must have been read for all clusters, otherwise the screenshots are useless I'm afraid. Can you please update?

Thoky81, 2021-06-20:
I tried to do it but no success. I have also tried to wake up sensor by moving the magnet.

Thoky81, 2021-06-20:
So I have removed the battery and added the sensor again and looks like success. Please find above updated pictures. The sensor is not visible in phoscon app but this time I can see it also in Home Assistant.

SwoopX, 2021-06-20:
You see me positively suprised as the device is not explicitly supported, but benefits from a fallback mechanism 😄 

Should be fully functional though, except for battery reporting.

Thoky81, 2021-06-20:
Looks like the battery is reporting in HA but is strangely low. I will try to change it later just to see if report will change too.

[image]

Thoky81, 2021-06-26:
I tried to change the battery but in fact the sensor is not changing any of the states in HA anymore. Can you include the support for the sensor in next beta pls?

closed 2021-07-02 by manup

Thoky81, 2021-07-31:
I have updated to version beta 2.12.3 where the sensors should have support according to release notes but its not showing up in Phoscon. Any idea pls?

Mimiix, 2021-07-31:
If you see devices within deCONZ but not in Phoscon, that is most of the time perfectly normal. Phoscon is closed source and needs devices to be added manually by their developer. However, the device will still be usable with the API/Home-Assistant/OpenHAB etc. You may want to go ahead and request support for your device at the Phoscon repo

Thoky81, 2021-07-31:
Thanks. 
I found another problem. When I removed the device from deconz I am unable to pair it again and that is strange since it should be already supported in 2.12.3.

Thoky81, 2021-07-31:
.. The device is showing as some unknown item.