dresden-elektronik/deconz-rest-plugin #5036: Anyway to set outdoor temp for Sinope Thermostat ?

matlafqc, 2021-06-19:
## Describe the question or issue you are having
I use Sinope TH1123ZB and I want to set the outdoor temperature value of the thermostat. I don't see any entity in HA to set the value. If I check deconz, I see the value but it's readonly. Can this be supported by deconz and HA integration ? Thanks.

## Screenshots
[image]

## Environment
 - Host system: Synology 
 - Running method: Marthoc Docker container
 - Firmware version: 266b0700
 - deCONZ version: 2.11.5
 - Device: ConBee II
 - Do you use an USB extension cable:  Yes
 - Is there any other USB or serial devices connected to the host system? If so: No

## deCONZ Logs

## Additional context

SwoopX, 2021-06-20:
I assume you're looking at the wrong place. You're trying to change a general attribute on the thermostat cluster which is not changeable. Try the `Sinope specific` cluster.

matlafqc, 2021-06-21:
Thanks ! I see the attribute and it's settable. How this king of attribute is exposed outside deconz, like to HA ?

closed 2021-07-19 by github-actions[bot]

BabaIsYou, 2022-03-22:
This cluster seems not accessible thru rest API.