dresden-elektronik/deconz-rest-plugin #5040: deconz on Debian bullseye amd64

matthiasklein, 2021-06-21:
## Describe the question or issue you are having

I am currently trying to install the ConBee2 stick under Debian bullseye amd64.

Apparently there are no packages for bullseye.
Is there a way to use the ConBee2 stick natively (without Docker) under Debian bullseye anyway?

## Screenshots
## Environment

Debian GNU/Linux 11 (bullseye)
Linux 5.10.0-7-amd64 #1 SMP Debian 5.10.40-1 (2021-05-28) x86_64 GNU/Linux

## deCONZ Logs
## Additional context

SwoopX, 2021-06-21:
https://deconz.dresden-elektronik.de/debian/

matthiasklein, 2021-06-23:
@SwoopX Thank you for the link. But there are only arm64 packages, and no amd64 packages.

But from there I found the ubuntu downloads: https://deconz.dresden-elektronik.de/ubuntu/stable/
It looks like they are working on Debian bullseye too.

SwoopX, 2021-06-23:
Ah, my bad. If you do it "in a rush", you don't catch the subtle differences between the archs 😄 Glad it worked out. Mind closing this one?

matthiasklein, 2021-06-24:
I think using the Ubuntu packages under Debian is not a permanent solution. As a workaround to test deconz OK, but in the long run it rather leads to a FrankenDebian.

So I would like to leave this issue open to encourage building amd64 packages for Debian besides arm64 packages.

closed 2021-07-24 by github-actions[bot]

Red-M, 2021-07-24:
I've run into this too, I should be able to use those packages under debian, but they're all arm64. Can't even use them on a RPi either because they're armhf and not arm64 so the inclusion of the amd64 arch would be appreciated

Siggi0904, 2021-08-22:
Hi, here the same.

Upgraded my system from buster to bullseye and got the same error.
Please release a bullseye target for Debian.

With Ubuntu is this a seperate port und should be released seperately.

As far rpi will release debian bullseye port, it is needed too,

Thanks.

olicooper, 2021-08-30:
+1 for bullseye port on rpi 👍 🙏 .. 

```log
pi@rpi4:~ $ sudo apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Ign:2 http://phoscon.de/apt/deconz bullseye-beta InRelease
Err:3 http://phoscon.de/apt/deconz bullseye-beta Release
  404  Not Found [IP: 144.76.96.194 80]
Hit:4 http://archive.raspberrypi.org/debian bullseye InRelease
Reading package lists... Done
E: The repository 'http://phoscon.de/apt/deconz bullseye-beta Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

pdelrot-samea, 2021-08-31:
Hello. I just have the same needs as the others, and the same result as olicooper.
bullseye support will be appreciated !
BR,
Pascal

trevon88, 2021-10-31:
Still no workaround? ....

Also would like to have a Bullseye edition, please!

Thank you!

shinepixel, 2021-11-12:
The current beta now supports Bullseye:

```
deb http://phoscon.de/apt/deconz bullseye-beta main
```

Thank you *deCONZ* team!

Bias313, 2022-03-17:
Is bullseye still not in stable?

pdelrot-samea, 2022-04-05:
Hello,
WIth (or without trusted) this line I got a "no public signature error".

```
pi@raspberrypi-pascal:~ $ more /etc/apt/sources.list.d/deconz.list 
deb [trusted=yes] http://phoscon.de/apt/deconz bullseye main
```

```
pi@raspberrypi-pascal:~ $ sudo apt update
Atteint :1 http://security.debian.org/debian-security bullseye-security InRelease
Atteint :2 http://deb.debian.org/debian bullseye InRelease                                              
Atteint :3 http://deb.debian.org/debian bullseye-updates InRelease                                      
Réception de :4 http://phoscon.de/apt/deconz bullseye InRelease [2 873 B]                               
Atteint :5 http://archive.raspberrypi.org/debian bullseye InRelease                                     
Ign :4 http://phoscon.de/apt/deconz bullseye InRelease                                                  
2 873 o réceptionnés en 1s (2 068 o/s)
Lecture des listes de paquets... Fait
Construction de l'arbre des dépendances... Fait
Lecture des informations d'état... Fait      
Tous les paquets sont à jour.
W: Erreur de GPG : http://phoscon.de/apt/deconz bullseye InRelease : Les signatures suivantes n'ont pas pu être vérifiées car la clé publique n'est pas disponible : NO_PUBKEY C9A78F8C43D8720A
```

It's a fresh RPi OS 64 bits, on a RPi4. Few weeks ago a RPi OS / Debian buster with deconz and HomeAutomation where installed on it since months.

Do someone has been able to install deconz on bullseye ?

Thanks,
Pascal

night-gold, 2023-05-10:
Just living a comment here to say that this is still not fixed... There is still no packages for debian and the repo configuration working on older version of debian (buster) does not work anymore. Still no amd64 files in the debian repo.

Mimiix, 2023-05-12:
@manup can you leave a response?

manup, 2023-05-12:
The Debian/Bullseye APT repository is not yet referenced, you can install manually by using the Ubuntu package package:

```
wget http://deconz.dresden-elektronik.de/ubuntu/beta/deconz-2.22.00-qt5.deb
sudo dpkg -i deconz-2.22.00-qt5.deb
```

jvarrone, 2023-05-24:
> 

This doesn't work on debian. the .sh files for setup and running still reference the ubuntu o/s.