dresden-elektronik/deconz-rest-plugin #5041: Current Products Corp E-Wand WIndow Covering

RMCob, 2021-06-21:
## Device

- Product name: E-Wand Gen II
- Manufacturer: Current Products Corp
- Model identifier: E-Wand
- Device type : Window Covering
 
## Screenshots
### Node
[image]

### Basic
[image]

### Power Config
[image]

### On/Off
[image]

### Poll Control
[image]

### Window Covering
[image]

#### OTAU
[image]

manup, 2021-07-13:
Related PR: https://github.com/dresden-elektronik/deconz-rest-plugin/pull/5068

RMCob, 2021-08-04:
Keep active

RMCob, 2021-08-20:
Keep active

RMCob, 2021-09-11:
Keep active

RMCob, 2021-10-03:
Keep active

closed 2021-11-02 by github-actions[bot]