dresden-elektronik/deconz-rest-plugin #5046: Yagusmart Zigbee ZW-EU-02, 2 Gang smart switch

talksocke, 2021-06-23:
## Device

- Product name: The device name as shown on the product or package.
- Manufacturer: As per deCONZ GUI Basic cluster.
- Model identifier: As per deCONZ GUI Basic cluster.
- Device type : Please remove all unrelated device types. 
  - Light
  - Lock
  - Remote
  - Sensor
  - Siren
  - Switch
  - Thermostat
  - Other: 

## Screenshots

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Smanar, 2021-06-23:
I think this device can work natively.
But it s not correclty included on the capture, you miss the second circle on the node title.
You can get it using commands available in the first circle. This issue happen too on sonoff device, they need to be included with the coordinator, it s not working using a router and the mesh.

talksocke, 2021-06-23:
> I think this device can work natively.
> But it s not correclty included on the capture, you miss the second circle on the node title.
> You can get it using commands available in the first circle. This [...]

closed 2021-06-23 by talksocke

talksocke, 2021-06-23:
I had already pressed the commands of the first circle several times, but the second one was not displayed. Hours later I looked again and suddenly it was there. :-) Then I searched again via Phoscon for new lights and the two relays were found. 

[image]

talksocke, 2021-06-23:
> I think this device can work natively.
> But it s not correclty included on the capture, you miss the second circle on the node title.
> You can get it using commands available in the first circle. This [...]

Thank you for this hint. :-)