dresden-elektronik/deconz-rest-plugin #5047: CONBEE 2 loses connections with XIAOMI WINDOWS DOOR SENSORS

sajikur, 2021-06-23:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: INTEL NUC
 - Running method: DECONZ/PHOSCON on HOME ASSISTANT
 - Firmware version: 26660700
 - deCONZ version: 26660700
 - Device: ConBee II 
 - Do you use an USB extension cable: YES
 - Is there any other USB or serial devices connected to the host system? If so: UPS

## deCONZ Logs

## Additional context

THANX

Mimiix, 2021-06-23:
Without any information, we can't help with anything.

Either way, please check other issues that hjave been submitted before. The wiki also contains information.

https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

sajikur, 2021-06-23:
Hi. Sorry for strange post but for some reason the description is 
incomplete. In your answer there is my problem. So I am following the 
solution described there. On reboot my window/door sensors going dead but 
every temp/hum sensors stay alive.
I'll update the result.
Thanx

sajikur, 2021-06-24:
Issue updated, thank you

Mimiix, 2021-06-24:
> […](#)

I am not sure what you mean. Can you provide logs and also add a screen of the deconz map?

closed 2021-07-24 by github-actions[bot]