dresden-elektronik/deconz-rest-plugin #5048: Aqara S2 Eigenstone wrong type

oktay89, 2021-06-23:
## Describe the bug

After adding Aqara S2 Eigenstone Thermostat, it gets recognized as model "lumi.airrtc.tcpecn02", which causes OpenHab 3 to add it as thing of type "dimmable light". This way it can't be controller correctly 

## Steps to reproduce the behavior

1. Go to "Devices"/"Sensors"
2. Click "Add new sensor"
3. Go to "Devices"/"Lights" and check the item type
## Expected behavior

The item should have type "Thermostat", so the room temperature can be checked and the devices temperature can be set.

## Screenshots

[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Openhabian
 - Firmware version: (26680700)
 - deCONZ version: (2.09.03)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? If so: Which? - no

## deCONZ Logs

## Additional context

https://de.aliexpress.com/item/4001263827795.html

Knowl3dgable, 2021-06-23:
@oktay89  While you explain to Aqara support that their firmware is cra**y, they need to rewrite it and shar the firmware, can you maybee also ask them to make the reporting configurable on the hum/temp sensors?

Mimiix, 2021-06-24:
This is the name the device gives. What is the restapi showing?

closed 2021-07-24 by github-actions[bot]