dresden-elektronik/deconz-rest-plugin #5050: IKEA FYRTUR: Question regarding direct bindings/group bindings with new FW (2.3.073)

froogl, 2021-06-24:
## Describe the question or issue you are having

I have connected a FYRTUR blind and a TRADFRI open/close remote to my deCONZ network. When pairing the open/close remote a new group is auto-created for direct group bindings. I used to be able to add the FYRTUR blind to that group and control it with the remote, even if deCONZ is down.

However after upgrading the blind firmware to the latest version (2.3.073) this does not work any more. After FW update I have removed the blind, reset it and re-added it to my deCONZ environment. Then I added it to the remote group for direct binding but the blind does not react to any group commands. It can only be controlled directly through deCONZ.

Can I create a direct device binding between the remote and the blind manually? Which clusters would need to be linked? Any other ideas where I can look up what's going on?

## Screenshots

This is the cover next to the remote:

[image]

## Environment

 - Host system: PC
 - Running method: Debian with deb packages
 - Firmware version: (26680700)
 - deCONZ version: (2.10.04)
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system?: no
 - FYRTUR blind FW: 2.3.073
 - TRADFRI open/close remote FW: 2.2.010

## deCONZ Logs

## Additional context

Actually I was only using the direct binding between the remote and the blind because it has served me as a reliable backup in case the blind was having issues. The blind was sometimes not reachable through deCONZ, but the remote could always control the blind and brought it back to the deCONZ network. The blind FW 2.3.073 seems to have fixed that issue and has been very stable for me but it would still be nice if direct binding could still be used to increase the reliability of the system.

ebaauw, 2021-06-24:
Direct control continues to work for me without any issues.  Been running 2.3.073 on my Fyrtur for a long time.

You need to create a binding on the open/close remote, from the client (grey) _Window Covering_ cluster to the group.  Make sure to wake the remote before pressing _Bind_ on the _Bind dropbox_ panel.  Double check that the binding succeed: when you press a button on the remote, `config/group` on the ZHASwitch resource should be updated with the bound group.  You really want to use a group binding (rather than a unicast binding) to prevent any routing issues between the remote (or rather, the remote’s parent router) and the blind.

Next make sure the blind is a member of the group bound to the remote.  Double check in the GUI, using the _Cluster info panel_ on the server (blue) groups cluster.

froogl, 2021-06-25:
Thank you very much for your explanation @ebaauw. Today I was going to check this but all of a sudden the group binding was working! I did not do anything, it just worked. Here's what I did yesterday:

1. Remove and pair FYRTUR again
2. Remove and pair TRADFRI open/close remote again
3. Add the FYRTUR cover to the device group of the TRADFRI open/close remote, which was created automatically by deCONZ.

However this was not working. I also set the "brightness" attribute of that group through the API and the cover did not move.

I let the components sit there for one day. When I checked back today the setup was suddenly working! So my impression is that it just takes a lot of time for the cover to join that ZigBee group. Weird, but great that it's working now.

closed 2021-06-25 by froogl

moritzgloeckl, 2022-06-29:
> So my impression is that it just takes a lot of time for the cover to join that ZigBee group. 

This is most certainly the case! Since 2.3.X its very slow so best advised to pair them and leave it for 20 minutes or so. Worked for me.