dresden-elektronik/deconz-rest-plugin #5051: Osram motion sensor Smart+

RitaSghe, 2021-06-24:
## Describe the question or issue you are having
I connected my new Osram motion sensor Smart+ to Conbee II and does not appear in the list of device, why?

## Screenshots
[image]

[image]

## Environment
Host system: (Raspberry Pi)
Running method: (Raspbian)
Firmware version: (26660700)
deCONZ version: (2.11.05)
Device: (ConBee II)
Do you use an USB extension cable: (yes)
Is there any other USB or serial devices connected to the host system? If so: usb pen

## deCONZ Logs

## Additional context

Mimiix, 2021-06-24:
Seems to be not paired correctly.

RitaSghe, 2021-06-24:
Ok

closed 2021-06-24 by RitaSghe