dresden-elektronik/deconz-rest-plugin #5052: Xiaomi Aqara Temperature sensor: how to get higher temperaure resolution (what is 0xFF0D 'Xiaomi Sensitivity') ?

hbkoenig, 2021-06-24:
I'm using deconz v2.12.1-beta with Xiaomi Aquara lumi.sensor_ht (round, T/H) and lumi.weather (square, T/H/P).

those sensors right now send data whenever temperature changes are > 0.5 K (or other parameters change, or button press).
for one sensor I'd like to get higher resolution for temperature and/or measurement freqency as the changes for this channel are very small. how can this be achieved ?

I noticed the '0xFF0D Xiaomi Sensitivity' rw parameter (uint8) in 'Basic' cluster info.
where can I find docs about that parameter, is it about the minimal temperature steps (or any other writable parameter) ?

if the sensor itself cannot be reconfigured, is there any chance  that i can trigger measurements via deconz (either deconz setup, or using REST API) ?

trying deconz "READ" button in "0402 Temperature measurement" clust info doesn't seem to trigger measurements, 
I always get identical results over minutes (with 0.01 K value resolution).

if Xiaomi Aqara and deconz both cannot provide better resolution, can you suggest any similar sensor (shouldn't be much larger) ?

thanks for any pointers or ideas !!

Harald

## Describe the question or issue you are having

for a single temperature channel (maybe 3 in future) I need higher data rate/resolution...

## Screenshots
here you can see the temperature channel in case (round T/P sensor right now) over 24h:
[image]

## Environment

 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 0x26680700
 - deCONZ version: v2.12.1-beta
 - Device: ConBee II 
 - Do you use an USB extension cable: yes (connected to USB2 port on RPi 4)
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Smanar, 2021-06-24:
Hello, it s the sensitivity for the vibration sensor.

SwoopX, 2021-06-24:
I'm afraid you're not gonna like the answers, but the short version:

> those sensors right now send data whenever temperature changes are > 0.5 K (or other parameters change, or button press).
> for one sensor I'd like to get higher resolution for temperature and/or [...]

Probably nowhere as it has been reverse engineered. As Smanar said, it's for the vibration sensor, so not usable.

> if the sensor itself cannot be reconfigured, is there any chance that i can trigger measurements via deconz (either deconz setup, or using REST API) ?

No chance, the device is responsible for taking the measurement.
 
> trying deconz "READ" button in "0402 Temperature measurement" clust info doesn't seem to trigger measurements,
> I always get identical results over minutes (with 0.01 K value resolution).

The device is a deep sleeper and must be waken up (physical button press) or when the device firmware feels like taking a new measurement, it gets reported automatically.
 
> if Xiaomi Aqara and deconz both cannot provide better resolution, can you suggest any similar sensor (shouldn't be much larger) ?

There's none. If yes, it's probably also from China where you most likely make similar or worse experiences.

closed 2021-07-24 by github-actions[bot]