dresden-elektronik/deconz-rest-plugin #5053: Yagusmart Tuya ZigBee Smart Light Switch with Touch Panel 2-Gang

erkikoiv, 2021-06-24:
## Device

- Product name: Yagusmart Tuya ZigBee Smart Light Switch with Touch Panel 2-Gang
- Manufacturer: _TZ3000_lupfd8zu
- Model identifier: TS0012
- Device type : Switch
  
## Screenshots

[image]

This image shows the two lights that are created after pairing:
[image]

The product is found here: https://www.aliexpress.com/item/1005002116676502.html?spm=a2g0s.9042311.0.0.3da24c4dselHuV

It pairs as a light not as a switch (creates 2 lights as 2-gang switch). It should support three functions: one tap, two taps and long press. These other two are of course not working as it is registered as light. I am actually using it via Domoticz but it is registered there the same way it appears in Phoscon.

Can you help on how to indentify these correctly?

Mimiix, 2021-06-24:
Can you please add deconz gui screenshots and not phoscon?

erkikoiv, 2021-06-24:
Would this be the right one (I am quite new to deconz):
[image]

erkikoiv, 2021-06-25:
And somehow this is not showing up as a repeater.

Mimiix, 2021-06-25:
> And somehow this is not showing up as a router.

I want to see all clusters and cluster info as asked in the template.

erkikoiv, 2021-06-25:
Did I get the right info (Basic)?
[image]

erkikoiv, 2021-06-25:
I have also another Tuya ZigBee 3.0 Smart Light Switch Module Relay from https://www.aliexpress.com/item/1005002096875229.html?spm=a2g0s.9042311.0.0.3da24c4dariOoh that I would need to get right config as well. I don't care that it is defined as well as light, but I would need this to act as a router (to get the Esik good signal). How can it be changed from End device to Router?
[image]
[image]

Smanar, 2021-06-25:
But it s a wired switch ?
Wired switch are light because they receive order.
Battery switch are sensor because they send order.

Your device is wired, so it can turn on the light or turn it off. How it can have double/long press ?

erkikoiv, 2021-06-25:
You are right, it is a wired switch and I think there was some misleading info about 3 function buttons. Sad, it is not a router switch. Then I think it works already OK, nothing else really needed. Thanks!

closed 2021-07-24 by github-actions[bot]