dresden-elektronik/deconz-rest-plugin #5055: ghost automation

jegow, 2021-06-25:
## Describe the bug
I have installed a new wall switch "aquara opple" (triple switch). When I toggle the first switch, all sockets already installed in the house (3 OSRAM) are toggled too. I have not installed any scenes or rules or anything like that. IOBroker is switched off.
How can I switch off this automatic or where does it come from?

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment
Conbee 2
Version 2.11.05
Firmware 9.5.202126680700
## deCONZ Logs

## Additional context

Mimiix, 2021-06-25:
Please check this issue:https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3670#issuecomment-743732232

It's a common problem with this switches. clicking the C solved it.

closed 2021-06-25 by Mimiix

jegow, 2021-06-25:
What does "clicking the C" mean? I use the Phoscon app in the firefox browser.

Mimiix, 2021-06-25:
@jegow On the back of the switch itself.