dresden-elektronik/deconz-rest-plugin #5062: Unable to pair IKEA Symfonisk

Botr16, 2021-06-27:
## Describe the bug
I'm unable to pair an IKEA Symfonisk. I tried to add via Phoscon but it doesn't pick up the switch. The switch doesn't appear in the deCONZ GUI either. I tried pressing the pairing button 4 times, long press for >10 seconds, a combination of both and swapped the battery. During the scan the Symfonisk is sitting a couple of cm from the ConBee .

## Steps to reproduce the behavior
1. In Phoscon click "add new switch"
2. Select "Other"
3. Press the pair button on the Symfonisk 4 times in less than 5 seconds.
4. The LED on the Symfonisk flashes 4 times, turns off then on and it stays on for about 01m15s.
5. The scan ends and Phoscon reads Failed to Connect

## Expected behavior
The switch is connected and joins the network

## Screenshots
NA

## Environment

 - Host system: Raspberry Pi 3 b+
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: 26390500
 - deCONZ version: 2.11.5
 - Device: ConBee I
 - Do you use an USB extension cable: No
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs
The suggested procedure to extract logs doesn't work on home assistant and the debug view doesn't allow copy-paste.
Nothing is logged to ERROR or ERROR_L2 during the scan window.
Smanar on discord suggested looking for a line like "device announce 0x%016llX (0x%04X) mac capabilities 0x%02X\n" 
Checkin INFO, INFO_2, APS, ZDP, ZCL, I couldn't find any line like that.
The following is a screenshot of the debug view, INFO level.
[image]

## Additional context
19 active devices in the network , from Philips, Xiaomi, Aqara, all end-devices.

Mimiix, 2021-06-27:
You need routers in your network.

Botr16, 2021-06-27:
> 
> 
> You need routers in your network.

I connected an osram plug and I was able to pair it after a few attempts. first it came up in deCONZ but not in Phoscon, then Phoscon said it was ready but the switch did not appear and finally it came up in Phoscon and started sending events.
Is there a repeater/router you'd recommend for the sole purpose of routing?

closed 2021-06-27 by Botr16