dresden-elektronik/deconz-rest-plugin #5066: ConBeeII stick disconnects without observeable intervention

tnaumann71, 2021-06-28:
## Describe the bug

After days and weeks runing without problems, all Zigbee devicees connected via ConBeeII disapeared in Homeassistant and Phoscon. Phoscon was disconnected from ConBeeII. I saw this in VNC because the S´startup screen was visible and it shout "disconnected" and clicking on "connect" didn't work.

What I did:
restarting deCONZ-AddOn - no effect
restarting Homeassistant via Server restart - no effect
restarting Host (using Supervisor > System > Host > Restart) - ConBeeII Stick was connected again.

This happend the second time to me and it wasn't funny. Because I wasn't at home over the weekend and my family couldn't use the lights.

I checked the logfiles from the Homeassistant web interface for Server, deCONZ-AddON but I couldn't see any hints (maybe the entries from the time when the stick disconnected already disapeared from the view).
## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: Intel NUC running Proxmox 6.4
 - Running method: Home Assistent deCONZ Add-on / Virtual Machine
 - Firmware version: 266B0700
 - deCONZ version: 2.11.05 / 9.5.2021
 - ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? - yes - WAREMA WMS Stick also with extension cable (but not in use by the system)
 
VM with:
Home Assistant OS 6.1
core-2021.6.6
supervisor-2021.06.6
deCONZ Current version: 6.9.0

Zigbee components:

Busch-Jäger Switches (with battery and with power)
Gledopto LED controller (normal and pro)
IKEA Tradfri Bulb
IKEA Tradfri Remote
Xiaomi Sensors

## deCONZ Logs

Please let me know what you need to find the root cause and find a solution.
Which logfiles do you need?

## Additional context

Mimiix, 2021-06-28:
To get debug logs in a GUI installation, open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

To get debug logs in a headless installation / ssh session, use the following commands:
      sudo systemctl stop deconz-gui
      sudo systemctl stop deconz
      /usr/bin/deCONZ -platform minimal --dbg-aps=1 --dbg-info=2 --dbg-error=2 > debug.txt

When done, exit the process with CTRL+C. You can now start the systemctl service again. Your logs will be in the debug.txt file in the current directory.

Be sure to capture at least a few minutes worth of logs. When sharing these logs, DO NOT attempt to paste them into a channel. Instead use a pastebin service such as this or this.

When done, don't forget to disable logging. This is especially important when running off an SD card, as it will cause unnessecary writes, and might shorten the life of your media.

tnaumann71, 2021-06-28:
I enabled the debug info via VNC.

From where can I get the log file?

Mit freundlichen Grüßen
Torsten Naumann

E-Mail:	***@***.***

Mimiix, 2021-06-30:
In the window it should show.

tnaumann71, 2021-07-02:
We are talking about a problem that happens after days and probable weeks. Is there a logifle with all these logged events?

Mit freundlichen Grüßen

Torsten Naumann

Smanar, 2021-07-02:
>I enabled the debug info via VNC.

You mean you have graphical OS, so can just check logs in deconz ?

>To get debug logs in a GUI installation, open deCONZ and click Help. After that, click Debug View. The following debug levels need to be enabled for proper logging: INFO, ERROR,ERROR_L2,APS,APS_L2.

tnaumann71, 2021-07-24:
This error occurs after days! graphical logging doesn't make sense in this case.

closed 2021-07-24 by tnaumann71