dresden-elektronik/deconz-rest-plugin #5073: MCT-350 SMA

luisgarcia87, 2021-06-30:
## Device

- Product name: MCT-350 SMA
- Manufacturer: Visonic
- Model identifier: MCT-350 SMA
- Device type : Please remove all unrelated device types. 
  - Sensor
  
## Screenshots

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

The device did not enroll by itself and it kept blinking 3 times every 5 seconds indicating that it is in pairing mode. And the IAS Zone was like this:
[](url)

After I added to the IAS_CIE_Address the mac of my zigbee gateway and wrote that info into the device and after that I executed the zone enroll response, the Zone State on the attributes in the IAS Zone changed from not enrolled to enrolled, also then zone status changes from 0x0025 to 0x0024 when I close the sensor, meaning that it is reading the status of it, but I can't see the sensor on the Phoscon App and I am unable to pair it using the Phoscon App. 

Any help is appreciated.

Mimiix, 2021-06-30:
Can you add a screenshot of the node in deconz and the cluster list?

The reason you wont see it in phoscon is because Phoscon needs to add support themselves. However, Home assistant should see it if the device is supported. The name of the device suggest it isn't .

Smanar, 2021-06-30:
It s a door sensor ? (not sure as like said Mimix, it miss the cluster list)
Do you have a linux machine to test code modification ?

closed 2021-07-30 by github-actions[bot]