dresden-elektronik/deconz-rest-plugin #5078: Devices sometimes stop working

DanielPeinhopf, 2021-06-30:
## Describe the question or issue you are having
Does anyone else have the problem that some devices stops working from time to time?
In my case currently it's an IKEA Tradfri Remote.
The device is marked as reachable and also the "Last seen" timestemp is up to date but no button events are received.
Sometimes it helps to restart Phoscon but then all devices become unreachable and i have to manually go to every single device and push a button to get it back online. I know that battery powered devices will need some time to get back online but last time i was waiting a few days and it didnt reconnect.

All other devices are working fine.

I also found some other posts according to this issue but i'am already on a newer firmware.

## Screenshots
[image]

## Environment
 - Host system: Raspberry Pi 3
 - Running method: Raspbian / Marthoc Docker container
 - Firmware version: 26390500
 - deCONZ version: 2.05.86 / 15.10.2020
 - Device: ConBee I
 - Do you use an USB extension cable: yes (before i was not able to pair devices)
 - Is there any other USB or serial devices connected to the host system?
   - Yes, a Zigbee stick, but also on an extension cable a few inches away from the Conbee.

Smanar, 2021-07-01:
>Sometimes it helps to restart Phoscon but then all devices become unreachable and i have to manually go to every single device and push a button to get it back online

All devices ? it mean all not working, or all devices ?
And if you just re-include the not working one, instead of restarting phoscon ?

Strange if the "lastseen" is up to date, I don't thing it s a connexion issue ....

Can you try to get logs during a button press with "info" and "info_l2" ?

DanielPeinhopf, 2021-07-01:
> All devices ? it mean all not working, or all devices ?
No, not allways.. I think it depends on how long Phoscon was offline in the meantime.
I now restarted my server two times and all sensors did reconnect.

> And if you just re-include the not working one, instead of restarting phoscon ?
Yes, i've now re-included it and it's now working again.
It's also strange that i always can only include two devices. Then the inclusion-process stops working and i need to reboot my Raspberry before i can include the next devices.

I tried to get some logs before re-including the non-working remote control but it didn't output anything when pressing a button.

Smanar, 2021-07-02:
>No, not allways.. I think it depends on how long Phoscon was offline in the meantime.

For information Xiaomi sensors leave if their parent as not available during some hours.

>It's also strange that i always can only include two devices. Then the inclusion-process stops working and i need to reboot my Raspberry before i can include the next devices

I have see a similar issue ATM, but after that I have see your deconz version 
>deCONZ version: 2.05.86 / 15.10.2020

I think, it will be usefull to first update your deconz version, lot of changes since your version.

DanielPeinhopf, 2021-07-04:
> For information Xiaomi sensors leave if their parent as not available during some hours.

Ah ok.. Good to know. All of my sensors, except the remote controls, are from Xiaomi or Aqara.

> I think, it will be usefull to first update your deconz version, lot of changes since your version.

Jep, i will update it in the next few days...

closed 2021-08-03 by github-actions[bot]