dresden-elektronik/deconz-rest-plugin #5085: TUYA - 4CH Smart Switch

nukleuz80, 2021-07-03:
Hi Guys,

can you please add a new TUYA device:

[TUYA - 4CH Switch](https://de.aliexpress.com/item/1005002519585205.html?albpd=de1005002519585205&acnt=494-037-6276&aff_platform=aaf&albpg=743612850714&netw=u&albcp=1705854617&sk=UneMJZVf&trgt=743612850714&terminal_id=44a5a02706a848e8acd5a1c051aeb6dd&tmLog=new_Detail&needSmbHouyi=false&albbt=Google_7_shopping&src=google&crea=de1005002519585205&aff_fcid=625ba7d1bace41e4b1b38bc44ecfff73-1625299738171-06429-UneMJZVf&gclid=CjwKCAjwlYCHBhAQEiwA4K21m01BJ30B7vMwV-G71yrJvY1ts84hxOIeiyVzyz1zTb61WFuFMyZPKRoCjAEQAvD_BwE&albag=67310370915&aff_fsk=UneMJZVf&albch=shopping&albagn=888888&isSmbAutoCall=false&aff_trace_key=625ba7d1bace41e4b1b38bc44ecfff73-1625299738171-06429-UneMJZVf&device=c&gclsrc=aw.ds)

## Device

- Product name: TUYA 4Channel Smart Switch
- Manufacturer: _TZ3000_excgg5kb
- Model identifier: TS0004
- Device type : Please remove all unrelated device types. 
  - Switch

## Screenshots

[image]

[image]

[image]

[image]

[image]

Please let me know, if you need more than these screenshots.

Thanks a lot! :)

PS: I can test and compile...

Smanar, 2021-07-03:
Hello, this device don't work natively ?

nukleuz80, 2021-07-03:
OK, sorry - after a second pairing process the device is in Phoscon and in API too.

Is it correct that only the first device shows the manufacturer and product in Phoscon?
[image]

All relais are controllable!

Smanar, 2021-07-03:
"More or less" ^^
Yes, it s normal, there a special code to correct that but you have a new manufacture code.

As you are able to compile code, I can make a modification to try if you want ?

nukleuz80, 2021-07-03:
OK :)

But we can make a bit cosmetic...

Smanar, 2021-07-03:
So you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_4ch https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I don't think you need to re-include the device, but it seem the code work better if the manufacture is empty than if it s "unknown"

nukleuz80, 2021-07-03:
I need a bit time to get it run - some probs with the raspberry...
I will let you know, when i´m done.

nukleuz80, 2021-07-03:
You made it:

[image]

Smanar, 2021-07-03:
Ok, thx, will make the PR

nukleuz80, 2021-07-03:
Do you have an idea why in DeConz are no "**connection lines**" anymore?

[image]

Smanar, 2021-07-03:
Ha yes, wait a little, ATM you have network problem.
So much routers are off line, I hope it s because you have just restarted deconz ?

nukleuz80, 2021-07-03:
The restart is 15 minutes ago…

Smanar, 2021-07-03:
Ok so 2 possiblities:
- you have disabled lines, button "neightbor links" on the top
- you have a device that block your network, can be visible on log.

nukleuz80, 2021-07-03:
Thanks - it was a device that blocked the network.
Deleting and re-including does the trick.

closed 2021-07-05 by manup