dresden-elektronik/deconz-rest-plugin #5088: MOES - smart brightness thermometer

jleeca, 2021-07-04:
Hi Guys,

Can you please add a new MOES device:

[Moes ZigBee Smart Brightness Thermometer Sensor](https://www.aliexpress.com/item/1005002524376481.html?spm=a2g0s.9042311.0.0.28904c4dUdVsiP)

I tried to add the device in Phoscon but it is not detected. Please let me know if more information.

Thanks,

**Device**
- Product name: Moes Smart Brightness Thermometer Sensor
- Manufacturer: _TYZB01_kvwjujy9
- Model identifier: TS0222

**Screenshots**
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Mimiix, 2021-07-04:
If you see devices within deCONZ but not in Phoscon, that is most of the time perfectly normal. Phoscon is closed source and needs devices to be added manually by their developer. However, the device will still be usable with the API/Home-Assistant/OpenHAB etc. You may want to go ahead and request support for your device at the Phoscon repo

Mimiix, 2021-07-04:
If you see devices within deCONZ but not in Phoscon, that is most of the time perfectly normal. Phoscon is closed source and needs devices to be added manually by their developer. However, the device will still be usable with the API/Home-Assistant/OpenHAB etc. You may want to go ahead and request support for your device at the Phoscon repo.

Closing

jleeca, 2021-07-07:
As explained by @Mimiix, I can confirm that the device is available in Home Assistant and usable so I'm closing the case.

closed 2021-07-07 by jleeca