dresden-elektronik/deconz-rest-plugin #5091: Low Battery Issue - Aqara Single Switch Module T1 (with neutral)

spoer78, 2021-07-05:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi)
 - Running method: *** Homebridge Raspbian v1.0.22 - Raspberry Pi 3 Model B Rev 1.2 ***
 - Firmware version: (26660700)
 - deCONZ version: (2.11.05)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

Mimiix, 2021-07-05:
@ebaauw Is this something we need to fix or needs to be done in HBH?

spoer78, 2021-07-05:
Just to add some more information that could clarify the above question if this is a HBH or Phoscon issue. I have a buttload of Aqara Door/Window Sensors and a Multi-Sensor connected to Phoscon and there i can actually see a Battery indicator (which is absolutely correct, because they have a battery). Some of the sensors needed some minutes/hours to update the battery level, but it works absolutely fine and flawlessly. When i check the Single Switch Module in Phoscon, which is listed as a "Light", there are no additional information (which is fine because there are none) ... or ... ok the module supports power consumption and if you check the HBH logs, you can actually see that the consumption is shown and updated. But tbh i do not care that much regarding the power consumption, i just would love to have no battery warning for a non existing battery :)

SwoopX, 2021-07-05:
Based on #4328, the device doesn't have any power configuration or IAS zone cluster. Hence, there is no exposure of any battery.

Can you please provide the REST API output of the device's light and sensor resources? For now, I change this issue to a user question unless proven otherwise.

spoer78, 2021-07-05:
> Can you please provide the REST API output of the device's light and sensor resources? For now, I change this issue to a user question unless proven otherwise.

Is this the required info?

<code>
{
    "etag": "c15f2a3415998d513338f1662055c9dd",
    "hascolor": false,
    "lastannounced": "2021-07-03T16:21:35Z",
    "lastseen": "2021-07-05T07:41Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.switch.n0agl1",
    "name": "Badezimmer Licht",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": "Aug  8 2020",
    "type": "On/Off light",
    "uniqueid": "54:ef:44:10:00:10:97:48-01"
}
</code>

SwoopX, 2021-07-05:
Yes, and the corresponding sensors please 😉

spoer78, 2021-07-05:
I am sorry, but which sensors do you mean? The above mentioned Aqara sensors were only mentioned for intense "boasting" :) ... They arent involved here. Or do i misunderstand something here?

For now its the Aqara Switch module hooked up with Phoscon, passed to Homebridge .. there i have 2 accessories for the device somehow, one "Badezimmer Licht" (which is the light) and one for the (not existing) Battery (see below)

```
Battery Service |  
-- | --
Status Low Battery | 1
Charging State | 2
Battery Level | 0%
Manufacturer | LUMI
Model | lumi.switch.n0agl1
Name | Badezimmer Licht
Serial Number | 54EF441000109748
Firmware Revision | Aug 8 2020
Homebridge Instance IP | 192.168.178.52
Homebridge Instance Port | 51413
Homebridge Instance Username | XX:XX:XX:XX:XX:XX
aid | 14
iid
```

and the one just for the light itself:

```

Lightbulb |  
-- | --
On | 0
Status Fault | 0
Status Active | 1
Manufacturer | LUMI
Model | lumi.switch.n0agl1
Name | Badezimmer Licht
Serial Number | 54EF441000109748
Firmware Revision | Aug 8 2020
Homebridge Instance IP | 192.168.178.52
Homebridge Instance Port | 51413
Homebridge Instance Username | XX:XX:XX:XX:XX:XX
aid | 14
iid
```

SwoopX, 2021-07-05:
Sensors are involved since you state the device has some battery information and that is solely exposed via a sensor. Also, many devices consistw of `/lights` and `/sensors` resources exposed via REST API. Our reposibility ends at deconz' REST API, so whatever any other software makes out of it is not in our hands.

So, if you don't have any sensors for that device, the battery stuff does not originate from deconz and you also don't have any power or consumption measurement (which, to my understanding, the device should have).

spoer78, 2021-07-05:
> Sensors are involved since you state the device has some battery information and that is solely exposed via a sensor. Also, many devices consistw of `/lights` and `/sensors` resources exposed via [...]

Sorry, i am an idiot ... here you go:

```
    "9": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": null
        },
        "ep": 1,
        "etag": "f72c774fa29284e63cedbe32dc7896f7",
        "lastseen": "2021-07-05T08:46Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Consumption 9",
        "state": {
            "consumption": 18,
            "lastupdated": "2021-07-05T08:46:06.764",
            "power": null
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAConsumption",
        "uniqueid": "XX:ef:XX:XX:00:10:97:XX-XX-0702"
    },
    "10": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": null
        },
        "ep": 1,
        "etag": "74d12a9a99cece86de84a3ede742ea53",
        "lastseen": "2021-07-05T08:46Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Power 10",
        "state": {
            "lastupdated": "2021-07-05T08:45:09.415",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAPower",
        "uniqueid": "XX:ef:XX:10:00:10:XX:XX-01-0b04"
    }
```

SwoopX, 2021-07-05:
Thanks. So there seems to be indeed some finetuning required. I'll check it out later.

spoer78, 2021-07-05:
> Thanks. So there seems to be indeed some finetuning required. I'll check it out later.

Thank you, much appreciated. If you need anything else, please let me know.

spoer78, 2021-07-06:
I have seen that you fixed it, amazing. Thank you. So I just have to wait till the merge has been accepted and the fix is in the next release or can I somehow "hotfix" my Phoscon? Probably pull the source + your fixes and compile it from scratch? Or would you say it will be in the next normal release "soonish"? For now, everything works fine, it's just "ugly" or I kinda fear that i get used to a "Battery warning" and will simply ignore other "real batteries" from now on. :)

SwoopX, 2021-07-06:
You don't patch Phoscon, but deconz, Phoscon is just another API client for the REST API plugin 😄 

Well, you could pull my branch containing the changes and compile it yourself. You just need to make sure you got the package `deconz-dev` installed. Instructions on how to compile can be found here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4842#issuecomment-855288853

However, please use the following as the first 2 commands instead

```
git clone --branch=xiaomi https://github.com/swoopx/deconz-rest-plugin.git xiaomi       # Clone repository
cd ./xiaomi
```

spoer78, 2021-07-07:
Thank you!

I just tried to compile it, but it vomited the following error message before my feet:

```
firmware_update.cpp: In member function 'void DeRestPluginPrivate::checkFirmwareDevices()':
firmware_update.cpp:548:73: error: 'instance' is not a member of 'deCONZ::DeviceEnumerator'
     deCONZ::DeviceEnumerator *devEnumerator = deCONZ::DeviceEnumerator::instance();
                                                                         ^~~~~~~~
make[1]: *** [Makefile.Release:965: release/firmware_update.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/xiaomi'
make: *** [Makefile:40: release] Error 2
```

Any ideas?

spoer78, 2021-07-07:
I assume its regarding the deconz-dev beta version, that is not installed...

--- >deconz-dev is already the newest version (2.12.00-raspbian-stretch-beta).

but i do have 2.11.05-raspbian-buster-stable

spoer78, 2021-07-07:
I tried to install the latest beta, did the whole dance like

```
pi@homebridge:~ $ sudo dpkg -i deconz-latest-beta.deb
(Reading database ... 97736 files and directories currently installed.)
Preparing to unpack deconz-latest-beta.deb ...
Unpacking deconz (2.12.01-raspbian-stretch-beta) over (2.12.01-raspbian-stretch-beta) ...
Setting up deconz (2.12.01-raspbian-stretch-beta) ...
Processing triggers for desktop-file-utils (0.23-4) ...
Processing triggers for mime-support (3.62) ...
```

Then i did the sudo apt-get install -f, which said everything fine, but when i checked it was still the old 2.11.05 stable version ... then i rebooted, why the heck not ... still the same ... then i went wild an uninstalled the 2.11 first, did the whole dance again ... the install -f wanted to remove all the deconz-dev dependencies because he said it wasn't installed anymore .. so i installed the old version again.

So everything still works, so i might not have f**ked it up completely, but i am still unable to get the latest deconz-dev beta installed or at least running. :/

spoer78, 2021-07-07:
For whatever reason, Phoscon now says the gateway has the version 2.21.01 / 14 / 06 / 2021,

But i still get the following error when i try to make it:

```
firmware_update.cpp: In member function 'void DeRestPluginPrivate::checkFirmwareDevices()':
firmware_update.cpp:548:73: error: 'instance' is not a member of 'deCONZ::DeviceEnumerator'
     deCONZ::DeviceEnumerator *devEnumerator = deCONZ::DeviceEnumerator::instance();
                                                                         ^~~~~~~~
make[1]: *** [Makefile.Release:965: release/firmware_update.o] Error 1
make[1]: *** Waiting for unfinished jobs....
make[1]: Leaving directory '/home/pi/xiaomi'
make: *** [Makefile:40: release] Error 2
```

spoer78, 2021-07-07:
To be thorough:

```
pi@homebridge:~/xiaomi $ apt list --installed | grep deconz

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

deconz-dev/unknown,now 2.11.05-raspbian-buster-stable armhf [installed]
deconz/now 2.12.01-raspbian-stretch-beta armhf [installed,local]
```

so deconz and deconz-dev are separate and so i still have the old version after the whole dance before?! :(

SwoopX, 2021-07-07:
You have apt battle dpkg...

Your sources for apt are set on stable, tha's why apt doesn't fetch it. And via dpkg, you only install deconz, not deconz-dev (in your particular case) 😉

spoer78, 2021-07-07:
🤦‍♂️ 

i am sorry :(

It's weird, you do it right once, suddenly it works 😁

spoer78, 2021-07-07:
make went fine, then i did the following:

```
sudo systemctl stop deconz
sudo systemctl stop deconz-gui
sudo cp /usr/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ/plugins/bak_libde_rest_plugin.so
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins/
sudo systemctl start deconz
```

which ended in a segmentation fault.

spoer78, 2021-07-07:
when i roll back the libde_rest_plugin.so its starting fine again.

SwoopX, 2021-07-07:
Hm, that is unexpected...

Can you please get me the backtrace? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4619#issuecomment-869139115

spoer78, 2021-07-08:
> Hm, that is unexpected...
> 
> Can you please get me the backtrace? [#4619 (comment)](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4619#issuecomment-869139115)

I am really sorry to bother you with this, but your help is highly appreciated.

As requested:

```
pi@homebridge:~/xiaomi $ gdb --args /usr/bin/deCONZ --platform minimal --dbg-info=2
GNU gdb (Raspbian 8.2.1-2) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "arm-linux-gnueabihf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/deCONZ...(no debugging symbols found)...done.
(gdb) r
Starting program: /usr/bin/deCONZ --platform minimal --dbg-info=2
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
libpng warning: iCCP: known incorrect sRGB profile
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 0x730b1430 (LWP 25767)]
[New Thread 0x724a7430 (LWP 25768)]
[New Thread 0x71868430 (LWP 25769)]

Thread 1 "deCONZ" received signal SIGSEGV, Segmentation fault.
0x70d64ce8 in DeRestPluginPrivate::addLightNode(deCONZ::Node const*) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
(gdb) bt
#0  0x70d64ce8 in DeRestPluginPrivate::addLightNode(deCONZ::Node const*) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#1  0x70d8a8dc in DeRestPluginPrivate::nodeEvent(deCONZ::NodeEvent const&) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#2  0x70eabfe8 in DeRestPluginPrivate::qt_static_metacall(QObject*, QMetaObject::Call, int, void**) () from /usr/share/deCONZ/plugins/libde_rest_plugin.so
#3  0x76035244 in QMetaObject::activate(QObject*, int, int, void**) () from /usr/lib/arm-linux-gnueabihf/libQt5Core.so.5
#4  0x76f1d5e8 in deCONZ::ApsController::nodeEvent(deCONZ::NodeEvent const&) () from /usr/lib/libdeCONZ.so.1
#5  0x0004570c in ?? ()
#6  0x00054834 in ?? ()
#7  0x000400c0 in ?? ()
#8  0x75a4b718 in __libc_start_main (main=0x7e800664, argc=1974984704, argv=0x75a4b718 <__libc_start_main+268>, init=<optimized out>, fini=0xd43e0, rtld_fini=0x76f714c4 <_dl_fini>, stack_end=0x7e800664) at libc-start.c:308
#9  0x00040abc in ?? ()
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb)
```

SwoopX, 2021-07-09:
Hm, that's very odd. Runs fine over here and I haven't touched the function which is referred to in the backtrace. Maybe a clean compile can help?

`make distclean`
`qmake && make -j2`

spoer78, 2021-07-10:
Thanks, yeah i assume the version "f**kery" that i did before messed up the build process... 

So i cleaned it, compiled it again, switched the file, started deconz again and this time it actually booted up nicely. But the battery warning is still there. So with my current track-record in "messing stuff up", I started from scratch, removed your pulled repo and did everything new. The compilation has finished, the md5 sum of the new file is identical to the one i compiled before. But anyway, stopped deconz, exchanged the file, started it up again ... still the same. Everything works nicely, but the battery warning is still there in homebridge. I even rebooted homebridge and the whole OS just to make sure there is nothing cached in the memory. One thing i am afraid of is that i need to re-bind the module from scratch, but that would mean getting to the module which is painfully slammed and buried in a way too small space behind a wall switch. So i really really hope there is a way around that. :)

The rest api output for those sensors are with the new lib are:

```
    "9": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "8f32e84027377959a2bb3e5692966226",
        "lastseen": "2021-07-10T06:31Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Consumption 9",
        "state": {
            "consumption": 33,
            "lastupdated": "2021-07-10T06:31:25.324",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAConsumption",
        "uniqueid": "xx:ef:xx:10:00:xx:xx:48-01-xxxx"
    },
    "10": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "bee5bfa4a45763fb2d31cc5177d49300",
        "lastseen": "2021-07-10T06:31Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Power 10",
        "state": {
            "lastupdated": "2021-07-10T06:31:03.381",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAPower",
        "uniqueid": "xx:ef:xx:10:00:10:xx:xx-01-0b04"
    }
```

and the light device itself:

```
{
    "etag": "2c6b57ba6294c9a4a5498601fbb0cb20",
    "hascolor": false,
    "lastannounced": "2021-07-03T16:21:35Z",
    "lastseen": "2021-07-10T07:00Z",
    "manufacturername": "LUMI",
    "modelid": "lumi.switch.n0agl1",
    "name": "Badezimmer Licht",
    "state": {
        "alert": "none",
        "on": false,
        "reachable": true
    },
    "swversion": "Aug  8 2020",
    "type": "On/Off light",
    "uniqueid": "xx:ef:xx:10:00:10:xx:xx-01"
}
```

SwoopX, 2021-07-10:
I guess you could delete the light in Phoscon, which resets the device, so you wouldn't need to physically fiddle around with it.

spoer78, 2021-07-11:
> I guess you could delete the light in Phoscon, which resets the device, so you wouldn't need to physically fiddle around with it.

Are you sure? If i delete it, it will get back into pairing mode? Because getting to the switch to push the button would really, really be a pain in the butt. :) Apropos but(t), i am getting a few more of these modules next week, one with "neutral" and one without. When i hook them up and pair them, i will see if those guys (or at least the one with neutral, in case dconz sees them as different devices types and the fix doesn't work with it yet) show a battery warning or not. If not, then i will try to simply delete and re-pair it. If that doesn't work, i will probably hate myself for a few days, then rip the switch apart again and pair it manually again. :)

SwoopX, 2021-07-11:
Then how about deleting the sensors via API?

https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#delete-sensor

spoer78, 2021-07-11:
> Then how about deleting the sensors via API?
> 
> https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/#delete-sensor

Will it recreate/auto-detect the sensors afterwards or are they "gone" then (till i re-pair them)?

SwoopX, 2021-07-11:
Yes, that should be the case.

spoer78, 2021-07-11:
> Yes, that should be the case.

I have deleted sensor 10, which removed 9 as well. When I checked for lights, the light was gone in Phoscon. I started the "Lights detection" again, it looked around for 3 minutes but did not find anything. But i can still toggle the light with the switch itself (like common folks, but hey). :D I tried pairing it again after i have toggled the switch, too, but did not work either.

So no, it did not auto detect the light afterwards (at least not for me) and I do not hate myself enough for today to dig out the module out of the wall. So i will wait till i get the other modules and will do it then.

spoer78, 2021-07-12:
Opened up the wall again and tried to re-pair the switch. It does all the usual stuff, but deconz does not detect the switch. I even moved the RPI right next to the switch, just in case something bothers him now, but the search doesn't find anything anymore. This probably has nothing to do with your bugfix but with me upgrading deconz completely before. Anyway, i have some config backups and i will try to do a restore and next time i should just ignore the battery warning and be happy that everything is running smoothly ... and that i should have a WAY BETTER backup-concept. :)

spoer78, 2021-07-12:
I restored the backups, it even listed the (other) devices again, but no connection to them. I reset the usb-stick and was able to add all the sensors again, which was "fine".... but when i tried to add the Single Switch module, it kindly suggested to me to go f**k myself. I am really, really unsure what has happened here, but for now i am unable to connect to the single switch module. I will setup everything on a separate sd-card from scratch and kinda hope that this will help somehow.

spoer78, 2021-07-12:
Rolled back to 2.11.05 and was able to add all devices including the single switch module again.

spoer78, 2021-07-12:
On an additional note, i got the "no neutral" version in the mail today, because some of my rooms have the "old wiring". I hooked it up, it works fine and has no issues regarding a false battery warning. I am running 2.11.05 / 09/05/2021 at the moment. So it seems that only the module with "neutral" has the battery issue at all.

spoer78, 2021-07-13:
FYI lumi.switch.l0agl1 works great and is the version of the switch without neutral, lumi.switch.n0agl1 is the version with neutral, which still has the battery-bug. I tried to test your fix, but as you can see above, it didn't work out that well.

I have moved the homebridge and deconz/phoscon from a raspi to a ubuntu vm running on my Synology, which works pretty well. I restored my backups and just had to restore the zigbee network configuration via the "hidden" menu. Running quite good so far. I try to ignore the battery warning for the switch and hope your fix will be included in one of the next releases. Then i will give it a try again...

spoer78, 2021-07-16:
Added another switch module with neutral (got them all now, hooray) and it also has the battery warning. How long do you think that it will take for your fix to be in the next release?

SwoopX, 2021-07-16:
> FYI lumi.switch.l0agl1 works great 

Don't you worry, it would also have battery if it would be allowed or needs to create any sensor resources 😃 

Not sure what went wrong in your case. Based on the documentation just the resource should have to be deleted without resetting the device. No clue if that might have silently changed over time to also make a reset by default...

I had hope that the changes would have already been included in the current version, but obviously not. Next beta is certain, my gut feeling says around 2 weeks.

spoer78, 2021-07-17:
> I had hope that the changes would have already been included in the current version, but obviously not. Next beta is certain, my gut feeling says around 2 weeks.

But when the next beta is out, i will still need to delete and re-add those wall modules probably again? Now i have 2 with neutral, so it would be double the pain to get to them. :) Isn't there some way to patch the database? Like deleting the battery status for those two guys or setting it to 100% or something fishy like that?

spoer78, 2021-07-17:
I know i know it looks like i hate myself but i want to give it a go again ... 

After a snapshot of my VM, i have switched to the beta channel and now i am running version 2.12.03 / 12/07/2021... So hopefully it prevents any version related problems ... 

So for the ones that a little bit slow here ... me for example ... i need to checkout your repo / change again, install deconz-dev for the latest beta, compile the rest api plugin again, switch it against the currently running one, and then i have to delete one of the sensor resources and pray?

Last time i deleted one, everything was gone and never came back...  was it the wrong one?

For example i have the following sensors:

```
    },
    "17": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "adsadsadq2saadad",
        "lastseen": "2021-07-17T07:53Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Consumption 17",
        "state": {
            "consumption": 6,
            "lastupdated": "2021-07-17T07:50:51.993",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAConsumption",
        "uniqueid": "XX:XX:XX:XX:XX:XX:XX:XX-XX-XXXX"
    },
    "18": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "adsadsadq2saadad",
        "lastseen": "2021-07-17T07:53Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Power 18",
        "state": {
            "lastupdated": "2021-07-17T07:52:34.752",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAPower",
        "uniqueid": "XX:XX:XX:XX:XX:XX:XX:XX-XX-XXXX"
    }
```

Which of those need to be deleted? The ZHAPower or the ZHAConsumption? I would kinda guess the ZHAPower or? I think the last time i have delete the ZHAConsumption and everything went downhill from there ... everything was gone then, the lights and the other sensor as weil.

SwoopX, 2021-07-29:
I now added a hopefully forced removal of the battery resource item. No need to reset or re-pair anything.

spoer78, 2021-07-29:
Thank you very much. I have just tried it and due to a completely new box (its now a Ubuntu VM) i have done the following:

```
git clone --branch=xiaomi https://github.com/swoopx/deconz-rest-plugin.git xiaomi
sudo apt-get install deconz-dev
cd xiaomi/
qmake && make -j2
sudo systemctl stop deconz
sudo systemctl stop deconz-gui # just for giggles, it did not run anyway
sudo cp /usr/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ/plugins/bak_libde_rest_plugin.so
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins/                      
sudo systemctl start deconz
```
Deconz is starting fine, Phoscon shows all the devices (funny bug btw, when you click on the lamps on the left for the rooms, only the Aqara modules with Neutral are switched on, without neutral nothing happens ... but when i click on the lights itself, everything works fine ... but i digress. :)

Back to the issue at hand. I restarted Homebridge as well just to be sure, but there are still the "Battery 0" indicators.

Have i done something wrong?

SwoopX, 2021-07-29:
Can you double check the REST API, just to be sure?

spoer78, 2021-07-30:
Sure, here you go:

```
Lights:

    "2": {
        "etag": "---",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2021-07-30T07:23Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Badezimmer Licht",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": "Aug  8 2020",
        "type": "On/Off light",
        "uniqueid": "54:ef:44:10:00:10:97:48-01"
    },
    "3": {
        "etag": "---",
        "hascolor": false,
        "lastannounced": null,
        "lastseen": "2021-07-30T07:20Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.l0agl1",
        "name": "Hauptraum Licht",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": "0x00000018",
        "type": "On/Off light",
        "uniqueid": "54:ef:44:10:00:0e:04:95-01"
    },
    "4": {
        "etag": "---",
        "hascolor": false,
        "lastannounced": "2021-07-16T14:08:55Z",
        "lastseen": "2021-07-30T07:22Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Küche Licht",
        "state": {
            "alert": "none",
            "on": false,
            "reachable": true
        },
        "swversion": "Aug  8 2020",
        "type": "On/Off light",
        "uniqueid": "---"
    }

Sensors:

    "9": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "---",
        "lastseen": "2021-07-30T07:21Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Consumption 9",
        "state": {
            "consumption": 97,
            "lastupdated": "2021-07-30T07:20:36.375",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAConsumption",
        "uniqueid": "---"
    },
    "10": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "---",
        "lastseen": "2021-07-30T07:21Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Power 10",
        "state": {
            "lastupdated": "2021-07-30T07:21:37.293",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAPower",
        "uniqueid": "---"
    },
    "17": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "---",
        "lastseen": "2021-07-30T07:24Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Consumption 17",
        "state": {
            "consumption": 216,
            "lastupdated": "2021-07-30T07:19:58.519",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAConsumption",
        "uniqueid": "---"
    },
    "18": {
        "config": {
            "battery": null,
            "on": true,
            "reachable": true,
            "temperature": 0
        },
        "ep": 1,
        "etag": "---",
        "lastseen": "2021-07-30T07:24Z",
        "manufacturername": "LUMI",
        "modelid": "lumi.switch.n0agl1",
        "name": "Power 18",
        "state": {
            "lastupdated": "2021-07-30T07:21:04.266",
            "power": 0
        },
        "swversion": "Aug 8 2020",
        "type": "ZHAPower",
        "uniqueid": "---"
    }
}
```
It looks like the sensors for the switches with neutral are still there, but there are none for the "no neutral". The "no neutral" is the one that has no battery warning.

SwoopX, 2021-07-30:
This is getting really annoying... Can you give me a screenshot of the available clusters from such a switch?

The last change should have killed the resource item upon sensor load from the DB. However, the only reason I can think of for it still being there is that the power configuration cluster sneaked in somehow...

spoer78, 2021-07-30:
Like i said, i am terrible sorry to bother/annoy you with this. :( Really ...

I guess the screenshots of the clusters can only be made via the deconz-gui .. so atm i am installing x11, vnc-server and stuff to get the gui up and running and will send you the screenshots of the cluster asap. Its an Ubuntu VM on a Synology ... so this probably will take a while. :D But i am on it

spoer78, 2021-07-30:
Somehow there is a qt5 issue with Ubuntu, as far as i could google... i got vnc up and running, but when i want to start the deconz-gui, i only get this:

```
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, xcb.

/usr/bin/deCONZ-autostart.sh: line 298: 4151482 Aborted                 (core dumped) $APPNAME $OPTIONS $EXTRA_OPTIONS
deCONZ - exit code 134
exit code 134 not handled
```

I kinda feel like i have wasted too much of your time already and only for a cosmetics issue. Sure i would love the warnings gone, just so i am not missing "real" warnings in the future by being numb regarding those alerts, but i am sure you have way better things to do and my setup seems to be far from the usual rpi stuff.

spoer78, 2021-08-03:
So after you have spend so much time fixing this, i kinda owed you to finally test it again .. so i made a snapshot and a phoscon gw backup ... then i switched your lib again and deleted the ZHApower .. the device disappeared again. So i restored the phoscon backup .. the device was there again, but greyed out and not working. So i deleted the device, ripped the wall switch aparat and tried to pair it again .. not found ... tried several times ... no go. Then i switched back the patched lib to the original and i could add the device without a problem.

So it might be a good idea to pull back the merge request, there seems to be a problem there. With the patched lib, you cannot add the device anymore.

closed 2021-08-10 by manup

spoer78, 2021-08-11:
Maybe it would be wise not to close this issue or at least pull back the merge request due to the problem that it does not fix the problem and prevents adding devices of this type. Sure i would love this fixed someday, but i would prefer the faulty battery-display above not using the device at all. :)