dresden-elektronik/deconz-rest-plugin #5094: Switching of Aqara Plug / Single Switch with iobroker

Beckmec, 2021-07-05:
Hello, I have a question/ problem,
Unfortunately, I am not an expert and a complete newcomer, so I apologize if I should forget one or the other information

## Describe the bug
I have problems the following two Aqara switch-types:

- Aqara Single Switch Module T1 (with neutral) [SSM-U01]
- Aqara Smart Plug [SP-EUC01]

If I add this via the Conbee 2 stick, at the start I can easily switch it on or off via the iobroker using a simple blockly script (object: "On" with the values "true" or "false").

If I restart the raspberry pi or the iobroker, the script no longer works. I see that the value of the object changes if i start the skript (the skript just controlles the value of the object), but physically the switch does NOT switch.

The whole time I can easily turn the switch on or off via the Conbee 2 stick directly (i.e. the Phoscon app) or via the physical switch on the device.

I have already removed and re-added the switches several times, always with the same result.
Both switches are also compatible with the Conbee 2 or deconz.

Has anyone ever had such a problem or an idea what else I could do?

Many Thanks

PS: My other aqara door / window sensors, temperature sensors and motion detectors work without any problems.

## Steps to reproduce the behavior

After a restart (raspberry Pi / iobroker) I can no longer switch the following two switch types via iobroker:
- Aqara Single Switch Module T1 (with neutral) [SSM-U01]
- Aqara Smart Plug [SP-EUC01]

If I restart the raspberry pi or the iobroker, the script no longer works. I see that the value of the object changes if i start the skript (the skript just controlles the value of the object), but physically the switch does NOT switch.

The whole time I can easily turn the switch on or off via the Conbee 2 stick directly (i.e. the Phoscon app) or via the physical switch on the device.

## Expected behavior
Actually I would expect that a correctly added switch can be switched "permanently" by one and the same sibple blockly script.

## Screenshots
State if Plug is ON:
[image]

State if Plug is OFF:
[image]

In the current error situation I see that exactly this value changes as desired by starting the script (true / false), but physically nothing happens at the switch.

If I press the switch on the physical switch on the device or via the Conbee 2 page (Phoscon app), then I can turn the switch on and off normally and then exactly the object shown changes.

## Environment
- iobroker (version 4.2.1)
- deconz Zigbee Adapter (version 1.3.16)
- script engine (version 5.1.3)

 - Host system: (Raspberry Pi 4B 8GB)
 - Running method: (Raspbian with iobroker)
 - Firmware version: (26660700)
 - deCONZ version: (2.11.05 / 9.5.2021)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes) -- only relevant for ConBee II
 - Is there any other USB or serial devices connected to the host system? NO

## deCONZ Logs

## Additional context

SwoopX, 2021-07-05:
> The whole time I can easily turn the switch on or off via the Conbee 2 stick directly (i.e. the Phoscon app) or via the physical switch on the device.

So when iobroker fails, it still works in Phoscon?

Beckmec, 2021-07-06:
Yes exactly. I can still switch the switches via Phoscon without any problems.

SwoopX, 2021-07-20:
Then it is an issue with iobroker. Phoscon and iobroker make use of the same mechanism (both are REST API clients) and since Phoscon is still working, there's no other option. Please reach out to their team.

closed 2021-07-20 by SwoopX