dresden-elektronik/deconz-rest-plugin #5096: Sensors are not displayed correctly in Phoscon (Home Assistant)

Treyfane, 2021-07-05:
## Describe the bug
Hey there,

I've got some 7 pcs. SILVERCREST / LIDL Silvercrest motion detectors, all bought from the same store out of the same batch of which I connected 4 pcs. some months ago using the Phoscon App, they're working without a problem.

Now I wanted to connect the remaining 3 and here's the problem:

Upon connecting the motion detectors via the Phoscon App (Sensors / Add Sensor / ...) the sensors are found, but will not be displayed in the Phoscon App. 

[image]

However, the sensors are correctly displayed in the deCONZ App (Presence 11, Presence 12 and Presence 13). 

[image]

They are also available as Devices / Entities in Home Assistant.

From what I can see when checking the network requests/response in the Phoscon App, it seems as if the API request "sensors?_=<number>" actually does return a JSON with all the sensors in it, the page is just not displaying it (correctly):

`{

    "1":{
        "config":{
            "configured":true,
            "on":true,
            "sunriseoffset":30,
            "sunsetoffset":-30
        },
        "etag":"<long etag here>",
        "manufacturername":"Philips",
        "modelid":"PHDL00",
        "name":"Daylight",
        "state":{
            "dark":false,
            "daylight":true,
            "lastupdated":"2021-07-05T17:38:01.359",
            "status":170,
            "sunrise":"2021-07-05T03:35:23",
            "sunset":"2021-07-05T19:33:02"
        },
        "swversion":"1.0",
        "type":"Daylight",
        "uniqueid":"<long uniqueid here>",
    },
    "11":{
        "config":{
            "battery":100,
            "duration":60,
            "enrolled":1,
            "on":true,
            "pending":[
            ],
            "reachable":true
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":"2021-07-05T17:57Z",
        "manufacturername":"LIDL Silvercrest",
        "modelid":"HG06335",
        "name":"Presence 11",
        "state":{
            "lastupdated":"2021-07-05T17:57:43.452",
            "lowbattery":false,
            "presence":false,
            "tampered":false
        },
        "type":"ZHAPresence",
        "uniqueid":"<long uniqueid here>",
    },
    "12":{
        "config":{
            "battery":100,
            "duration":60,
            "enrolled":1,
            "on":true,
            "pending":[
            ],
            "reachable":true
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":"2021-07-05T17:57Z",
        "manufacturername":"LIDL Silvercrest",
        "modelid":"HG06335",
        "name":"Presence 12",
        "state":{
            "lastupdated":"2021-07-05T17:57:51.264",
            "lowbattery":false,
            "presence":false,
            "tampered":false
        },
        "type":"ZHAPresence",
        "uniqueid":"<long uniqueid here>",
    },
    "13":{
        "config":{
            "battery":100,
            "duration":60,
            "enrolled":1,
            "on":true,
            "pending":[
            ],
            "reachable":true
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":"2021-07-05T17:57Z",
        "manufacturername":"LIDL Silvercrest",
        "modelid":"HG06335",
        "name":"Presence 13",
        "state":{
            "lastupdated":"2021-07-05T17:57:48.828",
            "lowbattery":false,
            "presence":false,
            "tampered":false
        },
        "type":"ZHAPresence",
        "uniqueid":"<long uniqueid here>",
    },
    "2":{
        "config":{
            "battery":100,
            "duration":60,
            "enrolled":1,
            "on":true,
            "pending":[
            ],
            "reachable":true
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":"2021-07-05T17:57Z",
        "manufacturername":"SILVERCREST",
        "modelid":"TY0202",
        "name":"BM EG Flur 1",
        "state":{
            "lastupdated":"2021-06-27T15:47:12.211",
            "lowbattery":false,
            "presence":false,
            "tampered":false
        },
        "swversion":"2019.12.12",
        "type":"ZHAPresence",
        "uniqueid":"<long uniqueid here>",
    },
    "3":{
        "config":{
            "battery":100,
            "duration":60,
            "enrolled":3,
            "on":true,
            "pending":[
            ],
            "reachable":true
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":"2021-07-05T17:39Z",
        "manufacturername":"SILVERCREST",
        "modelid":"TY0202",
        "name":"BM EG Flur 2",
        "state":{
            "lastupdated":"2021-07-05T17:25:02.371",
            "lowbattery":false,
            "presence":false,
            "tampered":false
        },
        "swversion":"2019.12.12",
        "type":"ZHAPresence",
        "uniqueid":"<long uniqueid here>",
    },
    "4":{
        "config":{
            "battery":100,
            "duration":60,
            "enrolled":3,
            "on":true,
            "pending":[
            ],
            "reachable":true
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":"2021-07-05T17:39Z",
        "manufacturername":"SILVERCREST",
        "modelid":"TY0202",
        "name":"BM EG Bad 1",
        "state":{
            "lastupdated":"2021-07-05T14:35:42.016",
            "lowbattery":false,
            "presence":false,
            "tampered":false
        },
        "swversion":"2019.12.12",
        "type":"ZHAPresence",
        "uniqueid":"<long uniqueid here>",
    },
    "6":{
        "config":{
            "heatsetpoint":1200,
            "locked":true,
            "mode":"auto",
            "offset":0,
            "on":true,
            "preset":"auto",
            "reachable":false,
            "schedule":{
            },
            "schedule_on":false,
            "setvalve":false,
            "windowopen_set":false
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":null,
        "manufacturername":"_TZE200_c88teujp",
        "modelid":"TS0601",
        "name":"Thermostat 6",
        "state":{
            "lastupdated":"none",
            "lowbattery":null,
            "on":null,
            "temperature":null,
            "valve":null
        },
        "type":"ZHAThermostat",
        "uniqueid":"<long uniqueid here>",
    },
    "7":{
        "config":{
            "battery":100,
            "duration":60,
            "enrolled":0,
            "on":true,
            "pending":[
            ],
            "reachable":true
        },
        "ep":1,
        "etag":"<long etag here>",
        "lastseen":"2021-07-05T17:38Z",
        "manufacturername":"SILVERCREST",
        "modelid":"TY0202",
        "name":"BM EG Küche 1",
        "state":{
            "lastupdated":"2021-07-05T17:24:44.898",
            "lowbattery":false,
            "presence":false,
            "tampered":false
        },
        "swversion":"2019.12.12",
        "type":"ZHAPresence",
        "uniqueid":"<long uniqueid here>",
    }

}`

I also see, that the manufacturer and modelid is different in between the sensors aded via Phoscon some month ago and the ones added recently (today). The sensors are in fact all the same model (HG06335 according to what#s written on them).

Maybe someone has a bright idea how to address this problem? 😸

Greetings,

          Dingo

## Steps to reproduce the behavior
see above

## Expected behavior
see above

## Screenshots
see above

## Environment
I'm currently using the following setup:

Home Assistant OS 6.1 on a Raspi 4
Supervisor: supervisor-2021.06.6
Core: core-2021.6.6
deCONZ: 6.9.0
ConBee II

[image]

## deCONZ Logs

## Additional context

Smanar, 2021-07-05:
Hello, phoscon is just an application, like HA.
But phoscon don't support all devices (less than HA), you realy want thoses device in phoscon ?
If I m right you can see them on HA ?

Treyfane, 2021-07-05:
Yes, I can see all them on HA. The problem now is, that only Phoscon is able to rename those devices to my likes (i.e. "Presence 4" → "BM EG Bad 1"). 

I can set new names for the devices in HA internally, but whenever I need to "reset" the HA Integration (by removing / reinstalling it), the devices are newly provided with their "Presence xx" names, which killa Automations, Scripts, ...

So yes, I'd really like those devices available in Phoscon 😸

Mimiix, 2021-07-05:
You need to ask for support in the phoscon repo.

Smanar, 2021-07-05:
But the HA plugin don't have an option to rename deconz device direclty in the API (because you are not the first one to ask for that) ?

Treyfane, 2021-07-05:
In HA (Configuration → Devices / Entities) I can give a device (or any entity within) a new name. But that change seems to be only a "cosmetical" change, since it seems not to be written back into deCONZ. After "reinstalling" the deCONZ Integration, all names are back to square one.

In deCONZ I have a field with the name in it, editable / writeable, which prompts a "sending user descriptor set request", but also does not seem to do anything. The names of the devices simply do not change.

The only part of software in this setup, which seems to be able to actually change a device's name is Phoscon. Which does not display all of the devices (whyever).

> You need to ask for support in the phoscon repo.

This is not helpful.

Smanar, 2021-07-05:
Yep, you can too using yourself the API https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/ , but I think you can ask for this feature for HA, I m using it on domoticz, usefull one.

And no, using "node information" on deconz is not the solution, better to use direclty the API, like do phoscon.

Mimiix, 2021-07-05:
This attitude isn't helpful either...

Treyfane, 2021-07-05:
> Yep, you can too using yourself the API https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/ , but I think you can ask for this feature for HA, I m using it on domoticz, usefull one.

That's a useful Idea, at least for me. Personally I can live with that for now. Thank you very much.

> And no, using "node information" on deconz is not the solution, better to use direclty the API, like do phoscon.
> But like said Mimix, we haven't impact on devices displayed or not in phoscon.

Given I'm a developer myself (or used to be), I can "work-around" the problem this way. And thanks to @Smanar 's explanation I now understand a bit better, how the whole thing works internally.

But I might sincerely suggest, that someone in charge here keeps the average user / customer in mind. Phoscon == Dresden Elektronik and from the Phoscon APP page everything in regards  to changelog / bugs links right here to **this Git**. For an outsider (or so to speak: regular buyer of the ConBee II) this Git **is** Phoscon / deCONZ. No "normal" person knows the difference in what you develop.

Again, from the outside view: Your product is broken, I bought it, please fix it. Easy as that. That includes Phoscon / deCONZ / the sitck itself and any API it uses in a setup advertised on the manufacturer's homepage.

If Dresden Elektronik sends their buyers / users this way to obtain information, then tell them either not to do that, put a big warning sign up front or just be helpful (like @Smanar). You know the stuff here, your visitors don't.

@Mimiix - on a personal note (I very seldomly do this): You are the only one in this conversation with an (very bad) attitude. You throw in a comment which is not helping at all (at least I didn't find a repo for the phoscon app), right after withdrawing the "Bug report" label without any (useful) explanation. You are acting like the almighty god of developers here, giving your customer a bad time. That's neither professional nor in any way the spirit of open source collaboration, heck, that's not even basic courtesy. You're not thinking for your customer (which I am), you're not acting solution-oriented, you're posing like a desk clerk: "Not my problem, go away". If that's how you usually do your job, please get yourself another one and preferrably one without any public contact.

SwoopX, 2021-07-05:
This is indeed a topic for Phoscon here, deconz is working fine. However, the explanation is as follows:

- The devices you see in Phoscon have the modelID `"modelid":"TY0202"` and `"manufacturername":"SILVERCREST"`
- Those you DON't see `"modelid":"HG06335"` and `"manufacturername":"LIDL Silvercrest"`
- Note the differences in both, modelid and manufacturername

Between your periods of adding the sensors, we've done some changes to give the LIDL devices a more readable naming, which was only loosely enfored at the beginning. So, if you'd now re-pair your visible sensors, they'd vanish as well and you might want to avoid doing that 😉 

So from visibility perspective, Phoscon needs to catch up a bit here (which is not in our hands). Regarding the sensor name, you can rename the devices anytime via deconz' REST API.

Treyfane, 2021-07-05:
Given the information I already got, I suspected something like that, thank you for the clarification. The suggestion using the API to rename the devices helped me a lot (also in regards to the thermostat I'm using there), already tried it and it worked out.

As for HA the renaming of Devices / Entities never seems to be written back to the original source (no matter what Integration you use), so I guess they'll not implement a functionality like that.

So I just hope, Phoscon at some point will catch up.

Mimiix, 2021-07-06:
> @Mimiix - on a personal note (I very seldomly do this): You are the only one in this conversation with an (very bad) attitude. You throw in a comment which is not helping at all (at least I didn't find a repo for the phoscon app), right after withdrawing the "Bug report" label without any (useful) explanation. You are acting like the almighty god of developers here, giving your customer a bad time. That's neither professional nor in any way the spirit of open source collaboration, heck, that's not even basic courtesy. You're not thinking for your customer (which I am), you're not acting solution-oriented, you're posing like a desk clerk: "Not my problem, go away". If that's your understanding of your job (here), please get yourself another one and preferrably one without any public contact.

Sorry if i made you feel this way, as it wasnt meant like that in any way and  I just keep things tidy and strict.  My communication just can be a bit harsh. I appologize. I am not a developer nor an employee of Dresden Elektronik.  I'm the community manager :) As far as i am concerned: You are right in the communication part and that's why there's a forum comming up in the following days to provide a more open streamlined communication platform for end users. 

once again: My applogies if i communicated wrongly.

manup, 2021-07-06:
I think we can move this issue to https://github.com/dresden-elektronik/phoscon-app-beta 
I've seen a few changes for Tuya/Lidl devices in the code, might already be fixed in the pwabeta. We'll have a look.