dresden-elektronik/deconz-rest-plugin #5097: Device name

jwestelynck, 2021-07-05:
## Device

- Product name: Animal compatible motion sensor
- Manufacturer: ADEO
- Model identifier: LDSENK10
- Device type :
  - Sensor

## Screenshots

### Node info
[image]

### Basic
[image]

### Power Configuration
[image]

### Identify
[image]

### Poll Control
[image]

### IAS Zone
[image]

### Diagnostics
[image]

Smanar, 2021-07-06:
Hello, do you have a linux machine with full OS to test code modification ?
Can you show the cluster list fom deconz too, it s the picture, with the node title and all clusters visibles with endpoint ?

jwestelynck, 2021-07-06:
Hello, for sure i can test code modification

please fin the requested screenshot :

[image]

Smanar, 2021-07-07:
Ok so first try

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch adeo_sensor https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

The code is untested on my side, If I m right you will have a presence sensor, not checked for battery.

jwestelynck, 2021-07-07:
Great !

Sensor are now found by Phoscon !

[image]

But don't appear in sensor list.

[image]

In deconz it appear as a motion sensor (presence)

[image]

Smanar, 2021-07-07:
On first try lol
No problem, batery is good too ?

Can make the PR ?

jwestelynck, 2021-07-07:
Ok :)
I can find it on Jeedom. How to check battery ?

[image]

Smanar, 2021-07-07:
And it works fine ?
You haven't "a place" for battery on jeedom ? else I known you have somewhere "informations brutes"

jwestelynck, 2021-07-08:
looks good !

[image]

tampered sensor is also working.

Smanar, 2021-07-08:
Thx, making the PR

closed 2021-08-01 by manup