dresden-elektronik/deconz-rest-plugin #5099: ZM-RT201 Zigbee motion sensor

Cassiopea1, 2021-07-06:
## Device

- Product name: Zigbee motion sensor
- Manufacturer: no idea
- Model identifier: ZM-RT201
- Device type : 
  - Sensor

## Screenshots
[image]
[image]
[image]
[image]

### Basic

### Identify

### Alarms

### Device Temperature

### Groups

### Scenes

### On/Off

### Level Control

### Color Control

### Simple Metering

### Diagnostics

### Other clusters that are not mentioned above

Mimiix, 2021-07-06:
Can you try to pair it a few times? There seem to be some clusters missing. The Basic cluster should always be there, so maybe after a re-pair it should show up.

Cassiopea1, 2021-07-06:
I tried several times to pair it, but I always get the same result.

RangeMethod, 2021-07-07:
I have the exact same Issue, 
what can we do to get this done?

erkikoiv, 2021-07-07:
I have the same issue, it is visible in deConz but not in Phoscon and doesn't show the Basic cluster. What is missing?

dvdfrancos, 2021-07-08:
+1 with the exact same Issue

mactoolz, 2021-11-06:
Hi there,

I have a PIR motion sensor from Tuya.

ZM-RT201. I've already found a post on GitHub. Unfortunately, the subject was not pursued any further. I would like to start that again or ask how you can solve this problem with the sensor.

The topic is still the same as the one that can be seen in the DeConz but not in the Phoscon APP.

Now I'm not sure how far the sensor is supported or not.

Regards

Cassiopea1, 2021-11-06:
Hi,

 

My solution was to buy a Tuya zigbee hub. I can not wait for Dresden elektronik to get in the mood to support it.

 

Regards

mtressl, 2021-12-03:
bump - same issue here

Malle355, 2021-12-09:
Same to me. My Bad that I didn't found this before I bought those devices...

mtressl, 2021-12-10:
But now with so many pieces of the PIR available, can we re-try and get them working?
@Mimiix anything we can do or help to get you the right information?

Marcello2022, 2021-12-10:
Same issue here - would much appreciate if dresden elektronik looked into this in order to get support for the sensor

Mimiix, 2021-12-10:
> But now with so many pieces of the PIR available, can we re-try and get them working? @Mimiix anything we can do or help to get you the right information?

Nope, we need a dev to look into it. But due to ddfs, we are kinda on hold.

closed 2022-01-09 by github-actions[bot]

DayRunner131, 2022-03-23:
Bump. Same issue. No Phoscon support but see it connected in deConz.

Mimiix, 2022-03-23:
@Smanar

Smanar, 2022-03-24:
Basic cluster are still invisible ?

marvin743, 2022-07-10:
Same issue for me. Too sad, that I already bought the devices. What’s about the support? :-(

Smanar, 2022-07-11:
We need the basic cluster information to know what is the device.
At least the manufacture name / model id.

marvin743, 2022-07-11:
> We need the basic cluster information to know what is the device. At least the manufacture name / model id.

Device: Tuya Mini Zigbee PIR Sensor [https://de.aliexpress.com/item/1005004295765145.html?spm=a2g0o.productlist.0.0.7db430d8c3iigl&algo_pvid=4787d341-485d-4df1-8752-e432db178861&algo_exp_id=4787d341-485d-4df1-8752-e432db178861-0&pdp_ext_f=%7B%22sku_id%22%3A%2212000028665580362%22%7D&pdp_npi=2%40dis%21EUR%21%2113.0%21%21%21%21%21%400b0a01f816575525324815854e7853%2112000028665580362%21sea](url)

Should be this device:
[image]

DayRunner131, 2022-07-11:
I am unfortunately not seeing a manufacturer name / model identifier.

[image]

Smanar, 2022-07-11:
@DayRunner131 can try the same DDF if you think you have the same device, else you can
- select the basic cluster 0x0000 to open the cluster information
- press the "read" button adn in same time awake the device with pressing shortly the button.
- It will fill the attribute table on the same panel.

DayRunner131, 2022-07-11:
@Smanar I am not seeing a basic cluster information tab... It's consistent with all of these that I connect.

[image]

marvin743, 2022-07-11:
> Hello, can try this DDF
> 
> ```
> {
>    "schema":"devcap1.schema.json",
>    "manufacturername":"_TZ3000_mcxw5ehu",
>    "modelid":"TS0202",
>    "product":"Presence sensor (without Tuya cluster)",
>    "sleeper":true,
>    "status":"Gold",
>    "subdevices":[
>       {
>          "type":"$TYPE_PRESENCE_SENSOR",
>          "restapi":"/sensors",
>          "uuid":[
>             "$address.ext",
>             "0x01",
>             "0x0500"
>          ],
>          "items":[
>             {
>                "name":"attr/id"
>             },
>             {
>                "name":"attr/lastannounced"
>             },
>             {
>                "name":"attr/lastseen"
>             },
>             {
>                "name":"attr/manufacturername"
>             },
>             {
>                "name":"attr/modelid"
>             },
>             {
>                "name":"attr/name"
>             },
>             {
>                "name":"attr/swversion"
>             },
>             {
>                "name":"attr/type"
>             },
>             {
>                "name":"attr/uniqueid"
>             },
>             {
>                "name":"config/duration"
>             },
>             {
>                "name":"config/enrolled",
>                "public":false
>             },
>             {
>                "name":"config/on"
>             },
>             {
>                "name":"config/pending"
>             },
>             {
>                "name":"config/reachable"
>             },
>             {
>                "name":"state/lastupdated"
>             },
>             {
>                "name":"state/lowbattery"
>             },
>             {
>                "name":"state/presence",
>                "awake":true,
>                "parse":{
>                   "fn":"ias:zonestatus",
>                   "mask":"alarm1"
>                }
>             }
>          ]
>       }
>    ],
>    "bindings":[
>       {
>          "bind":"unicast",
>          "src.ep":1,
>          "cl":"0x0500"
>       },
>       {
>          "bind":"unicast",
>          "src.ep":1,
>          "cl":"0x0001",
>          "report":[
>             {
>                "at":"0x0021",
>                "dt":"0x20",
>                "min":60,
>                "max":3600,
>                "change":"0x00000001"
>             }
>          ]
>       }
>    ]
> }
> ```

How can i do this? Is there any manual available?

Thanks!

Smanar, 2022-07-12:
@DayRunner131 your device is not completely included https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues
Without modelid and manufacture name, can do nothing.

@marvin743 https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/DDF-cheat-sheet
You just need to put a text file called what_youwant.json in the "devices" folder with this content.
And if you are using HA https://forum.phoscon.de/t/how-to-add-edit-a-ddf-on-home-assistant-using-text-editor/1839

WhimsySpoon, 2022-08-09:
I've just tried adding this device and the Edit DDF button on the drop down does nothing. For other devices it works fine.

Smanar, 2022-08-09:
The "edit DDF" doesn't open the editor ? Or it is empty ?