dresden-elektronik/deconz-rest-plugin #5100: Going crazy, entire zigbee network gone bonkers, loosing lights left and right for several days now

fribse, 2021-07-06:
## Describe the bug

I've previously had a light drop off here and there, no big deal, a power cycle and things are back to normal. But lately it's gone completely crazy.
I'm loosing half the lights every day, making the WAF plummet.
## Steps to reproduce the behavior

  1. Have all lights signed into the network
  2. Leave it overnight
  3. Chaos

The problems seems to be sporadic, and illogical, the lights don't loose power, but something is disturbing them from operating.
## Expected behavior

## Screenshots

[image]
[image]
[image]
## Environment

 - Host system: PC
 - Running method: Home Assistent deCONZ Add-on
 - Firmware version: (26670700)
 - deCONZ version: (2.11.05)
 - Device:ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? If so: Which? AEOTEC Z-WAVE stick

## deCONZ Logs

## Additional context

Smanar, 2021-07-06:
And not only ikea ?
Something special on the mesh ? can be a special zone ?

Have you tried to check logs for a clue ?

manup, 2021-07-06:
Could it be that some other 2.4GHz devices are interfering on the same channel? WiFi and Bluetooth are often good candidates.
Usually when this happens you might see some 0xE1 errors in the logs (see https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Zigbee-Error-Codes-in-the-Log).

fribse, 2021-07-06:
I have end devices (sensors) from heiman and aqara, they've played well for a very long time. I then tried adding some heiman zigbee sirens, and that made it go to hell. I've since removed the heiman sirens again, but the network doesn't settle, it keeps kicking stuff out, so both me and (more importantly) my wife, are going bat crazy over this.
I would love to provide some logfiles, any idea on how I can get logfiles out from within the home assistant addon?

fribse, 2021-07-06:
@manup I have 2.4 GHz at channel 1 and 11, the zigbee says channel 15, but I don't know if it's the same 'numbers' for the channel they use.
I've done some wifi scanning, and the neighbours doesn't seem to interfere that much, but my wifi scanner does not show the zigbee as far as I know.
If I change the channel, I guess I have to redo all devices, right?

AutoHass8, 2021-07-07:
Your ZigBee channel 15 is in the sidelobe of WiFi channel 1. Try setting your WiFi channels to 1 & 6 and your ZigBee channel to 25, so they are as well separated as possible. ZigBee channel 25 will then be in the sidelobe of WiFi channel 11, in case of pesky neighbours...

I had some issues early on using ZigBee channel 15 on my ConBee I, so I made the above move right away. Don't remember if I had to redo any devices, but I've never had a network issue since.

Mimiix, 2021-07-07:
@manup @Smanar Would this be a possible bug in deCONZ or is this a user question?

@fribse We've got a forum since yesterday, feel free to take a look :) 
https://www.phoscon.de/forum

Smanar, 2021-07-07:
I don't think it s from a deCONZ bug, else we will have more users with the problem.

I don't remember how to get log on HA, but now you have access on the deconz GUI inside HA itself no ?

fribse, 2021-07-07:
@Mimiix , ok, I'll use that as well :-)

@Autohass8, darned, I feared that, I can see that out new neighbour is using channel 11, but there is nothing above 13, typical. for now I will move my ap with channel 1 to use channel 11, and the channel 11 to use channel 6 (my site survey says that it's better this way towards neighbours :-)
So the zigbee is now still on 15, where there is not much trafic now, and the background noise are a bit lower.

AutoHass8, 2021-07-07:
In that case you might consider moving ZigBee to channel 11, to get out of the WiFi channel 6 sidelobe. 
Have a look at this: https://www.metageek.com/training/resources/zigbee-wifi-coexistence.html

fribse, 2021-07-08:
I've moved it to 11 in the deconz, so far it is more stable. I did loose connection to all the lights etc. but as far as I can see, they are coming back...
Now I just have to wait to see if it has helped

fribse, 2021-07-09:
It didn't help enough, so now I've reset the entire deconz, uninstalled it from HA, reinstalled it, and I'm starting over, darned what a job... But for now the zigbee network has been rocksolid...

Smanar, 2021-07-09:
> It didn't help enough, so now I've reset the entire deconz, uninstalled it from HA, reinstalled it, and I'm starting over, darned what a job... But for now the zigbee network has been rocksolid...

So it mean, the problem was from something in the database ?

fribse, 2021-07-10:
@Smanar It would seem so yes. First of the channel clashing has been resolved, secondly the entire system is reset.
While enlisting all the sensors and switches again, I do have some problems with items appearing without me resetting anything, and leaving small odd items with invalid 4-hexcode names.

fribse, 2021-07-10:
Looking in deconz, after doing the reset, the network seems cleaner as well, most sensors, even if being xiaomi, or the spirit thermostat has attached to a 'local' router?

closed 2021-08-02 by fribse