dresden-elektronik/deconz-rest-plugin #5101: Loosing connection to all devices on daily basis

hesspoint, 2021-07-06:
## Describe the bug
I have a Raspbee II in a raspberry pi 4B. I have several sensors switches etc connected.

sometimes daily the devices stop working.  E.g. ne motion detectors don’t dedtect anything anymore and the last contact in the phoscon app is hours ago. Often that happens over night. When I restart the gateway via the phoscon app > gateway >power > restart then evertything reconnects and works fine until the next disconnect happens.

## Steps to reproduce the behavior
Not reproducible but happens sometimes once in 24 or once in 48h

## Expected behavior

## Screenshots

## Environment
Raspbee II
Version 2.11.05 / 6/23/2021
Firmware 26690700

Raspberry pi 4B on raspbian

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-07-07:
Hello, you don't receive update, but are you still able to send request ? Or you have only sensors ?

hesspoint, 2021-07-07:
@Smanar not sure what you mean exactly with send requests? You mean e.g. pressing a switch?
If yes then yes that also stops working. So there is no Communication anymore with any device switch, light or sensor. 
I can still access the phoscon web app all right. I see all devices but they don’t react. I can also see and access the gateway e.g. restart via the phoscon web app (not by pulling the plug). So it is not that the pi crashed and took everything with it. The pi is also running as it should. 
and all devices seem to have lost connection about the same time.

Mimiix, 2021-07-07:
Hi!

Can you please provide: 

- Logs 
- Screenshot of deCONZ Gui

this would give a bit more context.

I'll change to User Question for now, as it is hard to classify this as a bug and we can't make out if it really is.

Smanar, 2021-07-07:
For information deconz use 2 connexions :
- A permanent one, the websocket
- A classic one with REST request.

If the first one have issue, phoscon can't receive notification, so for him all devices have DC, but the second one is still working, and no, It s for that I m asking if you have only sensor/switch, to send request you need a bulb for exemple, a device that can receive order from phoscon.

But, Mimiix is right, we realy need logs.

closed 2021-08-06 by github-actions[bot]