dresden-elektronik/deconz-rest-plugin #5103: Hue lights drop out of network after toggling mains power

froogl, 2021-07-07:
## Describe the bug

As soon as a Hue light is disconnected from mains power it does not come back to the ZigBee network after powering it on again. The light needs to be reset and re-added to deCONZ in order to control it again. 

## Steps to reproduce the behavior

  1. Add a Hue light to the deCONZ network
     * I have tried this with LWB006 [Hue Lux, FW 5.130.1.30000] and LLC010 [Hue iris, FW 5.127.1.26581]
  2. They can be controlled normally through deCONZ
  3. Turn off mains power from the Hue light
  4. Turn mains power of Hue light back on
  5. The light turns on (default behavior) but does not come back to the ZigBee network. After a while deCONZ reports it as "unavailable".

For the light to come back these steps are required:

   1. Reset light through Hue dimmer switch or Living Colors remote
   2. Click "Add light" in deCONZ
   3. The light will be found and added to the network again

Resetting the device by removing it through deCONZ does not make any difference.

## Expected behavior

The light should not drop out of the network after power-cycling it.

## Screenshots

## Environment

 - Host system: PC
 - Running method: Debian packages
 - Firmware version: 26680700
 - deCONZ version: 2.11.05
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

Log with first communication error:

```
20:54:26:204 enqueue event config/localtime for /config/
20:54:26:206 Wait 1s till query finished
20:54:26:225 rule event /config/localtime: 20:54:25.204 -> 20:54:26.204 (1000ms)
20:54:26:717 poll node 00:17:88:01:10:57:c4:24-0b
20:54:26:721 Poll light node Pendelleuchte
20:54:26:804 read attributes of 0x001788011057C424 cluster: 0x0006: [ 20:54:26:808 0x0000 20:54:26:811 ]
20:54:26:814 add task 1362 type 19 to 0x001788011057C424 cluster 0x0006 req.id 159
20:54:26:818 Poll APS request 159 to 0x001788011057C424 cluster: 0x0006
20:54:26:895 Poll APS confirm 159 status: 0xE9
20:54:26:896     drop item attr/modelid
20:54:26:897     drop item attr/swversion
20:54:26:898     drop item state/bri
20:54:26:901 0x001788011057C424 error APSDE-DATA.confirm: 0xE9 on task
20:54:26:904 Erase task req-id: 159, type: 19 zcl seqno: 15 send time 0, profileId: 0x0104, clusterId: 0x0006
```

When the light is powered back on no entries appear in the log. I'd be happy to provide a more meaningful log, please just tell me what exactly is required.

## Additional context

Other ZigBee components on the network are not affected. My TRADFRI bulbs, OSRAM plugs and Busch-Jaeger switches come back to the network after a power outage. Only the Hue lights seem to have this problem. The ZigBee mesh is not bad in my case as I have many routers.

To me it kind of feels like the Hue lights are not correctly provisioned by deCONZ when adding them. I am not a ZigBee expert but to me it feels like they are not properly storing the key required to join the network. They only seem to keep the key while they remain mains-powered. I have a direct binding configured between one of my Busch-Jaeger switches and a Hue light. So that switch can control the light even if my Conbee is unplugged.

**When I power-cycle the Hue light while the Conbee is unplugged the direct binding also falls apart**. That's why I think there is a provisioning issue with Hue lights, which results in them not being permanent members of the ZigBee network (only "temporary members" while they remain mains-powered).

SwoopX, 2021-07-07:
> I am not a ZigBee expert but to me it feels like they are not properly storing the key required to join the network.

I wonder how deconz comes into play if the device cannot store the network settings persistently in its NVRAM.

froogl, 2021-07-07:
> > I am not a ZigBee expert but to me it feels like they are not properly storing the key required to join the network.
> 
> I wonder how deconz comes into play if the device cannot store the network settings persistently in its NVRAM. Have you searched the net for this behavior?

The thing with the network/key settings is only a vague theory from my side. DeCONZ performs the initial provisioning of these lights. And that's what _might_ be going wrong. Maybe it's a key incompatibility or something like that. I have searched the net for this and there are no similar reports. The Hue lights are on the official compatibility list for deCONZ so I'd expect that they are well-tested. And they are generally known for their stable ZigBee implementation.

But again, that is just a theory from my side. I honestly don't know what the reason for these dropouts could be and I don't know how to debug this. That's why I am asking what is needed to get to the root of this.

I'll try to provide anything that might be helpful. Network screenshots (doubt that they are helpful), logs, ... .

> My lights all behave fine.

That's great. Do you have any Hue lights (especially LWB006 or LLC010)?

## Screenshots

Maybe these screenshots are helpful. I have not changed anything from the default values:

[image]
[image]

SwoopX, 2021-07-07:
What you describe is that the device seems to not store the network settings persistently in its NVRAM. The fact that you did a direct binding to the BJ swtich even adds more to this, as the devices are able to communicate directly without the need of a coordinator by using broadcasts. If they, however, do not share the same settings, then there's of course no communication.

However, while writing this, I recall that especially Osram plugs are pretty shi**y repeaters and I know lots of ppl having trouble with those. If the lights would be connected to one of those, but they refuse to forward any communication, this could also be an explanation. You could check what happens if you discommect the plugs and try to pair the lights directly with the coordinator, disconnect a light and then reconnect to power.

Other than that, there's barely any chance to see what's going on without a zigbee sniffer. Eventually, the deconz logs show anything when the light is reconnected to power (DBG_INFO, DBG_INFO_L2, DBG_ERROR, DBG_APS, DBG_APS_L2), but I wouldn't expect anything. A screenshot you your mesh could give insights in your current setup.

I personally have some LWB006 all within reach of at least 5 on Osram repeating devices.

ebaauw, 2021-07-07:
Could you double check if you can control the light through a group (e.g. /groups/0).  Then after power-cycle check that the light still responds to the group commands.  If it does, you're probably facing an issue with Zigbee routing.  The device hasn't "dropped" from the network; the network doesn't know how to reach it.  Best advice for that case is: don't power down your smart lights and ditch any routers causing issues.  If it does not react to group commands after power cycle, I suspect a faulty device (although it would be weird coincidence on two lights at the same time).

There's no such thing as a "temporary" joining of a Zigbee network: the device knows the network parameters (key, channel, etc), or it doesn't.  These settings should be persisted across power cycles on/by the device in non-volatile memory.

froogl, 2021-07-09:
> However, while writing this, I recall that especially Osram plugs are pretty shi**y repeaters and I know lots of ppl having trouble with those. If the lights would be connected to one of those, but [...]

@SwoopX Thanks for your answer. I actually have only one OSRAM plug right now (current firmware). On an older firmware it caused issues with my XIaomi devices, but not with the most recent one. But as you have suggested I have unplugged it and let my ZigBee network rearrange without it. It's still showing the same behavior that **none of my Hue lights** survive a power outage.

> What you describe is that the device seems to not store the network settings persistently in its NVRAM. The fact that you did a direct binding to the BJ swtich even adds more to this, as the devices are able to communicate directly without the need of a coordinator by using broadcasts.

> Eventually, the deconz logs show anything when the light is reconnected to power (DBG_INFO, DBG_INFO_L2, DBG_ERROR, DBG_APS, DBG_APS_L2), but I wouldn't expect anything. 

I have enabled the debug levels you have suggested but after reconnecting the lights the log does not reveal any information about them. They seem to be pretty much dead. This is a log file from deCONZ:

https://gist.github.com/froogl/5a7b524f363f56af317d3247a398801f

The device 001788011057C424 is powered on at 13:55:00 but there is no sign of life after that moment.

> A screenshot you your mesh could give insights in your current setup.
 
These are two screenshots, one with the OSRAM device and one without it. Please note that I have removed the names of family members in the device names :)

[image]

[image]

As you can see I have quite a few Philips Hue devices (MAC address starting 00:17) and they all show that behavior. We are talking about Hue iris, Hue Lightstip, Hue Lightstrip plus, Hue Lux and Hue ambience candles here. So it's not a single device, but rather all of them.

I found a similar report about Hue lights (#1614) but in that case a newer firmware seemed to have helped. My Hue devices are all on the latest firmware. And in that old report a simple scan for lights helped (without resetting them), which is also not the case here.

> I personally have some LWB006 all within reach of at least 5 on Osram repeating devices.

I think this all used to be working for me as well some time ago. But like 2 months ago I was creating direct bindings between my Busch-Jaeger scene controller and some lights and all of a sudden all routers **except IKEA routers** fell of the ZigBee mesh. I reconnected all of them by resetting and searching for lights again. I think this might be the day my Hue lights broke. It's hard to tell what happened back then but I think that was the breaking event. It would be nice to know how to recover from this state now.

> @froogl, Could you double check if you can control the light through a group (e.g. /groups/0). Then after power-cycle check that the light still responds to the group commands.

@ebaauw Thanks for your input. Unfortunately my Hue devices also do not respond to group commands after a power-cycle. 

> If it does not react to group commands after power cycle, I suspect a faulty device (although it would be weird coincidence on two lights at the same time).

Well that would mean that all of my Hue devices are faulty and I have quite a few of them. I also added one of the LWB006 lights back to my Hue bridge (which I am still using for Hue Entertainment in the living room) and it's working on there. So I can say that this is not an issue with the light itself.

I normally keep all of my ZigBee devices powered, but in case of a power outage I am in big trouble. And these things usually happen when you are not around :)

So what I am thinking is that there has to be some sort of an invalid state somewhere within the network. The big question is how to detect that and recover from it.

ebaauw, 2021-07-09:
> Unfortunately my Hue devices also do not respond to group commands after a power-cycle.

This is weird.  Do you have a UPS on the server running deCONZ, or does that also power-cycle?

If you power-cycle a single Hue light, does it also drop off the network?

And if you power down the coordinator device, and then power-cycle a single Hue light?

And if you only stop deCONZ, but keep the coordinator powered, and then power-cycle a single Hue light?

> So what I am thinking is that there has to be some sort of an invalid state somewhere within the network.

The only (theoretical) scenario I can think of, is that the lights boot up too soon, way before the coordinator/deCONZ is up, try and re-join the network, but conclude that it isn't there and somehow leave it.  Or are told by some router that they should leave.

Did you recently change and of the network parameters (like channel), causing _NWK Update ID_ to change?  I've had issues some time that not all devices picked up the new id.  My best practice is not to touch the network parameters, and simply start a new network if you want a change (or: make sure you have the parameters you want before joining devices).

froogl, 2021-07-09:
@ebaauw Thanks for your quick response.

> This is weird. Do you have a UPS on the server running deCONZ, or does that also power-cycle?

No, the deCONZ host/cooridnator does not power-cycle. However I have stopped deCONZ and unplugged the Conbee a couple of times.

> If you power-cycle a single Hue light, does it also drop off the network?

Yes. In fact I have not tried power-cycling all Hue lights at once yet (like in a real power-outage scenario). Only light-by light. The issue is that after resetting the lights I am losing all direct bindings and stored scenes, that's why a real power-outage is an actual nightmare right now.

> And if you power down the coordinator device, and then power-cycle a single Hue light?

I have tried that, same result. I have tried this with a single Hue bulb when the coordinator was down and the bulb also did not come back to the network. The direct binding with my Busch-Jaeger switch also did not work when I power-cycled the Hue light while the coordinator was down, that's why I assume that it's not the Cooridnator itself, but some weird issue within the network.

> And if you only stop deCONZ, but keep the coordinator powered, and then power-cycle a single Hue light?

Ahh, I have not tried that. Just did this but unfortunately it's the same result:

1. Stop deCONZ
2. Power-cycle single Hue light
3. Test direct binding with single Hue light -> fails
4. Start deCONZ
5. Device is reported as unavailable after a while and cannot be controlled any more (not even through group commands)

> The only (theoretical) scenario I can think of, is that the lights boot up too soon, way before the coordinator/deCONZ is up, try and re-join the network, but conclude that it isn't there and [...]

Hmm, okay. But the coordinator and all other routers are up my case. After a real power outage this would be different, but right now everything is up.

> Did you recently change any of the network parameters (like channel), causing _NWK Update ID_ to change? I've had issues some time that not all devices picked up the new id. My best practice is not to touch the network parameters, and simply start a new network if you want a change (or: make sure you have the parameters you want before joining devices).

Just like you I never mess with these parameters. Setting a channel was the only thing I did when I started with deCONZ. I did that before adding any device. Since then I have not touched a thing in the network parameters dialog, I am even a little afraid when opening it :smiley: And I am usually running deCONZ in headless mode so nothing could have changed this by accident.

But there is one thing I have not tried yet: simulating a real power outage by cutting mains power to my entire house (including all ZigBee routers and the coordinator). Maybe I have one device which is in a bad state which needs a power-cycle. If you guys cannot think of anything else I guess then this is the last thing to try. The right time for this needs to be chosen carefully though, because when I am actually going to lose all of my Hue lights at once it will require a lot of work to reset and re-pair them (especially setting up scenes and direct bindings takes a lot of time).

But if you have any other suggestions which would keep me from power-cycling the house (and potentially re-doing all my deCONZ setup) I'd be super happy!

ebaauw, 2021-07-09:
> The issue is that after resetting the lights I am losing all direct bindings and stored scenes

That’s what a reset is supposed to do.  A light should not reset on power cycle, though.  What direct bindings are you talking about?  Afaik, Hue lights don’t support attribute reporting, so there’s no use to have any bindings on a Hue light.  Hue lights might be destinations of bindings on controllers, but those are stored on the controller.  Usually, you’d use bindings to a group, however.

> simulating a real power outage by cutting mains power to my entire house (including all ZigBee routers and the coordinator). Maybe I have one device which is in a bad state which needs a power-cycle.

In that case, you could just power-cycle the non-Hue routers, I suppose.

SwoopX, 2021-07-09:
It's irritating that there's no `DeviceAnnce of LightNode: 0x04cf8cdf3c764934 Permit Join: 0` message when the light was powered on. Based on what I see in the code, this can only be if the announcing device is not part of the `LightNode` vector, so unknown?

There's also quite a number of errors in the log, but for me this is in line with your observations and no real surprise then. Maybe it's worth to have a look at the DB if there are any anomalies...

Before trying to recreate the whole network, I'd suggest to put the stick into another system with deconz installed, the network setting are kept on the stick. When deconz is running, you should see the nodes which are still able to communicate pop up real soon. To have them up and running again, it should suffice to just search for either lights or sensors (without resetting them). Sleepy End devices like e.g. the Tradfri remote need to be waken up to get recognized again while the search is running.

ebaauw, 2021-07-09:
> It's irritating that there's no DeviceAnnce of LightNode: 0x04cf8cdf3c764934 Permit Join: 0 message when the light was powered on.

Some lights, like the LIDL Livarno Lux, don't send a _Device Announcement_ on boot.  I have one E27 Aqara light with mac prefix 001584 (instead of 04cf8c) that does send a _Device Announcement_ , but I haven't seen any other Xiaomi lights.

froogl, 2021-07-11:
> What direct bindings are you talking about?

Yeah, sorry. I meant group bindings. I am binding some of the rockers of my Busch-Jaeger switches to a ZigBee group and then I am adding my lights to these groups. I do that to be as independent as possible from the gateway.

> In that case, you could just power-cycle the non-Hue routers, I suppose.

Unfortunately not possible, as my Busch-Jaeger devices are hardwired (in-wall) so I need to flip the automatic circuit breaker :)

> It's irritating that there's no `DeviceAnnce of LightNode: 0x04cf8cdf3c764934 Permit Join: 0` message when the light was powered on. Based on what I see in the code, this can only be if the announcing device is not part of the `LightNode` vector, so unknown?

Yeah, it also feels like the light is unknown to the network when it comes back. That's why I came up with the theory that it has lost it's key/config.

> There's also quite a number of errors in the log, but for me this is in line with your observations and no real surprise then. Maybe it's worth to have a look at the DB if there are any anomalies...

But AFAIK DB errors would not matter for group bindings, right? In this case at least the group binding should survive the power cycle while the Conbee was unplugged I guess.

I have looked at my DB with sqlite and I cannot see anything unusual. All SELECTs work, the DB has not grown to an unusual size and I also do not see any DB-related errors in the logs.

> Before trying to recreate the whole network, I'd suggest to put the stick into another system with deconz installed, the network setting are kept on the stick. When deconz is running, you should see the nodes which are still able to communicate pop up real soon.

This did not work, unfortunately. The devices showed up again (with generic names) but they were still dropping from the network after a power-cycle. So it's probably not directly related to the Conbee/DeCONZ.

I just power-cycled my entire house and some Hue lights actually came back (of same type and FW as other lights which are dropping). One LTW012 came back, another LTW012 in the same room got stuck. Both are running the latest 1.50.2_r30933 firmware.  And some of my Busch-Jaeger routers (wall switches/relays) have also dropped from the network (not all of them). So I come to conclude that a fresh start is required.

I have written a small script which backs up my device names and scene configuration through the REST API so at least this step will save me some time. When recreating the network I'll start with just IKEA, Hue and Busch-Jaeger and see how this goes (no Xiaomi, no OSRAM).

froogl, 2021-07-16:
So, I started with a fresh DeCONZ/ZigBee network. I double-checked that the network key and PAN ID have changed after a gateway reset. I added my Hue components, my Busch-Jaeger ZLL switches/dimmers and my IKEA lights. No OSRAM, no Xiaomi sensors. Just essential lights, motion sensors and dimmer switches.

And the result is rather frustrating: I am getting the same behavior. After a power-cycle some lights/routers do not come back to the network. In the end I can say the following:

- IKEA components are rock-solid (I only have the current TRADFRI stuff with the latest ZigBee 3.0 firmware). They always come back to the network no matter what.
- Some Hue lights drop out of the network after a power cycle. FW and different HW versions do not matter here. It's erratic.
- Some Busch-Jaeger components also keep dropping from the network. Apparently only devices with four-rockers are affected, but only the routers. The battery-operated wall switches (end devices) are not affected, even if they have 4 rockers.

So in the end I think the culprit might be the Busch-Jaeger devices. I have read somewhere that they have very low memory, which might be a problem especially for the 4-rocker variants. If I am right with this I still don't understand why they are tearing down some of the Hue lights with them. Is it because they are both ZLL-only devices? The IKEA routers are ZHA devices (ZB 3.0), that's the only difference I can think of.

On the other hand: my OSRAM plug is ZLL-only as well and it comes back reliably.

Any thoughts on this one? @ebaauw @SwoopX

SwoopX, 2021-07-19:
Tbh, I'm a bit running out of ideas here. I mean, it can always happen that a devices get fubar, but more than one would be a very strange coincidence. It's also ironic that the IKEA lights are doing just fine since they are known to sometimes fall off the network every now and then (but probably for other reasons).

Not too sure about the BJ either, as I also got a 4-gang. They indeed have the annoyance of a very limited binding capability (no more than 4) restricting convenience a bit, but also those can back flawlessly whenever I had to pop the fuse. However, there's always plenty of routers available in my case (not lights, but plugs).

It would be really interesting to see a sniffer log of the network. I somehow can't imaging the lights will leave the network "on purpose" due to whatever reason. Even if they would leave a message in the debug log if I recall correctly. Like said, the lights in my hallway are also always powered off physically, but come back though I have to admit they're older lights. Did yours show that behavior ever since? Might be worth to see if other users may experience something similar although I guess it's not a too common scenario that people fully power them down. Other than that, I'd probably reach out to Hue support to ask for support or known issues.

Sorry for not having anything better to offer.

froogl, 2021-07-20:
@SwoopX Thanks for your reply. I am also running out of ideas... I have migrated some of the "Harakiri" Hue bulbs and BJ switches back to the Hue bridge where they are working fine. No more dropouts after a power-cycle. So I guess they are not faulty.

> Not too sure about the BJ either, as I also got a 4-gang. They indeed have the annoyance of a very limited binding capability (no more than 4) restricting convenience a bit, but also those can back [...]

How many nodes do you have in your network? If you actually have one of these BJ switches, maybe you could do some testing for me: remove it from deCONZ, reset it and pair it again (with current stable deCONZ/Conbee firmware). Then pop the fuse to check if it survived it. You need to be a little adventurous for this though.

**And my theory that only the 4-gang version is affected has proven wrong**. Just yesterday I tested another full power outage and all of the BJ routers (1-gang, 2-gang and 4-gang) have dropped out of the network this time. The three remaining Hue bulbs survived the power outage this time. Which makes me think that there really is something odd going on with these BJ devices..

I think I also have enough routers in my network, even without the BJ routers. Still a few bulbs, a TRADFRI signal repeater and one OSRAM smart plug.

> It would be really interesting to see a sniffer log of the network. I somehow can't imaging the lights will leave the network "on purpose" due to whatever reason.

Yeah, I know. I'd also love the sniff the network but I am still lacking the hardware for that and the Conbee II apparently cannot do that. And I kind of suspect that there won't be any traffic when the devices are powered back on.

> Did yours show that behavior ever since?

No, definitely not. They used to work just fine. I am regularily testing emergency scenarios (what happens if ZigBee GW is down or after a power outage) and this used to work with deCONZ and the ConBee II. That's why I thought a fresh network might fix it but unfortunately it did not.

> Might be worth to see if other users may experience something similar although I guess it's not a too common scenario that people fully power them down.

You are right about the power cycle. I guess not many people regularly test emergency scenarios like a full power or gateway outage. So it might be that this issue might only come up for people in the future.

> Other than that, I'd probably reach out to Hue support to ask for support or known issues.

The thing is: the Hue bulbs work on the Hue hub. So I doubt support is going to provide any help on this.

> Sorry for not having anything better to offer.

That's okay. I understand that this is a very specific issue which is extremely hard to debug. If you are willing to go through a full reset and pair-procedure with one of your BJ switches that would be cool though. I'd also understand if you do not want to touch your production setup for this.

I am going to make more tests and report back here. Since the network is working as long as there is no power outage I can live with this risk for a while. I am planning on testing Zigbee2Mqtt with a CC2652RB stick in the future to see if I can get the devices working reliably there. If I can get them to work I still suspect that it might be some sort of provisioning issue. Maybe within the Conbee II firmware?

froogl, 2021-07-28:
Another update: I have connected two of my BJ switches to a CC2652RB stick (together with some OSRAM plugs) and they are working great there. They always come back after a power outage. So this issue seems to be specific to deCONZ or the Conbee.

SwoopX, 2021-07-29:
Hm, that's strange... However, I'm tempting to agree that it's an issue on deconz side but I wonder how to get a hold of it 🤔 

However, while testing some other stuff on the weekend, I stumbled over some oddities when analyzing the network traffic. Not sure if that might be related in any way but it goes (very simplified) like this: deconz is using a mechanism to check if a previously unreachable mains powered device might be reachable meanwhile with a special request. I have 5 mains powered routers in my DB where 1 was active, 1 was powered down and 3 are not part of the network anymore. I noted that, within my environment, those requests stopped after about 15 mins and that only 3 out of 4 inactive devices were queried.

I've reported that already internally and hoping for some feedback soon. Unfortunately, I cannot check anything myself since in the deconz core where I don't have access to.

froogl, 2021-08-29:
I have updated the Conbee II to the latest firmware (26720700) and it seems to have solved the issue. All nodes came back after a full power cycle of my entire house. :tada: 

So it likely was a firmware issue. I don't fully understand why it also affected group bindings, but hey, it's working now. If similar issues are coming up it might be good to point users to FW 26720700 for testing @SwoopX @manup ;)

closed 2021-08-29 by froogl

cobch, 2021-09-28:
Hi,
My firmware version of conbee II is 26660700. How could I update the firmware since is says updated on Phoscon?

Mimiix, 2021-09-28:
Hi @cobch I think its best to check the wiki or ask on the forum.