dresden-elektronik/deconz-rest-plugin #5105: API brightness value does not reflect actual state occasionally

TheTrueRandom, 2021-07-07:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

Smanar, 2021-07-08:
Hello, so first phoscon use the API, so if the value is good in phoscon, it need to be good too on the API.
Deconz have a special working mode, when you send a "on" request, it change the value to "on" in the API, then send the zigbee request, and update the API when it receive the return from the bulb.

So you can have the good value some seconds after, when the device make the return state.

TheTrueRandom, 2021-07-08:
Hi @Smanar 
the API is never good, from the scenario above, I can do the API requests for hours, the values stay the same as in the screenshots. I also don't know where the UI gets the values from (as it has to be the API) but strangely it takes about 10 seconds after loading the UI until the value jumps from 0 to the actual state. The API however always shows 0 for brightness. Also I noticed the same behavior is also for hue and sat. That definitely looks like a bug.

Smanar, 2021-07-09:
When I say the API update the value from return from device, it s only for the light entry, not the group one, it don't work same.
On my side I never use group value, not reliable, for exemple if you have a bulb in 2 group, one group "on" , and the second one "off"

TheTrueRandom, 2021-07-09:
@Smanar that would be fine, but as you can see from the screenshot it's not. The light also shows state: `"bri": 0` and that's incorrect.

Smanar, 2021-07-09:
Ha yes you miss too this last return state with good value.

TheTrueRandom, 2021-07-09:
So how come the UI shows the correct value?

Smanar, 2021-07-10:
That is another question ^^.

It s possible look at websocket return in same time, using a websocket client as plugin in browser, or just enable developer tools in browser.
Probably a "timing" issue.

TheTrueRandom, 2021-07-11:
No it is not a timing issue. Reloading the page will always show the correct result, after some time.
So I reload the UI, the value shows 0 (as the API does).
after about 10s, sometimes more, sometimes less, the value jumps to the correct state, only on the UI!

TheTrueRandom, 2021-08-02:
Issue still active and quite bothering. I provided easy steps to reproduce it.

Mimiix, 2021-08-03:
I'll ask Manup to check it :)

TheTrueRandom, 2021-09-01:
The issue is still not solved

TheTrueRandom, 2021-09-21:
@Mimiix can you please reopen the issue it is still not fixed

Mimiix, 2021-09-22:
Sure :) Make sure to keep it open tho

KipK, 2022-01-31:
Still unsolved. I confirm we don't get brightness feedback through API.
For now I have to trick by getting the brightness of one member of the group.