dresden-elektronik/deconz-rest-plugin #5107: All connected IKEA devices drop from the network 5-10m after deconz is started

delta9690, 2021-07-08:
## Describe the question or issue you are having

Hello,

I am running deconz via the Home Assistant OS addon on a Raspberry Pi 3B+ with a Raspbee 2. All IKEA devices (I currently only have IKEA lights and switches, but I have an Aqara temperature sensor arriving tomorrow) will connect up fine after deconz is started and respond to commands via deconz, phoscon, and hue essentials normally, but 5-10 minutes after startup the lights and switches gradually drop off the network (they lose all links to neighbors and stop responding to commands). Power cycling the lights does not change this whatsoever. After rebooting deconz, the behavior happens exactly the same way. Sorry in advance for any formatting booboos. I'm new to using Github.

## Screenshots

## Environment

 - Host system: Raspberry Pi 3B+
 - Running method: Home Assistant OS deconz addon
 - Firmware version: 26690700
 - deCONZ version: 2.11.5
 - Device: Raspbee 2
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? **None**

## deCONZ Logs

Logs:

`21:44:58:043 dev /dev/ttyS0
21:44:58:045 DB UPDATE device_descriptors SET data = x'f2e0a164000100012100', timestamp = 1625715898 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 242 AND type = 4
21:44:58:047 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 242, 4, x'f2e0a164000100012100', 1625715898 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
21:44:58:048 GW firmware version: 0x26690700
21:44:58:369 DB UPDATE device_descriptors SET data = x'0104010500010300000a00190003010020000005', timestamp = 1625715898 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 1 AND type = 4
21:44:58:371 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010500010300000a00190003010020000005', 1625715898 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
21:45:08:040 dev /dev/ttyS0
21:45:08:041 GW firmware version: 0x26690700
21:45:08:852 discovery for zombie 0x90fd9ffffe00fdbd dropped, last try was 453 s ago
21:45:18:041 dev /dev/ttyS0
21:45:18:043 GW firmware version: 0x26690700
21:45:24:212 discovery for zombie 0x000b57fffef376da dropped, last try was 144 s ago
21:45:28:040 dev /dev/ttyS0
21:45:28:041 GW firmware version: 0x26690700
21:45:31:990 Current channel 11
21:45:32:003 CTRL got nwk update id 1
21:45:32:016 Device TTL 3565 s flags: 0x7
21:45:38:040 dev /dev/ttyS0
21:45:38:042 GW firmware version: 0x26690700
21:45:39:573 discovery for zombie 0x000b57fffef57d3a dropped, last try was 271 s ago
21:45:48:040 dev /dev/ttyS0
21:45:48:042 GW firmware version: 0x26690700
21:45:58:044 dev /dev/ttyS0
21:45:58:046 GW firmware version: 0x26690700
21:46:08:040 dev /dev/ttyS0
21:46:08:041 GW firmware version: 0x26690700
21:46:11:732 discovery for zombie 0x000b57fffef376da dropped, last try was 191 s ago
21:46:18:039 dev /dev/ttyS0
21:46:18:042 GW firmware version: 0x26690700
172.30.32.2 - - [07/Jul/2021:21:46:20 -0600] "GET /novnc/websockify HTTP/1.1" 101 8169545 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
localhost - - [07/Jul/2021 21:46:20] 127.0.0.1:5900: Client closed connection
localhost - - [07/Jul/2021 21:46:20] 127.0.0.1:5900: Closed target
Ignoring interrupted syscall
21:46:27:229 discovery for zombie 0x000b57fffef57d3a dropped, last try was 319 s ago
21:46:28:044 dev /dev/ttyS0
21:46:28:046 GW firmware version: 0x26690700
172.30.32.2 - - [07/Jul/2021:21:46:28 -0600] "GET /pwa/index.html HTTP/1.1" 200 113153 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/ingress.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
172.30.32.2 - - [07/Jul/2021:21:46:29 -0600] "GET /api/6C64FE0F7D/config?_=1625715989180 HTTP/1.1" 200 266 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
172.30.32.2 - - [07/Jul/2021:21:46:29 -0600] "GET /api/6C64FE0F7D/config?_=1625715989181 HTTP/1.1" 200 266 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
172.30.32.2 - - [07/Jul/2021:21:46:29 -0600] "GET /api/6C64FE0F7D/groups?_=1625715989182 HTTP/1.1" 403 92 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
172.30.32.2 - - [07/Jul/2021:21:46:29 -0600] "PUT /api/6C64FE0F7D/config HTTP/1.1" 403 92 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
21:46:30:349 New websocket 127.0.0.1:55112 (state: 3) 
21:46:31:989 Current channel 11
21:46:32:002 CTRL got nwk update id 1
21:46:32:015 Device TTL 3505 s flags: 0x7
172.30.32.2 - - [07/Jul/2021:21:46:35 -0600] "PUT /api/6C64FE0F7D/groups/19/action HTTP/1.1" 403 103 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
172.30.32.2 - - [07/Jul/2021:21:46:36 -0600] "PUT /api/6C64FE0F7D/groups/19/action HTTP/1.1" 403 103 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
172.30.32.2 - - [07/Jul/2021:21:46:37 -0600] "PUT /api/6C64FE0F7D/groups/19/action HTTP/1.1" 403 103 "https://aglet.duckdns.org:8123/api/hassio_ingress/MYLwylXl9yOE_0sAOkJfbdkEG9X0haj1e4mU3LR11aQ/pwa/index.html" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/91.0.4472.124 Safari/537.36"
21:46:38:044 dev /dev/ttyS0
21:46:38:046 GW firmware version: 0x26690700
21:46:43:724 discovery for zombie 0x90fd9ffffe00fdbd dropped, last try was 47 s ago`

Let me know if you need the logs from the debug view in deconz. VNC viewer wasn't playing nice.

## Additional context

Mimiix, 2021-07-08:
Hi!

> Let me know if you need the logs from the debug view in deconz. VNC viewer wasn't playing nice.

We really like to have these 😅

delta9690, 2021-07-08:
Lol, not an issue, but it'll have to wait until (my) morning (about 8 hours). I don't currently have the energy to fix the (unrelated) networking issues that are breaking VNC.

delta9690, 2021-07-08:
```
08:15:31:521 created username: 5159C37F23, devicetype: Home Assistant
08:15:31:635 dev /dev/ttyS0
08:15:31:854 dev /dev/ttyS0
08:15:31:854 COM: /dev/ttyS0 / serialno: , RaspBee
08:15:32:521 Announced to internet https://phoscon.de/discover
08:15:32:523 discovery server date: Thu, 08 Jul 2021 14:15:32 GMT
08:15:32:523 	 local time seems to be ok
08:15:33:917 Device firmware version 0x26690700 RaspBee II
08:15:33:956 unlocked max nodes: 200
08:15:34:192 Device protocol version: 0x010E
08:15:34:213 new node - ext: 0x00212effff06e6f2, nwk: 0x0000
08:15:34:386 Current channel 11
08:15:34:426 CTRL got nwk update id 1
08:15:34:439 CTRL ANT_CTRL 0x03
08:15:34:541 Device protocol version: 0x010E
08:15:34:702 Current channel 11
08:15:34:741 CTRL got nwk update id 1
08:15:34:755 CTRL ANT_CTRL 0x03
08:15:35:791 DB UPDATE device_descriptors SET data = x'10400f3511472b0040002b0000', timestamp = 1625753735 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 0 AND type = 2
08:15:35:793 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'10400f3511472b0040002b0000', 1625753735 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
08:15:35:982 LightNode 1: Configuration tool 1 added
08:15:35:985 DB UPDATE device_descriptors SET data = x'0104010500010300000a00190003010020000005', timestamp = 1625753735 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 1 AND type = 4
08:15:35:986 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010500010300000a00190003010020000005', 1625753735 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
08:15:36:115 DB UPDATE device_descriptors SET data = x'f2e0a164000100012100', timestamp = 1625753736 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 242 AND type = 4
08:15:36:117 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 242, 4, x'f2e0a164000100012100', 1625753736 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
08:15:36:196 ZDP discovery done in 330 ms
08:15:42:088 dev /dev/ttyS0
08:15:42:096 GW update firmware found: /usr/share/deCONZ/firmware/deCONZ_ConBeeII_0x26660700.bin.GCF
08:15:42:096 GW firmware version: 0x26690700
08:15:45:998 saved node state in 0 ms
08:15:45:999 sync() in 0 ms
08:15:52:071 dev /dev/ttyS0
08:15:52:073 GW firmware version: 0x26690700
08:16:02:063 dev /dev/ttyS0
08:16:02:065 GW firmware version: 0x26690700
08:16:12:064 dev /dev/ttyS0
08:16:12:066 GW firmware version: 0x26690700
08:16:14:292 device state timeout ignored in state 3
08:16:14:453 device state timeout ignored in state 3
08:16:14:612 device state timeout ignored in state 3
08:16:14:667 Device protocol version: 0x010E
08:16:14:873 Current channel 0
08:16:14:908 CTRL got nwk update id 1
08:16:14:919 CTRL ANT_CTRL 0x03
08:16:15:018 CTRL got stack debug assert code: 0x2021, type: 0x03
08:16:15:173 Device protocol version: 0x010E
08:16:15:355 Current channel 11
08:16:15:395 CTRL got nwk update id 1
08:16:15:408 CTRL ANT_CTRL 0x03
08:16:15:504 CTRL got stack debug assert code: 0x2021, type: 0x03
08:16:16:033 DB UPDATE device_descriptors SET data = x'10400f3511472b0040002b0000', timestamp = 1625753776 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 0 AND type = 2
08:16:16:035 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'10400f3511472b0040002b0000', 1625753776 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
08:16:16:839 DB UPDATE device_descriptors SET data = x'f2e0a164000100012100', timestamp = 1625753776 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 242 AND type = 4
08:16:16:841 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 242, 4, x'f2e0a164000100012100', 1625753776 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
08:16:17:635 DB UPDATE device_descriptors SET data = x'0104010500010300000a00190003010020000005', timestamp = 1625753777 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 1 AND type = 4
08:16:17:638 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010500010300000a00190003010020000005', 1625753777 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
08:16:21:013 Current channel 11
08:16:21:026 CTRL got nwk update id 1
08:16:21:039 Device TTL 3594 s flags: 0x7
08:16:21:052 Warning frame counter 2752266 (0x0029FF0A) lower than previous one 2752267 (0x0029FF0B)
08:16:21:053 Raise frame counter to 2752567 (0x002A0037)
08:16:22:067 dev /dev/ttyS0
08:16:22:068 GW firmware version: 0x26690700
08:16:32:063 dev /dev/ttyS0
08:16:32:065 GW firmware version: 0x26690700
08:16:42:063 dev /dev/ttyS0
08:16:42:065 GW firmware version: 0x26690700
08:16:45:460 Skip idle timer callback, too early: elapsed 937 msec
08:16:46:026 saved node state in 0 ms
[... 292 lines cut ...]
08:19:37:820 Mgmt_Lqi_req zdpSeq: 94 to 0x00212EFFFF06E6F2 start index 1
08:19:37:822 APS-DATA.request id: 123, addrmode: 0x03, addr: 0x00212effff06e6f2, profile: 0x0000, cluster: 0x0031, ep: 0x00 -> 0x00 queue: 0 len: 2 tx.options 0x00
08:19:37:874 APS-DATA.confirm id: 123, status: 0x00 SUCCESS
08:19:37:876 APS-DATA.confirm request id: 123 -> confirmed, timeout 1625753977
[... cut ...]
08:19:40:374 APS-DATA.indication request id: 135 -> finished
08:19:40:376 APS-DATA.request id: 135 erase from queue
08:19:40:378 ZDP status = 0x00 -> SUCCESS
08:19:40:380 ZDP Mgmt_Lqi_rsp zdpSeq: 95 from 0x00212EFFFF06E6F2 total: 3, startIndex: 2, listCount: 1
08:19:40:381     * neighbor: 0x000B57FFFEF376DA (0x08C0), LQI: 255, relation: 0x02 rxOnWHenIdle: 1
08:19:40:701 discovery for zombie 0x000b57fffef376da dropped, last try was 66 s ago
08:19:42:104 dev /dev/ttyS0
08:19:42:113 GW firmware version: 0x26690700
```

Smanar, 2021-07-08:
Hello, and after rebooting deconz, all is working again, for  5 mn ?

delta9690, 2021-07-08:
It's hit or miss. Sometimes the bulbs will operate for a few minutes, other times they immediately fail to respond (but the links stay green in deconz for 5-10m).

delta9690, 2021-07-08:
[image]
Loading the configuration causes the lights to respond to commands for about 30 seconds after loading is complete before they fall off the network again.

SwoopX, 2021-07-08:
You might wanna check out this one here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3898

delta9690, 2021-07-09:
> You might wanna check out this one here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/3898

Going through those steps (specifically, flashing to an old version and then updating to latest) did make it a bit better but it's far from working properly. I was a ke to pair everything without issue, and it's been grouped, but now if the network sits idle for more than 10 minutes, I have to power cycle a connected device (seems to work on any of them) before it becomes controllable again.

Specifically: 
1. The lights stop responding to commands, but Phoscon/Hue Essentials both behave normally
2. The lights must be set to off in the control software, it does not work from on.
3. Any light can be power cycled (or deConz restarted) and the network becomes responsive again for another 10m or so.

Logs to come when I'm actually awake.

delta9690, 2021-07-11:
Works best when work doesn't call me in for a lethally timed shift.

Anyway, to reiterate:
1. The lights stop responding to commands, but Phoscon/Hue Essentials/deConz all behave as if all devices are connected and working. (if it sits idle long enough, devices slowly start to show up as being unreachable, but it takes about an hour)
2. The lights must be set to off in the control software, it does not work from on.
3. Any light can be power cycled (or deConz restarted) and the network becomes responsive again for another 10m or so.

Should also be mentioned that the Aqara temperature sensor works flawlessly.

And some debug logs:

`18:59:14:381 Device protocol version: 0x010E
18:59:14:590 Current channel 11
18:59:14:629 CTRL got nwk update id 1
18:59:14:642 CTRL ANT_CTRL 0x03
18:59:14:738 CTRL got stack debug assert code: 0x2021, type: 0x03
18:59:15:116 DB UPDATE device_descriptors SET data = x'10400f3511472b0040002b0000', timestamp = 1625965155 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 0 AND type = 2
18:59:15:118 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'10400f3511472b0040002b0000', 1625965155 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
18:59:16:636 DB UPDATE device_descriptors SET data = x'f2e0a164000100012100', timestamp = 1625965156 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 242 AND type = 4
18:59:16:638 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 242, 4, x'f2e0a164000100012100', 1625965156 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
18:59:18:158 DB UPDATE device_descriptors SET data = x'0104010500010300000a00190003010020000005', timestamp = 1625965158 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 1 AND type = 4
18:59:18:160 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010500010300000a00190003010020000005', 1625965158 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
18:59:25:345 0xD0CF5EFFFE1FB447 error APSDE-DATA.confirm: 0xD0 on task
18:59:37:837 0x000B57FFFEF376DA error APSDE-DATA.confirm: 0xD0 on task
18:59:49:107 Current channel 11
18:59:49:120 CTRL got nwk update id 1
18:59:49:133 Device TTL 3565 s flags: 0x7
18:59:49:146 Warning frame counter 3483894 (0x003528F6) lower than previous one 3483904 (0x00352900)
18:59:49:147 Raise frame counter to 3484204 (0x00352A2C)
18:59:49:199 0x14B457FFFE6D5FFB error APSDE-DATA.confirm: 0xD0 on task
19:00:00:440 0x000B57FFFEB25489 error APSDE-DATA.confirm: 0xD0 on task
19:00:00:442 max transmit errors for node 0x000B57FFFEB25489, last seen by neighbors 490 s
19:00:11:740 0x000B57FFFEC0C840 error APSDE-DATA.confirm: 0xD0 on task
19:00:11:741 max transmit errors for node 0x000B57FFFEC0C840, last seen by neighbors 59 s
19:00:13:052 Skip idle timer callback, too early: elapsed 927 msec
19:00:16:632 device state timeout ignored in state 3
19:00:16:791 device state timeout ignored in state 3
19:00:16:951 device state timeout ignored in state 3
19:00:17:056 Device protocol version: 0x010E
19:00:17:133 0x14B457FFFE6D60EF error APSDE-DATA.confirm: 0xA7 on task
19:00:17:273 Current channel 0
19:00:17:312 CTRL got nwk update id 1
19:00:17:325 CTRL ANT_CTRL 0x03
19:00:17:424 CTRL got stack debug assert code: 0x2021, type: 0x03
19:00:17:458 Device protocol version: 0x010E
19:00:17:669 Current channel 11
19:00:17:708 CTRL got nwk update id 1
19:00:17:721 CTRL ANT_CTRL 0x03
19:00:17:819 CTRL got stack debug assert code: 0x2021, type: 0x03
19:00:18:313 DB UPDATE device_descriptors SET data = x'10400f3511472b0040002b0000', timestamp = 1625965218 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 0 AND type = 2
19:00:18:315 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 0, 2, x'10400f3511472b0040002b0000', 1625965218 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
19:00:20:636 DB UPDATE device_descriptors SET data = x'f2e0a164000100012100', timestamp = 1625965220 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 242 AND type = 4
19:00:20:638 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 242, 4, x'f2e0a164000100012100', 1625965220 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
19:00:22:958 DB UPDATE device_descriptors SET data = x'0104010500010300000a00190003010020000005', timestamp = 1625965222 WHERE device_id = (SELECT id FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2') AND endpoint = 1 AND type = 4
19:00:22:960 DB INSERT INTO device_descriptors (device_id, endpoint, type, data, timestamp) SELECT id, 1, 4, x'0104010500010300000a00190003010020000005', 1625965222 FROM devices WHERE mac = '00:21:2e:ff:ff:06:e6:f2'
19:00:29:538 0x000B57FFFED9608B error APSDE-DATA.confirm: 0xD0 on task
19:00:29:540 max transmit errors for node 0x000B57FFFED9608B, last seen by neighbors 204 s
19:00:40:749 0x000B57FFFE9B666C error APSDE-DATA.confirm: 0xD0 on task
19:00:49:107 Current channel 11
19:00:49:120 CTRL got nwk update id 1
19:00:49:134 Device TTL 3568 s flags: 0x7
19:00:52:040 0x90FD9FFFFE90E435 error APSDE-DATA.confirm: 0xD0 on task
19:00:52:042 max transmit errors for node 0x90FD9FFFFE90E435, last seen by neighbors 480 s
[... cut ...]
19:06:53:864 APS-DATA.request id: 115, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0005, ep: 0x00 -> 0x00 queue: 2 len: 3 tx.options 0x00
19:06:53:919 APS-DATA.confirm id: 115, status: 0x00 SUCCESS
19:06:53:921 APS-DATA.confirm request id: 115 -> confirmed, timeout 1625965613
19:06:53:943 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8005, lqi: 5, rssi: 1
19:06:53:945 APS-DATA.indication request id: 115 -> finished
19:06:53:946 APS-DATA.request id: 115 erase from queue`

Smanar, 2021-07-11:
Are you sure for you power supply ?
Have you tried with bluetooth and wifi disabled ?

delta9690, 2021-07-11:
Just tried it with a bunch of different power supplies, didn't help.

Official Raspberry Pi power supply
5v/3a canakit supply
5v/3a phone adapter
5v/4a bench supply

Bluetooth and WiFi are already disabled in config.txt.

closed 2021-08-03 by delta9690