dresden-elektronik/deconz-rest-plugin #5109: Aqara RTCGQ13LM

u20p17, 2021-07-09:
## Device

- Product name: Aqara RTCGQ13LM high precision motion sensor
- Manufacturer: Aqara 
- Model identifier: RTCGQ13LM 
- Device type : 
  - Sensor

## Screenshots

[image]
[image]

### Basic
[image]

### Identify
[image]

### Other clusters that are not mentioned above
Occupancy sensing:
[image]
power configuration:
[image]
OTAU:
[image]

SwoopX, 2021-07-10:
Can you please update occupancy sensing and power configuration cluster screenshots? The values still contain the defaults; you must use the read button and make sure the sensor is awake by triggering its function 2-3 secs prior to reading.

Before you do that, can you please update your firmware? I hope you're using a Conbee II 😉 
[deCONZ_ConBeeII_0x266f0700.bin.zip](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6795994/deCONZ_ConBeeII_0x266f0700.bin.zip)
It has some special magic included which should tickle even more out of the device. After firmware update, please reset and re-pair the device.

u20p17, 2021-07-11:
Hi,
thanks for your answer. I now updated from 0x266d0700 to 0x266f0700. After that I deleted the node and addedit again --> after that I couldn't read in the values with my HomeAssistant deconz running... I now tried to connect the Conbee2 stick to my laptop and there it worked.
node info:
[image]

Basic:
[image]

identify:
[image]

LUMI_specific:
[image]
[image]

occupancy:
[image]

OTAU:
[image]
[image]

power_configuration:
[image]

BR

SwoopX, 2021-07-11:
Great, that helps. Thanks! You you please get me some further info?

For the power configuration cluster, please double click on attribute's 0x0020 value and read the reporting configuration (at the bottom). Do the same for occupancy sensing cluster on attribute 0x0010. I'd like to stick to the defaults here if they're reasonable...

Would you be able to compile the plugin to test the changes?

u20p17, 2021-07-12:
Hi,
good to hear. :) For 0x0020  of the power configuration cluster it worked, but for 0x0010 in occupancy sensing cluster I always get 'reading config failed'.

[image]
[image]

Regarding compilation: I am not the expert in this so I would prefer, if you could do that for me if possible :)

SwoopX, 2021-07-12:
Thanks again. So the reporting for power values are a little bit weak, will set them to report once an hour. The occupancy sensing it a bit odd, that's neither an "I support it" nor an "I don't support it". I guess there'll be no harm anyway, so I wouldn't bother too much.

> Regarding compilation: I am not the expert in this so I would prefer, if you could do that for me if possible :)

As I don't have the device, that's kinda pointless, although I'd love to 😉

u20p17, 2021-07-12:
Thank you for your effort! Can you give ma a short description how I could compile it? With some help from you I should be able to do that :)

u20p17, 2021-07-14:
Hi @SwoopX , not sure if you are waiting for me to compile and test a new version for this or if you are working on this topic and will come back to me when you have finished? (Please do not get me wrong, I don't have that big time pressure with the new sensors and do not want to put you under stress... just that I know when I should do some tests for you again :) )

SwoopX, 2021-07-14:
All good. I'm just not sure how to cater with setting the sensitivity on the sensor. Was hoping for a better answer than I got from the other devs 😉 

Once this is sorted out, I'll give you some guidance on how to compile and test, given that you don't use HA.

u20p17, 2021-07-15:
Hi, thank you very much!

Just du be clear: I am using HomeAssistant with the Conbee2 stick for my system. :)

L3H0, 2021-07-27:
@u20p17 Hi, can you tell me where you purchase this device?

closed 2021-08-02 by manup

u20p17, 2021-08-12:
Hi @SwoopX ,
today I updated my conbee 2 firmware to 26700700 and tried to connect to this sensor with Phoscon, but I wasn't able to connect... Do I also have to update phoscon? If noit, how could I generate a log file which you might need for analyses?

BR

u20p17, 2021-10-01:
Hi @SwoopX ,
After a while I had time again to test the new sensors with HA. Therefore I updated to v6.10.0 (changelog: Bump deCONZ to 2.12.6) and the Conbee2 Stick to 26700700. Then I tried to add the sensors with Phoscon, but without success. I was not able to add them. Furthermore I checked if they get displayed on deconz and since the update I do not get any additionb node there (like with the old software version). What could be the problem? If you need any log file please let me know.
BR

SwoopX, 2021-10-03:
I'd assume you do not correctly reset the device. Also the selected snipped is not really helpful since it's just 9 seconds. What you see there has nothing to do with the motion sensor.

I don't know HA, so when it comes to any related question, I cannot be of any help.

u20p17, 2021-10-04:
ok, I will try again to reset the sensor. Furthermore I could also try to set up another raspi with deconz running on it (without HA) and check if it is working there... 

Regarding the log: what would be helpful for you in order to check the problem (if the same also appears with the secound raspbi...)

Thanks!

7leman7, 2021-10-04:
[image]

I can confirm I was able to delete and successfully add back RTCGQ13LM sensor in deCONZ 2.12.6 running as HA addon. I can finally see the sensitivity settings etc. Will see how it works (before that - when added as generic sensor - it was acting weirdly).

I also have ConBee II, FW 26680700. I was adding the sensor via Phoscon app (it does not appear in Phoscon, only in deCONZ).

u20p17, 2021-10-04:
@7leman7 : Good to hear that at least you can add the sensor in deconz. I now tried it again, but I am not able to do it. Maybe you can describe what you did?

i did it now the following:
1.) removed the batteries for half a minute.
2.) inserted it again and pressed the button in the rear very long (until the blue light started flashing.)
3.) opened Phoscon Menu --> sensors and there I pressed add new sensor an as manufacturer i have chosen 'others'.
4.) after that I pressed again the button on the back until it flashed again but in phoscon it couldn't add it.
5.) After that I opened deconz and also there I can not see the node anymore.

I now also tried a secound (completely new) sensor and there the same is happening.

@SwoopX: Do you have an explanation why @7leman7 can see it in deconz (and if i understand correctly, the sensor is working like expected with the change in sensitivity) but it is not possible for him to add the sensor to phoscon?

7leman7, 2021-10-04:
I did not remove batteries, just deleted the respective node in deconz.

After that I restarted HA, opened Phoscon, started sensor add process (as "other"), pressed the button on the sensor for 5 seconds (sensor blinked 3 times), waited until Phoscon showed add process status as "ready", pressed the button - that's it.

Sensor does not show in Phoscon, but in deCONZ I get new node "Presence **" where ** is the node number. I understand this is the expected behavior.

7leman7, 2021-10-04:
BTW, from what I can see during the last hour the sensor works OK.

u20p17, 2021-10-04:
ok, I now tried with the latest firmware(26720700) and with the 26680700, but with the same result:
[image]

I can not add it in phoscon nor can I see it in deconz. If I look at the debug view can just see the following ( If you check this @7leman7  do you also get those messages? (you can display it by pressing help-debugview)

[image]

SwoopX, 2021-10-04:
You seem to have more than one Phoscon tab open accross all devices at your place. That interferes with opening the network properly.

u20p17, 2021-10-04:
I now closed all phoscon and deconz tabs and then tried to connect again. Same behaviour than in my last post (one hour ago)... 

I update the conbee2 stick always on my windows pc and the reconnect it to the pi + rebooting the whole system - could this somehow be the problem?

7leman7, 2021-10-04:
> ok, I now tried with the latest firmware(26720700) and with the 26680700, but with the same result: [image]
>
> I can not add it in phoscon nor can I see it in deconz. If I look at the debug view can [...]

I don't have the "unhandled APS-DATA" lines, other than that my log looks pretty similar, but I don't really know what I'm looking at :)

Also, I don't have the inclusion timeframe in the logs already (rebooted HA since then).

u20p17, 2021-10-04:
i now found out that all the unhandled APS-DATA error come from a hue smart plug - when I disconnect it and delete the node no errors occur anymore... 
Furthermore I tried to use the conbee 2 with 26720700 and 2.12.06 on my windows pc and there I can connect the sensor and it get displayed as node in Deconz ( in Phoscon I cannot see it - so the same as for @7leman7 - why isn't it also displayed in phoscon as a motion sensor?). Then I deleted the node on the win pc again and connected the conbee2 to my raspberry pi with the same software versions running. Here I can not connect the sensor via phoscon... strange behaviour...

u20p17, 2021-10-04:
so now i was also able to connect the sensor with deconz and phoscon running in HA on my pi. it had the conbee2 connected with an elero centero Transmitter Stick on one usb hub and after disconnecting the elero stick it weas working - maybe the conbee didn't get enough power? However this is working now.

Now only one question is left for me: Why isn't it also displayed in phoscon as a motion sensor? If this is indented, how can I change the name of the node in deconz?

7leman7, 2021-10-07:
Yeah, same for me: after reading "PIR Occupied To Unoccupied Delay" shows 120, config call via rest-api shows duration 60 (also changed to 60 myself after joining the sensor). Maybe I was wring above, but I can say fir sure I did my change via deCONZ, not via rest-api.

Really unsure what's going on, in the meanwhile I was bale to get back all clusters (by numerous "read simple descriptor" calls) except for lumi specific cluster...

7leman7, 2021-10-08:
Okay, so after numerous deletions and inclusions I finally managed to pair the sensor with Lumi Specific Cluster available again.

With that I can change sensitivity attribute in Lumi Specific Cluster and change PIR Occupied to Unoccupied delay to 60 in Occupancy Sensing cluster (writing works, reading back gives 60 as just written).

What I did was press the pairing button for 5 sec (until 3 blinks) and and then pressing the button once a second until paired ("Done" message appearing in Phoscon "Add sensor" menu). Still the process seems to be a bit flimsy as Phoscon showed the sensor to be successfully paired even though it apparently did not recognise it correctly during several previous attempts.