dresden-elektronik/deconz-rest-plugin #5114: Aqara Mini Switch / Button Automations broken with v2.12.2-beta

peer69, 2021-07-13:
## Describe the bug

Updating to v2.12.2-beta broke automations for Aqara Mini Buttons / Switches (maybe even other automations but atm I only use these via Phoscon). ## Steps to reproduce the behavior

Automations (lights on/off, calling scenes) do not work via Aqara Switches when running v2.12.2-beta. Reverting to .1-beta fixes this. 
## Expected behavior

## Screenshots

Switches working as expected with v2.12.1-beta. 
## Environment

 - Host system: ESXi Host
 - Running method: (Ubuntu 18.04)
 - Firmware version: (266E0700)
 - deCONZ version: (2.12.2-beta)
 - Device: (ConBee II)
 - Do you use an USB extension cable: (yes)
 - Is there any other USB or serial devices connected to the host system? No. 

## deCONZ Logs

## Additional context

Mimiix, 2021-07-13:
Can you please provide logs?

peer69, 2021-07-13:
Not yet. As this breaks my wife's current work environment I'll have to wait for updating again to not cause any serious real world issues.

Mimiix, 2021-07-13:
Without any logs or pointers, we can't do anything in this case to fix it.

For anyone facing the same issue and ending up here:  Please open a new issue with corresponding logs.

I'll close this issue.

closed 2021-07-13 by Mimiix