dresden-elektronik/deconz-rest-plugin #5115: Eurotronic Zigbee Thermostat SPZB0001 didn't work with 2.12.2

bruesche, 2021-07-13:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi / PC / NAS)
 - Running method: (Raspbian / Ubuntu / Home Assistent deCONZ Add-on / Marthoc Docker container / Windows / Virtual Machine)
 - Firmware version: (26xxyy00)
 - deCONZ version: (2.xx.yy)
 - Device: (ConBee I / ConBee II / RaspBee I / RaspBee II)
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II
 - Is there any other USB or serial devices connected to the host system? If so: Which?

## deCONZ Logs

## Additional context

SwoopX, 2021-07-14:
It seems that iobroker is sending a string value to the API, but the API expects a numeric value. A check on the data type has been recently introduced in accordance with the API documentation.

bruesche, 2021-07-15:
What can I do? Change the value Type (of it is possible)? And what to do if it is not possible? 

What will be happend with values like 21.5 degress?

thanks for your help!

bruesche, 2021-07-15:
Hi, 

I looked at the value type of heatsetpoint in iobroker. This is numeric. 
[image]

SwoopX, 2021-07-15:
How can you tell if 2500 (numeric) is send instead of "2500" (string) from the screenshot?

bruesche, 2021-07-15:
Thought because the value type is „Zahl“. Thought That „Zahl“ will transfere it as Numerus. You are Right That I didn‘t know how it is transfered. 

So what can I do that my Thermostats work correctly?

SwoopX, 2021-07-15:
I'd recommend to have that checked on the iobroker side to confirm what type of value is send. Other than that, you can of course always directly send to the API as well and check deconz logs for any errors.

bruesche, 2021-07-15:
Oh, now Problem starts. Not an expert in That (Iobroker - where to Check or deconz api - how to check Logs or sent Events)

bruesche, 2021-07-15:
Van you give me advice 

or

is here someone whith same Problem „Thermostat“ and have more knowhow than me

SwoopX, 2021-07-15:
Check out https://dresden-elektronik.github.io/deconz-rest-doc/getting_started/

bruesche, 2021-07-16:
ok, api is running. I can sent values to heatsetpoint over api and the thermostat is working. when I use iobroker (adapter deconz or adapter hue extended) the value heatsetpoint will not be set. The value didn't change. 

Didn't know where to Look for log files. Can you help me?

bruesche, 2021-07-16:
The experiment continues... Found "error" in iobroker log.

deconz.0 | 2021-07-16 14:16:08.886 | warn | (14412) {"address":"/sensors/17/config/heatsetpoint","description":"invalid value, 600, for parameter heatsetpoint","type":7}
-- | -- | -- | --

It is really an invalid value. Now the question "how to solve"...

SwoopX, 2021-07-16:
This error message is indeed thrown when the wrong data type is provided. As this is evidentally not a deconz issue, please seek support from iobroker team on this.

closed 2021-07-16 by SwoopX