dresden-elektronik/deconz-rest-plugin #5117: Aqara smart switch stopped working in beta Оtava Ё

myxor, 2021-07-13:
## Describe the bug

Aqara Smart Switch WXKG11LM Rev.2 stopped working with beta Оtava Ё release. It works with previous release very well.

## Steps to reproduce the behavior

1. Use beta Оtava Ё 
2. Press button on Aqara Smart Switch WXKG11LM Rev.2
3. No event is registered in connected Home-Assistant
4. Switch back to previous release
5. Press button
6. Event in Home-Assistant is registered

## Expected behavior

1. Press button
2. Event in Home-Assistant is registered

## Screenshots

## Environment

 - Host system:  Pine64 RockPro64
 - Running method: Marthoc Docker container
 - Firmware version: 26680700
 - deCONZ version: v2.12.1-beta
 - Device: ConBee II
 - Do you use an USB extension cable: yes
 - Is there any other USB or serial devices connected to the host system? No

## deCONZ Logs

> 20:11:25:772 Node data 0x00158d0002904e7f profileId: 0x0104, clusterId: 0x0012
20:11:25:773 0x00158D0002904E7F: update ZCL value 0x01/0x0012/0x0055 after 0 s
20:11:25:773 Multi state present value: 0x0001 (1), lumi.remote.b1acn01
20:11:25:773 [INFO] - No button map for: lumi.remote.b1acn01, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT (0x0012), command: ATTRIBUTE_REPORT (0x0A), payload: 5500210100, zclSeq: 8
20:11:25:773 ZCL attribute report 0x00158D0002904E7F for cluster: 0x0012, ep: 0x01, frame control: 0x18, mfcode: 0x0000 
20:11:25:773    payload: 5500210100

## Additional context

Mimiix, 2021-07-13:
@SwoopX

Mimiix, 2021-07-14:
**There's more reports of this problem. I've asked @manup to look into it. I'll mark this as confirmed for now, please wait for the outcome.**

canedje, 2021-07-14:
I do have the same problem.
Switched back to 2.12.01.
Is working again

SAOPP, 2021-07-14:
Hm, just checked this after saw discord's message... All works fine at my side. Aqara wireless one/two rockers, with zero line, opple at all my three conbee2 sticks.

Smanar, 2021-07-14:
>0:11:25:773 [INFO] - No button map for: lumi.remote.b1acn01, unicast to: 0x0000, endpoint: 0x01, cluster: MULTISTATE_INPUT 

Can you check your JSON file ? 
For Raspberry it s /usr/share/deCONZ/devices/button_maps.json

You need to have inside

```
        "xiaomiSwitchB1acn01Map": {
            "vendor": "Xiaomi",
            "doc": "WXKG11LM 2018 remote",
            "modelids": ["lumi.remote.b1acn01"],
            "map": [
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "2", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double press"],
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Hold"],
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "255", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Long released"]
            ]
        },
```

rayaar, 2021-07-14:
I have the same issue, and I can confirm that the entry is present in /usr/share/deCONZ/devices/button_maps.json. I have not yet checked the startup-log yet.

myxor, 2021-07-14:
From docker log:

> [...]
14:05:01:416 [INFO] - Found file containing button maps. Parsing data...
14:05:01:435 [INFO] - Button maps loaded.
[...]
4:05:04:004 [INFO] - No button map for: lumi.remote.b1acn01, unicast to: 0x0000, endpoint: 0x01, cluster: BASIC (0x0000), command: 0x0A, payload: 01FF421A0121BD0B03281C0421A81305211600062401000000000A219B89, zclSeq: 40
[...]

Inside the button_maps.json file i can see this JSON element:

```
 "xiaomiSwitchB1acn01Map": {
            "vendor": "Xiaomi",
            "doc": "WXKG11LM 2018 remote",
            "modelids": ["lumi.remote.b1acn01"],
            "map": [
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "1", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "Normal press"],
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "2", "S_BUTTON_1", "S_BUTTON_ACTION_DOUBLE_PRESS", "Double press"],
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Hold"],
                [1, "0x01", "MULTISTATE_INPUT", "ATTRIBUTE_REPORT", "255", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Long released"]
            ]
        },
```
which looks exactly like the one you were looking for.

Remc000tje, 2021-07-14:
same here
In Deconz the switches do work, but no activity in Phoscon, Homebridge or Home assistant.
version 2.12.02. How do I reverse to 2.12.01?

Smanar, 2021-07-14:
>which looks exactly like the one you were looking for.

Yep exactly, but all seem fine ....

SwoopX, 2021-07-14:
Meanwhile @manup confirmed the bug. Apparently, the correct button map is not chosen for yet unknown reason. He'll check it and has the device available for testing.

S474N, 2021-07-15:
> same here
> In Deconz the switches do work, but no activity in Phoscon, Homebridge or Home assistant.
> version 2.12.02. How do I reverse to 2.12.01?

Do you have standard od dockerized version?

Remc000tje, 2021-07-15:
i am running the dockerized version, an let watchtower update the containers to latest.

S474N, 2021-07-15:
As you see, it's not good idea :)

In standard docker version with YML file you can use manually version, what you need:

```
  deconz:
#    image: marthoc/deconz:latest
    image: marthoc/deconz:2.12.01
```

closed 2021-07-15 by manup

Remc000tje, 2021-07-16:
> As you see, it's not good idea :)
> 
> In standard docker version with YML file you can use manually version, what you need:
> 
> ```
>   deconz:
> #    image: marthoc/deconz:latest
>     image: marthoc/deconz:2.12.01
> ```

In Synology Docker: 
- Export container settings (xxxx.json)
- search marthoc/deconz:latest and change to marthoc/deconz:2.12.01
- Stop and delete container
- Import changed xxxx.json
- start container.

oh, and I stopped Watchtower.... :-)

thanks for helping out.