dresden-elektronik/deconz-rest-plugin #5121: Cannot connect Hive Thermostat any more

gee-jay-bee, 2021-07-15:
## Describe the bug

I have a Hive thermostat (SLT3) and Heat / Hot Water Controller (Computime SLR2) that have worked flawlessly for > 6 months with Deconz. This week, I updated to marthoc/deconz:latest and things stopped working. I deleted the two nodes via VNC and then used the Phoscon webapp to try to re-pair. The two nodes both come back into the network (visible in VNC) and talk to each other (i.e. I can control the boiler correctly, the thermostat correctly shows state) but the device is no longer listed in the Phoscon web app under sensors and as a result is no longer available for control via the REST API (i.e. no climate entities in Home Assistant — HA is running on a separate machine).

## Steps to reproduce the behavior
1. Delete nodes
2. Factory reset receiver and thermostat
3. Put Deconz in pairing mode (search for sensors on webapp)
4. Re-pair receiver and thermostat

## Expected behavior
1. Thermostat listed in WebApp under sensors
2. Entity exposed via REST API

## Screenshots
I have one, but I can’t attach .. will submit and try on a different machine

## Environment

 - Host system: Raspberry Pi 4
 - Running method: Raspbian / Marthoc Docker container
 - Firmware version: 26390500
 - deCONZ version: attempted pairing with multiple versions: 2.11.05 (marthoc/deconz:stable), 2.12.02 (marthoc/deconz:latest) and an older version (2.09.03)
 - Device: RaspBee I
 - Do you use an USB extension cable: (yes / no) -- only relevant for ConBee I/II — N/A
 - Is there any other USB or serial devices connected to the host system? If so: Which? No

## deCONZ Logs

## Additional context
I can read the various clusters on the nodes via VNC and responses come back and look sane.

gee-jay-bee, 2021-07-15:
[image]

Smanar, 2021-07-15:
Effectively they are no more reconized by the API ....
Can you show again the basic attribute (the SL3 was working too ? I don't see it the code), cluster 0x0000, the attribute table on the bottom (you probably need to press "read" and make the device awake)

And if you can the log with "info" and "info_l2" during the inclusion try (I will need it after if all seem fine on basic cluster).

gee-jay-bee, 2021-07-15:
Apologies if this is the wrong info, I took it for both nodes:

[image]
[image]

gee-jay-bee, 2021-07-15:
Yes, the two devices had been working together (I watched progress made around #2979 last year and bit the bullet and it “just worked”)

Smanar, 2021-07-15:
Ok so the slt3 is not in deconz, so if you pair the 2 device one with the other can explain your issue.
If you try to include only the slr2, it don't work ?

gee-jay-bee, 2021-07-15:
By working together, I thought I had them both paired with deconz but seems not.

Before it went wrong, I could control heat and hot water both from HA and from the SLT3. At one point, the battery on that device died (late winter) and things stopped working via HA. Only by repairing the SLT3 did things start working again!

I think you’re suggesting that I try to pair just the SLR2 and leave the SLT3 powered off and reset. Assuming I see the SLR2 appear as heat + hot water in deconz, I could then try adding the SLT3 to the SLR2 and see what happens?

gee-jay-bee, 2021-07-15:
Tried it, success!

The SLR2 paired and as soon as I did a read on cluster 0000, it was recognised as a thermostat.

I then powered up the SLT3 and it joined the network and can now control the SLR2 whilst the hot water and heat are now exposed to HA.

I guess deconz needed to read something from the SLR2 before the SLT3 connected to claim it somehow?

Smanar, 2021-07-15:
I think it s like for bulb.
- include the bulb
- include the remote with the reset procedure
- pair them using deconz.

Work, but if you do

- include the bulb
- pair the remote with the bulb

Don't work

If you pair the bulb with the remote using the pairing procedure, they leave the previous network to pair each other.

>I guess deconz needed to read something from the SLR2 before the SLT3 connected to claim it somehow?

Yes, it's possible too, reading the basic attribute  can trigger (or finish) an inclusion, if phoscon is still in "add new device"

BTW the SLT3 is not in the API, so I don't think you can see it in HA, do you need to include it too in the API ?

gee-jay-bee, 2021-07-15:
Understood, thanks for the explanation.

I'm not sure exactly extra would be exposed to HA by making deconz aware of the device. I can't do things like set schedule or boost but I manage all that in HA anyway. It might be handy to know the battery level if available.

Is it easy to do?

Smanar, 2021-07-16:
Can try.

If I look the cluster, you can have 
Cluster 0x0001 > the battery level
Cluster 0x0201 > a ZHAthermostat
Cluster 0x0402 > A ZHATemperature

I just need a capture with the attribute table on this cluster (to see wich one are used or not)

gee-jay-bee, 2021-07-16:
[image]

gee-jay-bee, 2021-07-16:
[image]

gee-jay-bee, 2021-07-16:
[image]

Smanar, 2021-07-16:
You don't have more usefull attributes on thermostat cluster ? like the 0x0012 ?

BTW I haven't asked first, but have you a linux machine with full OS to test code modifications ?

gee-jay-bee, 2021-07-16:
Oh yeah, good catch. Lots of extra stuff in there but the table isn't very large. Is there a way to expand or do I need to screenshot and scroll?

Yes to Linux environment ... Is it ok to mount into docker or do I need to run native on the pi?

gee-jay-bee, 2021-07-16:
[image]
[image]
[image]
[image]

gee-jay-bee, 2021-07-16:
There's some AC related stuff too but it doesn't look to be set (which makes sense given that Hive doesn't have air-con :))

Smanar, 2021-07-17:
Ok so easy for me ^^, have just mimic the stl2 code and add slt3. So if can't test it, I think it s possible to do it in blind mode
It can't work on docker OS, you need a real one. (it s possible in reality on docker, but more complicated, https://github.com/dresden-elektronik/deconz-rest-plugin/issues/4841#issuecomment-852157541 )

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
So for you, after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch slt3 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

gee-jay-bee, 2021-07-17:
Ok, mixed results I'm afraid.

On the positive side, deconz recognised the device. I used the docker info you shared, it went smoothly apart from a docker build error trying to apt install deconz which I was able to fix. All other functionality seemed fine and HA was able to connect etc.

Unfortunately, things went back to almost how they were when I raised the bug and whilst the receiver and thermostat could talk, HA lost the ability to send commands via the REST API. I attempted re-pairing multiple times using the initial read of the cluster to get deconz to recognise the SLR2 as the thermostat and then adding the SLT3. The only difference in observed behaviour was that 1 thermostat was listed on the web app (whereas I saw nothing in my original report).

However, I then went back to the standard docker image (marthoc/deconz:latest) and experienced exactly the same problem. I tried this numerous times, trying to change the timing of pairing but it made no difference. 

Eventually I remembered that I wasn't running latest when I got things working again but I was running marthoc/deconz:stable. As soon as I switched over to stable, everything started working again (using the pairing from latest!). For fun, I performed a full factory reset on the Hive equipment and re-paired against stable ... all good. I then switched to latest and it stopped working. Switching back to stable worked again.

Looks like I'm sticking with stable for now ...

My feeling is that your change probably would work if it were applied to stable as the base but because latest is broken, it also doesn't work.

Smanar, 2021-07-18:
But my changes was on the lastest version.
Wich one version is the stable used in marthoc/deconz:stable ?

But I realy can't help you using docker, it s not so easy than classic OS.

So to resume on the "official docker image" it works, but not using my changes ?

gee-jay-bee, 2021-07-18:
I'm comfortable with docker so I can translate from native OS as needed.

I built your change against latest (2.12) because I wasn't sure that I could build your changes against stable (2.11). Is it possible to work out which version of the deconz-dev package I need? If yes, I can mod my environment to compile your change against that (I think).

Results:

- 2.12 with your change: recognises both devices as thermostats, both listed in web app under sensors, can't control from HA via REST API.
- 2.12 without change: recognises SLR2 not SLT3 (as expected) but cannot control from HA via REST API
- 2.11: as 2.12 but control from HA is working

I think your hunch about the new validation in 2.12 is probably right so I will raise a bug in the HA integration (I assume it is sending with wrong data type). Interestingly, the state is updated in HA (wrongly -- I ask it to heat the request fails but it sets the state to "heat") so it seems to ignore failed requests.

Ultimately, this was probably the issue that originally tripped me up and I then got in a mess with pairing which I didn't need to do!

Smanar, 2021-07-18:
@peterguy04 if you are still here ^^ ? I think it's something for you.

You have a way to acess deconz logs in ame time ? I think if you enable "http", you will see the error message send by the API.

>Interestingly, the state is updated in HA

Yes, it s because state are updated by the websocket, it s the classic request that have been updated recently.

gee-jay-bee, 2021-07-18:
Whilst I figure out how to enable dbg-http, I increased the logging on the latest docker image and sent commands from HA to switch the hot water on and off. I then used grep SLR2 to just look for anything interesting for that device:

`deconz    | 16:51:25:585 [WARNING] - Missing cluster in sensor fingerprint: 0x001E5E09020BEEC4 - 0x0832 (SLR2), endpoint: 0x05, cluster: 0x000A, payload: 0100, zclS
eq: 45
deconz    | 16:51:25:587 [INFO] - No button map for: SLR2, unicast to: 0x0000, endpoint: 0x05, cluster: 0x000A, command: 0x00, payload: 0100, zclSeq: 45
deconz    | 16:51:29:313 [INFO] - No button map for: SLR2, unicast to: 0x0000, endpoint: 0x06, cluster: 0x0201, command: 0x0A, payload: 1200299808, zclSeq: 46
deconz    | 16:51:35:594 [WARNING] - Missing cluster in sensor fingerprint: 0x001E5E09020BEEC4 - 0x0832 (SLR2), endpoint: 0x05, cluster: 0x000A, payload: 0100, zclS
eq: 47
deconz    | 16:51:35:594 [INFO] - No button map for: SLR2, unicast to: 0x0000, endpoint: 0x05, cluster: 0x000A, command: 0x00, payload: 0100, zclSeq: 47
deconz    | 16:51:45:630 [WARNING] - Missing cluster in sensor fingerprint: 0x001E5E09020BEEC4 - 0x0832 (SLR2), endpoint: 0x05, cluster: 0x000A, payload: 0100, zclS
eq: 48
deconz    | 16:51:45:631 [INFO] - No button map for: SLR2, unicast to: 0x0000, endpoint: 0x05, cluster: 0x000A, command: 0x00, payload: 0100, zclSeq: 48
deconz    | 16:51:49:319 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('108', 'Thermostat', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-05-0201', '20190604', '{"lastupdated":"2021-07-18T15:51:17.680",
"on":false,"temperature":3018}', '{"heatsetpoint":2100,"mode":"off","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep":5,"i
n":[513],"p":260}', 'normal', '1')
deconz    | 16:51:49:325 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('109', 'Thermostat 109', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-06-0201', '20190604', '{"lastupdated":"2021-07-18T15:49:42.7
36","on":false,"temperature":2100}', '{"heatsetpoint":2200,"mode":"heat","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep"
:6,"in":[513],"p":260}', 'normal', '1')
deconz    | 16:51:55:692 [WARNING] - Missing cluster in sensor fingerprint: 0x001E5E09020BEEC4 - 0x0832 (SLR2), endpoint: 0x05, cluster: 0x000A, payload: 0100, zclS
eq: 49 `

Does that DB sql line indicate that the command to turn on ("mode:heat") and off ("mode:off") are making it into deconz?

To confirm, the device didn't actually change state when this happened.

gee-jay-bee, 2021-07-18:
Not seeing any warnings about http in the log coming out of the docker container. 
Using grep heatsetpoint gives me this ... basically the dB update from before + action on the WebSocket.

Am I reading this correctly that HA is reflecting the state that Deconz is advertising? Just want to be sure I understand before I raise a bug on HA! 

`deconz    | 17:08:18:043 enqueue event config/heatsetpoint for /sensors/108
deconz    | 17:08:18:047 Websocket 192.168.18.180:44334 send message: {"config":{"heatsetpoint":2100,"mode":"off","offset":0,"on":true,"reachable":true,"schedule_on
":false},"e":"changed","id":"108","r":"sensors","t":"event","uniqueid":"00:1e:5e:09:02:0b:ee:c4-05-0201"} (ret = -1094660568)
deconz    | 17:09:14:238 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('108', 'Thermostat', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-05-0201', '20190604', '{"lastupdated":"2021-07-18T16:06:19.836",
"on":false,"temperature":3005}', '{"heatsetpoint":2100,"mode":"off","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep":5,"i
n":[513],"p":260}', 'normal', '1')
deconz    | 17:09:14:255 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('109', 'Thermostat 109', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-06-0201', '20190604', '{"lastupdated":"2021-07-18T16:09:11.5
66","on":false,"temperature":2100}', '{"heatsetpoint":2200,"mode":"off","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep":
6,"in":[513],"p":260}', 'normal', '1')
deconz    | 17:10:05:003 Websocket 192.168.18.180:44334 send message: {"config":{"heatsetpoint":2200,"mode":"heat","offset":0,"on":true,"reachable":true,"schedule_o
n":false},"e":"changed","id":"109","r":"sensors","t":"event","uniqueid":"00:1e:5e:09:02:0b:ee:c4-06-0201"} (ret = -1094660568)
deconz    | 17:10:20:224 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('108', 'Thermostat', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-05-0201', '20190604', '{"lastupdated":"2021-07-18T16:09:45.327",
"on":false,"temperature":3005}', '{"heatsetpoint":2100,"mode":"off","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep":5,"i
n":[513],"p":260}', 'normal', '1')
deconz    | 17:10:20:238 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('109', 'Thermostat 109', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-06-0201', '20190604', '{"lastupdated":"2021-07-18T16:09:45.5
01","on":false,"temperature":2100}', '{"heatsetpoint":2200,"mode":"heat","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep"
:6,"in":[513],"p":260}', 'normal', '1')
deconz    | 17:10:24:812 Websocket 192.168.18.180:44334 send message: {"config":{"heatsetpoint":2200,"mode":"off","offset":0,"on":true,"reachable":true,"schedule_on
":false},"e":"changed","id":"109","r":"sensors","t":"event","uniqueid":"00:1e:5e:09:02:0b:ee:c4-06-0201"} (ret = -1094660568)
deconz    | 17:11:20:238 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('109', 'Thermostat 109', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-06-0201', '20190604', '{"lastupdated":"2021-07-18T16:09:45.5
01","on":false,"temperature":2100}', '{"heatsetpoint":2200,"mode":"off","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep":
6,"in":[513],"p":260}', 'normal', '1')
deconz    | 17:12:20:209 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState
, mode) VALUES ('108', 'Thermostat', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-05-0201', '20190604', '{"lastupdated":"2021-07-18T16:11:20.509",
"on":false,"temperature":3008}', '{"heatsetpoint":2100,"mode":"off","offset":0,"on":true,"reachable":true,"schedule":null,"schedule_on":false}', '{"d":769,"ep":5,"i
n":[513],"p":260}', 'normal', '1') `

Smanar, 2021-07-18:
`17:08:18:047 Websocket 192.168.18.180:44334 send message: {"config":{"heatsetpoint":2100,"mode":"off","offset":0,"on":true,"reachable":true,"schedule_on ":false},"e":"changed","id":"108","r":"sensors","t":"event","uniqueid":"00:1e:5e:09:02:0b:ee:c4-05-0201"} (ret = -1094660568) deconz | 17:09:14:238 DB sql exec REPLACE INTO sensors (sid, name, type, modelid, manufacturername, uniqueid, swversion, state, config, fingerprint, deletedState , mode) VALUES ('108', 'Thermostat', 'ZHAThermostat', 'SLR2', 'Computime', '00:1e:5e:09:02:0b:ee:c4-05-0201', '20190604', '{"lastupdated":"2021-07-18T16:06:19.836", "on":false,"temperature":3005}', `

This is when deconz send websocket request to update client

But you can't change the value yourself using HA, so the problem is in the other side, it s the request make with HA that is probably not good, for exemple if it send "3200" instead of 3200.

Deconz send error message when the request is not good, but IDK HA to help you to find the good log, you haven't access to the deconz GUI ?

`17:10:05:003 Websocket 192.168.18.180:44334 send message: {"config":{"heatsetpoint":2200,"mode":"heat","offset":0,"on":true,"reachable":true,"schedule_o n":false},"e":"changed","id":"109","r":"sensors","t":"event","uniqueid":"00:1e:5e:09:02:0b:ee:c4-06-0201"}`

`17:10:24:812 Websocket 192.168.18.180:44334 send message: {"config":{"heatsetpoint":2200,"mode":"off","offset":0,"on":true,"reachable":true,"schedule_on ":false},"e":"changed","id":"109","r":"sensors","t":"event","uniqueid":"00:1e:5e:09:02:0b:ee:c4-06-0201"}`

But the value have moved too in deconz at a moment ?

gee-jay-bee, 2021-07-18:
I don't understand why the state is changing. I can only request the change from HA (nothing I can change in the phoscon web app). Is it just coincidence that the node is sending new state to deconz which then refreshes the web socket clients?

I can access the phoscon GUI via VNC, should I try to turn on heat on one of the clusters?

Smanar, 2021-07-18:
Phoscon don'y support thermostat.

> I can only request the change from HA

But it seem working, as on my previous link we can see a "mode" change ?

>Is it just coincidence that the node is sending new state to deconz which then refreshes the web socket clients?

There is no refresh, the websocket connexion is permanent

>I can access the phoscon GUI via VNC, should I try to turn on heat on one of the clusters?

You mean deconz GUI ? No, if you change it using the application, you are bypassing the API (but you will have at least the return)
But on the app go at help / debug view, with the flag "http", you will see HTTP requests between HA and the API. So if you make a bad request using HA (when trying to change the mode), you will see the error message.

gee-jay-bee, 2021-07-18:
> But it seem working, as on my previous link we can see a "mode" change ?

No, it isn't working. The SLR2 unit on the wall doesn't change state. Everything else in the chain (deconz / web socket / HA) seems to think it has. This is the only difference between stable and latest -- the boiler doesn't actually fire!

> But on the app go at help / debug view, with the flag "http", you will see HTTP requests between HA and the API. So if you make a bad request using HA (when trying to change the mode), you will see the error message.
> 
> 

Ok I can look at this but do I need to or is the issue deeper in deconz?

peterguy04, 2021-07-18:
hey @Smanar , mine are on stable right now and works well i tried a re-pair tonight and no issues, sounds like its only broken in new beta? maybe i should avoid it for now lol i mean i could fire up another container to test it but it wouldnt be till next weekend as i am back at work tomorrow. 

method of connecting thats been near flawless so far (other than a sensor search and read basic occasionally): remove batteries out of SLT , put SLR in pair mode and start a phoscon sensor search , when it changes from double amber flash to single flash, start another sensor search , put the SLT near the SLR and then put the SLT in pair mode, you need the SLR to go green.

tbh the SLT is useless in home assistant , its serves no real purpose but the SLR requires it to function as it uses it as the temp sensor. ALL operation is done via SLR only via API

Smanar, 2021-07-19:
So SLT2 and SLT3 are used only for reading ? Can't be used to send request ?
But the SLR need to work ?

Still waiting someone on the version 2.12 to be sure.

gee-jay-bee, 2021-07-19:
Right.

- If the SLR2 is paired with deconz, it will not respond to commands via the REST API unless the SLT3 is paired with it (lights on the SLR2 flash orange until SLT3 joins and then it turns green and is controllable)
- When I was manually reading the clusters on the SLT3, it seems that the SLR2 reads the temperature measurement from the SLT3 and reports that in its own temperature measurement cluster
- the only functionality I really expected to gain from the SLT3 being known to deconz is the battery level as I guessed the boost buttons for heat and hot water aren't generic

Let me know if you need more logging out of latest, I can easily switch between images.

Smanar, 2021-07-20:
Ha ok so to resume the SLR2 NEED a SLTX to read values/mode, can't work alone

>I did try to send a heat command to the SLT3 from HA when it appeared using your branch but nothing happened. However, this was against latest so I don't actually know for sure if this is expected or not!

I think it s not possible, try to send it to the SLR2, this one is an active device, more chance for it is able to receive command (at opposite of sensor device).

gee-jay-bee, 2021-07-20:
Yeah. In fact, when the batteries died on my SLT3 over the winter, the SLR2 stopped working and we had no heating or hot water that morning so being able to alert on the battery level in HA would be very handy 😀

Smanar, 2021-07-20:
So It can work with last changes.
Just need to make the API working if you use the SLR2 to send command . This one is not controlable using HA ?

gee-jay-bee, 2021-07-20:
The SLR2 cannot be controlled with 2.12 (latest).
Downgrading to 2.11 (stable) is working fine.

From HA's perspective, both behave the same way it's just that the device doesn't actually do what it says it is doing on 2.12.

If there's a specific scenario / log you want me to try with 2.12, let me know.

Smanar, 2021-07-21:
> The SLR2 cannot be controlled with 2.12 (latest).
> Downgrading to 2.11 (stable) is working fine.

This is exactly my problem, I can't say if the problem is from deconz or not.

Do you know how to use the API ?
Else I realy need the request made by HA, and the answer from deconz, I think this is visible somewhere in HA log, but it s too visible on deconz, on the application on help/debug view with the flag "http"

gee-jay-bee, 2021-07-21:
Preconditions:

- Boiler is firing, SLR2 green light shows boiler is firing
- Running stable image

Test:

- Upgrade to latest image
- Use HA to switch boiler on and off
- check that SLR2 does not respond to commands

Results:

Boiler continues to fire for the duration of the test.

Extracted from WebApp, API debugger (WebSocket view):

{
    "17:12:01:243": {
        "config": {
            "heatsetpoint": 2200,
            "mode": "off",
            "offset": 0,
            "on": true,
            "reachable": true,
            "schedule_on": false
        },
        "e": "changed",
        "id": "109",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:1e:5e:09:02:0b:ee:c4-06-0201"
    },
    "17:12:11:458": {
        "config": {
            "heatsetpoint": 2200,
            "mode": "heat",
            "offset": 0,
            "on": true,
            "reachable": true,
            "schedule_on": false
        },
        "e": "changed",
        "id": "109",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:1e:5e:09:02:0b:ee:c4-06-0201"
    },
    "17:12:48:475": {
        "config": {
            "heatsetpoint": 2200,
            "mode": "off",
            "offset": 0,
            "on": true,
            "reachable": true,
            "schedule_on": false
        },
        "e": "changed",
        "id": "109",
        "r": "sensors",
        "t": "event",
        "uniqueid": "00:1e:5e:09:02:0b:ee:c4-06-0201"
    },
    "17:13:08:521": {
        "e": "changed",
        "id": "109",
        "r": "sensors",
        "state": {
            "lastupdated": "2021-07-21T16:13:08.556",
            "on": true,
            "temperature": 2100
        },
        "t": "event",
        "uniqueid": "00:1e:5e:09:02:0b:ee:c4-06-0201"
    }
}
                    
Extracted from Deconz log:

17:15:29:577 HTTP API PUT /api/2AB1C4DEF7/sensors/109/config - 192.168.x.x
17:15:29:578 Text Data: 	{"mode": "heat"}
17:15:29:578 ApiMode: 0
17:15:29:579 [{"success":{"/sensors/109/config/mode":"heat"}}]
17:15:35:216 HTTP API PUT /api/2AB1C4DEF7/sensors/109/config - 192.168.x.x
17:15:35:216 Text Data: 	{"mode": "off"}
17:15:35:217 ApiMode: 0
17:15:35:217 [{"success":{"/sensors/109/config/mode":"off"}}]

I can find no useful logging out of HA during the test.

gee-jay-bee, 2021-07-21:
I repeated the test on stable, all controlled as expected.

Obviously, no debugger on the WebApp.

Deconz log is the same:

17:23:43:059 HTTP API PUT /api/2AB1C4DEF7/sensors/109/config - 192.168.x.x
17:23:43:060 Text Data: 	{"mode": "heat"}
17:23:43:060 ApiMode: 0
17:23:43:061 [{"success":{"/sensors/109/config/mode":"heat"}}]
17:23:49:649 HTTP API PUT /api/2AB1C4DEF7/sensors/109/config - 192.168.x.x
17:23:49:649 Text Data: 	{"mode": "off"}
17:23:49:650 ApiMode: 0
17:23:49:650 [{"success":{"/sensors/109/config/mode":"off"}}]

Smanar, 2021-07-21:
> 17:23:43:061 [{"success":{"/sensors/109/config/mode":"heat"}}]
> 17:23:49:650 [{"success":{"/sensors/109/config/mode":"off"}}]

So for both you have success messge, but in one the srl2 don't answer and the other it's working ? So the problem is not from HA ....

gee-jay-bee, 2021-07-21:
I can't say if deconz doesn't ask or SLR2 doesn't answer but I agree that HA seems to be ok.

Do I need to raise bug this somewhere else if it isn't the REST plug-in causing the issue?

Smanar, 2021-07-21:
It can be the RESt plugin ^^, don't worry you are at good place for the moment.
There is so much change on the last version 1.12 compared to 1.11, hard to check all the changes ...

Can you show some logs from deconz with "info" "info_l2" "error" "error_l2" and "aps" when making the same command with the bugged version (1.12) ?

gee-jay-bee, 2021-07-21:
[with-log-levels.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/6859018/with-log-levels.log)

gee-jay-bee, 2021-07-21:
Setting the heat point on the water thermostat doesn't do anything (in reality, it is just a water heater so only does on/off AFAICT).

Knowing whether the heat point has been set on the heating thermostat is a bit hit and miss because the SLT3 sometimes reports NO SIGNAL (I don't tend to use it so I don't know if this is normal or not).

- The SLT3 displays a snowflake if the heating is off so manually put heating to active on the SLT3 (this command doesn't work from HA)
- When I set the heat point and the SLT3 doesn't report NO SIGNAL, it does show that the heat point has been updated by HA
- The log shows success

> 23:02:55:868 [{"success":{"/sensors/108/config/heatsetpoint":1600}}]
23:02:55:898 HTTP API PUT /api/2AB1C4DEF7/sensors/108/config - 192.168.x.x

FWIW -- the SLT3 also shows the current temperature and this is reflected correctly back to HA on the SLR2's temperature measurement cluster.

I'd say it is only the mode selection that is failing.

Smanar, 2021-07-22:
Honnestly, I don't have thoses device, and the working mode is not easy to understand, it's for that I have asked to peterguy04 ^^.
But for zigbee side :
- Better to send request to the SLR2, not the SLT, the SLR2 is a router so It can receive all request, SLTX are end device, and lot of them are lazzy devices. It seem the SLR2 can update itself the SLTX after some time.
- the "success" message just mean the request have been send, not sure the device have received it.

So the "heatsetpoint" request work with HA (if send to the slr2) but not the "mode" ? thx, will be easier to check in the code.

gee-jay-bee, 2021-07-22:
To be clear, I'm not sending anything to the SLT3 from HA. Where I use the SLT3 above, it is on the unit itself (it has buttons and a dial). With 2.12, that's the only way to make the mode change to do the test.

Smanar, 2021-07-22:
NP, Have found the problem.
The code is now updated on the same branch, if you can update to test the last code.

And BTW tell me if the battery state is working ^^ ?

gee-jay-bee, 2021-07-23:
I built a new image with the change but I got a segfault just after starting the container (deconz initialised and I saw logging about startup).

I reverted for now but will rebuild / retest later.

Smanar, 2021-07-23:
Can be the new line

`attributeList.insert(THERM_ATTRID_SYSTEM_MODE, (quint32)match.value);`

But the synthax don't look so bad, but I have just tested the compilation on my side.

gee-jay-bee, 2021-07-23:
It was an error on my side (built against 2.12 deconz-dev package but ran against 2.11 docker).

Just got it running properly in a test docker image and control of hot water is now working again. I followed your instruction to permit join in deconz and then read the basic clusters on the SLT3 and it recognised the device. It shows up in HA, including the battery level. 

I will do some more testing tomorrow (including enabling the heating) just to make sure all seems ok after a day of real use and report back but this looks promising. 😀

gee-jay-bee, 2021-07-24:
Everything seems to be working OK.

The battery level is reported -- it's 100%, so only time will tell how often it updates!

FWIW -- I forked the repo you shared with me for testing with docker and got it working with the RPI 4 so that it can build against either latest or stable as the base docker image. In principle, it could target specific packages but I didn't have the need.

Thanks for your help on this :)

gee-jay-bee, 2021-07-24:
https://github.com/gee-jay-bee/docker-deconz-rest-plugin-test

Smanar, 2021-07-24:
" Allow switching between latest and stable docker image" ?
I don't use docker on my side so I don't relay understand all thoses stuff, but there is so much user using dockers.

closed 2021-08-10 by manup