dresden-elektronik/deconz-rest-plugin #5124: Conbee II Basic Cluster info says it is ZCL 2

RMCob, 2021-07-15:
## Describe the bug
The Basic Cluster info for the Conbee II indicates that it is ZCL 2

## Steps to reproduce the behavior
 1 - Select the Gateway
 2 - Select the Basic cluster
 3 - Press 'Read'

## Expected behavior
  ZCL should be 3

## Screenshots

## Environment
  Host system: Raspberry Pi 4
  Running method: Raspian
  Firmware version: 2666f0700
  deCONZ version: 2.12.02 (beta)
  Device: ConBee II
  Do you use an USB extension cable: yes
  Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

manup, 2021-07-20:
Do you mean the "ZCL Version" attribute? According to the current specification it should be set to `2`.

https://www.nxp.com/docs/en/user-guide/JN-UG-3115.pdf

https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf

RMCob, 2021-07-20:
Sorry for the confusion. I thought the ZCL Version meant the Zigbee version the device supports, which is 3.0.

closed 2021-07-26 by RMCob