dresden-elektronik/deconz-rest-plugin #5125: On/off button of Ikea remote (5 buttons) isn't working

ChristianPorsche, 2021-07-15:
## Describe the bug

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment

 - Host system: (Raspberry Pi 4)
 - Running method: (Raspbian)
 - Firmware version: (26660700)
 - deCONZ version: (2.11.05)
 - Device: (ConBee II )
 - Do you use an USB extension cable: (Yes) 
 - Is there any other USB or serial devices connected to the host system? (No)

## deCONZ Logs

## Additional context

ChristianPorsche, 2021-07-15:
I see on this page that there is a fix release in the beta software. https://phoscon.de/en/changelog/

How could I install the beta?

Smanar, 2021-07-15:
On a Raspberry pi, if you have used the package method (you need to change the branch, I think you are on the stable one,  https://www.phoscon.de/en/conbee2/install#raspbian)
`sudo apt install deconz`

Else just instal the file http://deconz.dresden-elektronik.de/raspbian/beta/

moschnetwork, 2021-07-17:
hey guys,

thank you for this great work.

I got a problem with my Ikea Tradfri (5 Buttons / Round) Remotes:

Every time if I press the power button for more than 3 seconds (1001), a short press (1002) is no longer possible.
The same applies to the pairing of the remote controls in the Phoscon app. Here I see that the Ikea Remote has been recognized, in the next step I should press the power button for 10 seconds ... After pairing, however, the short press 1001 is no longer recognized.

However, if I omit the 10 seconds press during the Phoscon app pairing process and close the popup window, but only then does the short press 1002 work. As soon as the long power button is pressed (1001) then the short press (1002) unfortunately no longer works.

readjust the process with all my 20 Ikea-Tradfri Remote to Control Sonos. Thank you guys

moschnetwork, 2021-07-17:
update to beta and now the 1001 Bug is fixed. Thank you guys ;)

ChristianPorsche, 2021-07-17:
> update to beta and now the 1001 Bug is fixed. Thank you guys ;)

@moschnetwork how did you installed the beta?

SwoopX, 2021-07-17:
Duplicate of #4900 and resolved.

closed 2021-07-17 by SwoopX