dresden-elektronik/deconz-rest-plugin #5129: Deconz package not found on raspberry debian

xiki808, 2021-07-17:
## Describe the question or issue you are having
I have debian buster on raspberry pi 4 and I cannot install deconz apt package on the system, it gives me: E: Unable to locate package deconz

I am following the Raspbian steps as per the Installation steps. I have tried same steps on raspbian and I had no problems.

What can I do to install deconz on debian on raspberry pi?

## Screenshots
## Environment
 - Host system: Raspberry Pi
 - Running method: on raspberry Debian
 - Device: ConBee II

## deCONZ Logs
## Additional context

Smanar, 2021-07-17:
In waiting for an update, can try this link perhaps https://deconz.dresden-elektronik.de/debian/ ?

xiki808, 2021-07-17:
Thank you this worked!

closed 2021-07-17 by xiki808