dresden-elektronik/deconz-rest-plugin #5130: Nedis Smart Door or Window Sensor pairs, but stays 'unreachable'.

brujoand, 2021-07-17:
## Describe the bug
After adding the sensor through phoscon the device show up but is not reachable, and no further communication betwoon deconz and the devices seems to happen.

## Steps to reproduce the behavior
1. Hold the pairing button on the sensor for 5 seconds
2. Initiate pairing on the conbee
3. Smile as it pairs
4. Cry as it stays in the state "Not reachable"

## Expected behavior
I would expect a device that has successfully paired to stay reachable

## Screenshots
No screenshots, but here is a dump from the API

```json
  "36": {
    "config": {
      "battery": 100,
      "enrolled": 0,
      "on": true,
      "pending": [],
      "reachable": false
    },
    "ep": 1,
    "etag": "6c2c9488fb768c2f9034e0fe742bf8de",
    "lastseen": null,
    "manufacturername": "Tuyatec",
    "modelid": "RH3001",
    "name": "Verandadør",
    "state": {
      "lastupdated": "none",
      "lowbattery": null,
      "open": null,
      "tampered": null
    },
    "type": "ZHAOpenClose",
    "uniqueid": "68:0a:e2:ff:fe:f1:f7:f4-01-0500"
  },
```

## Environment

 - Host system: PC
 - Running method:  Marthoc Docker container in Kubernetes
 - Firmware version: 26660700
 - deCONZ version: 2.12.03 / 12/07/2021
 - Device: ConBee II 
 - Do you use an USB extension cable: yes
 - No

## deCONZ Logs
`11:34:32:089 [INFO] - No button map for: RH3001, unicast to: 0x0000, endpoint: 0x01, cluster: 0x0001, command: 0x0A, payload: 2000201E210020C8, zclSeq: 18`

Pretty much just a bunch of these

## Additional context
Around 30 devices, mostly ikea lights and switches.

Smanar, 2021-07-17:
Ha yes, the enrollement have not worked for this device.

>"enrolled": 0,

Can you show some logs with the flag "IAS_ZONE", it can work too after some time.

brujoand, 2021-07-17:
Funny thing,about an an hour ago I walked over to the sensor and pressed the button once. Just to check that it actually is on, blinks or anything. Immediately it started working. And now everything is still functioning as expected. Very weird, but a simple solution I guess.

Smanar, 2021-07-17:
It can be normal ^^.
The enrollement process need lot of request, and as the device is a lazy one that go to sleep, can take some time if it fail during inclusion.

closed 2021-07-17 by brujoand