dresden-elektronik/deconz-rest-plugin #5138: CASA.ia Temperature probe CTHS-317-ET

arenoux, 2021-07-19:
## Device

- Product name: Temperature Probe CTHS-317-ET
- Manufacturer: CASAIA
- Model identifier: CTHS317ET
- Device type : Please remove all unrelated device types. 
  - Sensor

## Screenshots

### Node INFO

[image]

### Basic
[image]

### Identify

[image]

### Device Temperature

[image]

### Other clusters that are not mentioned above

###### Power configuration
[image]

SwoopX, 2021-07-20:
Please read the attributes of the power configuration cluster and update the screenshot accordingly, as this is required  to process the battery values appropriately.

arenoux, 2021-07-20:
Hello @SwoopX.
I've just updated the Power Configuration and Temperatur Measurement clusters section with a screenshot (after reading the values)

Smanar, 2021-08-09:
Hello.

@SwoopX have you started something ?
@arenoux have you a linux machine with full OS to test code modification ?

SwoopX, 2021-08-09:
Nope

delphiki, 2021-08-18:
I'd love to see this device integrated as well! :)

arenoux, 2021-08-22:
Thanks for your job @Smanar .

I'm on holidays right now (not at home) it tried to launch a build (with docker procedure) on my raspi3 remotely but seems my raspi froze and does not respond right now... Will try again once back home beginning of September

Smanar, 2021-08-23:
Yep same for me, my raspberry 3 freeze, I need to remove "-j2" on my side.
But even with that the first compilation need so much time.

delphiki, 2021-08-28:
I can confirm that it's working with Home Assistant :+1: 

[image]

Smanar, 2021-08-28:
Thx, PR submited

arenoux, 2021-09-08:
Thank you so much all for this work :+1:

closed 2021-09-25 by manup