dresden-elektronik/deconz-rest-plugin #5139: Error upgrading Conbee II Firmware 0x266f0700

PrimaryInc, 2021-07-19:
## Describe the question or issue you are having
Error trying to upgrade firmware conbee II to 0x266f0700
See screenshot for error message.

Any idea what I'm doing wrong here?

## Screenshots

[image]

[image]

## Environment
DeConz Version v2.13.03
Firmware version 26580700

 - Host system: PC
 - Running method:  Windows / Virtual Machine (vSphere 7)
 - Firmware version: 26580700
 - deCONZ version: 2.13.03
 - Device: ConBee II
 - Do you use an USB extension cable:  yes
 - Is there any other USB or serial devices connected to the host system? Yes: rfx433Mhz, Sigma Z-wave, see screenshot, Note!! rfx433 usb device removed during upgrade.

Mimiix, 2021-07-19:
You are using a vm. 

Try it on a native machine :)

PrimaryInc, 2021-07-20:
@Mimiix, thanks for your reply. Upgrade has always worked on vm using conbee stick, (gen 1) So I could move it to another pc, that is possible, biut stil the question is why this is happening on conbee II, is it vSphere 7 that is a problem, something else? DeConz been runing on windows server from day 1 and for a few years without seeing this issue. Would be great to understand why this is happening, and make a note of it in the upgrade guide if it cannot be done.

Mimiix, 2021-07-20:
Probably because of the Disconnect thats caused by the update. The VM software is not able to properly re-attach it again. I am not knowledgeable on the deeper knowledge on this.

Smanar, 2021-07-20:
I don't know what is vSphere but there is that in the upgrade guide

> Warning: It is strongly advised to use a native installation of one of the update methods. Virtual Machines may work but are not supported, using a VM while trying to update firmware is on your own risk.

PrimaryInc, 2021-07-21:
@Mimiix, ok then upgrade of host it is. Thanks again.

@Smanar, vSphere is VMware. As stated worked for years but no longer. But yeah, your point is taken :) Thanks!

closed 2021-07-21 by PrimaryInc

Mimiix, 2021-07-21:
@PrimaryInc Perhaps something changed in how vspehere does USB mounting. Or how the GCFlasher works. I have really no clue. _Something_ must have changed _Somewhere_.

PrimaryInc, 2021-07-21:
@Mimiix, for sure :) Non the less, updatet usin an pc, works just fine.