dresden-elektronik/deconz-rest-plugin #5143: Device name

EugenG84, 2021-07-20:
Could you please add Tuya Smoke detector?

## Device
- Product name: Bakeey Tuya Smoke Alarm ZigBee Fire Detector 3C Fire Smoke Alarm
- Manufacturer: _TZE200_t5p1vj8r
- Model identifier: TS0601
- Device type:
- Other: Smoke detector

## Screenshots
[image]

### Basic

[image]

OTAU Cluster

[image]

Node Info
[image]

Mimiix, 2021-07-20:
Can you please also post a url on the forum as explained here:  https://forum.phoscon.de/t/read-me-device-request-read-me/23/2 o

EugenG84, 2021-07-20:
https://forum.phoscon.de/t/tuya-zigbee-smoke-detector-tze200-t5p1vj8r/171

Smanar, 2021-07-28:
Hello, do you have a linux machine with full OS to test code modification ?

EugenG84, 2021-07-28:
Hi,

i am using Home Assistant Supervised (debian 10 with docker).

Smanar, 2021-07-28:
I can't help you with docker but you have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Compiling-the-REST-plugin-for-device-specific-testing

On normal OS after having installed deconz :

```
sudo apt install deconz-dev
git clone --branch tuya_smoke1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

Branch is "tuya_smoke1"

I think the device will be created, but not sure it will trigger the alarm, I have used a "deducted" value.

Onshuis, 2021-09-14:
Ok from our conversation elsewhere, here is my smoke detector device:

    Product name: Choifoo Tuya Smart Smoke Alarm Detector
    Manufacturer: _TZE200_uebojraa
    Model identifier: TS0601
    Device type: RFD
    Other: Smoke detector

Screenshots:

[image]

[image]

[image]

[image]

Well i hope this is what you meant. Otherwise please tell me if i can do something myself?
Thanks in advance.

Mimiix, 2021-09-15:
@Smanar Ping

Smanar, 2021-09-15:
Ok so have updated the code, can use same command,
From scratch 
```
sudo apt install deconz-dev
git clone --branch tuya_smoke1 https://github.com/Smanar/deconz-rest-plugin.git
cd deconz-rest-plugin
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

If you have already do it
```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

I haven't found tuya command for this device, so I m using the same one than for the first device (Don't worry, can read them in log later if needed)

Onshuis, 2021-09-15:
That was fast, thanks i will test if it works.

Onshuis, 2021-09-15:
Well i did it real quick, but i dont know the procedure for sure. Im guessing the same as with a device that is supported.
So i did the qmake and make and copy the library, Then i restarted my system. 
Then i open vnc to my linux machine, then i stopped deconz. I started deCONZ.
I opened phoscon and tried to pair.  But phoscon runs out. I see the led blinking and stop blinking like before.
In deCONZ i see the device, but it's not recognized.
Now maybe im doing something wrong?

Onshuis, 2021-09-16:
Well i tried several times different things, 
Can i do anything else? I did resetting device with battery taking out, delete device in deCONZ several times.
But it keeps coming back with 0xXXXX not supported. Perhaps i'm doing something wrong that i have no clue of?

Smanar, 2021-09-16:
IDK, for the whitelist I m using

```
{ VENDOR_TUYA, "TS0601", telinkMacPrefix }, // Tuya Zigbee Smoke detector
with
const quint64 telinkMacPrefix     = 0xa4c1380000000000ULL;
#define VENDOR_TUYA                 0x1141
```

Can be an issue with the tuya cluster not always visible (but it s strange, as you can see it in deconz). Have a made a modification

```
cd deconz-rest-plugin
git pull
qmake && make -j2
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
```

To update.
If it still not working can you share logs during the inclusion ? with "info" and "info_l2" ?

Onshuis, 2021-09-16:
Hello and thanks for all the effort. It is still not working. When i do `/usr/bin/deCONZ --platform minimal --dbg-info=I2 > tuyasmoke1_debug`
I get "tuyasmoke1_debug: permission denied." If i try to run as sudo su i get all sorts of other errors and deCONZ will not run at all.

If i do it without writing to a file, my console gives all the info, but it is so much can not copy everything and put it in a file myself.
When i try other commands it says:"This plugin does not support propagateSizeHints()
For instance if i do "/usr/bin/deCONZ --platform minimal --dbg-info=I2 > /home/onshuis/tuyasmoke1_debug

So i feel like a mouse in a wheel going round and round never getting to the end. I tried searching for what you said "info" and info_I2" during inclusion, and i thought it would be the commands that are in the compiling REST plugin information page, but now i am not so sure. :-(

It's ubuntu 20.04 focal headless, i have a conbeeII, normally i run deconz as a systemd, when i do all this i need to stop the service. I login with vnc and i open a terminal and then start deCONZ and i can see all the lines. But with this debug log, i realy need to learn as i have no clue what i am doing, or if it is permissions. But thanks for so far and all your time. How can i make this debug thing work ? Any ideas :-)

Onshuis, 2021-09-17:
Well it was a bit of a hassle, but a learned a thing or two in the process :-)
Here is the debug: [https://pastebin.com/yPL8bcvM](url)

Smanar, 2021-09-17:
Have made a change to have more log.
Can you update your code and share the same log ?

>This plugin does not support propagateSizeHints()

This error message is normal

Onshuis, 2021-09-17:
YES !!! Thanks again :-)
I've updated the code, it is now seen as Fire74. 
There is no battery.
Here is the log: https://pastebin.com/kjxWLZLG

Smanar, 2021-09-17:
```
17:09:47:842 Tuya debug 4 : Address 0xA4C138DD3EDCC321 Payload 003c0104000101
17:09:47:843 Tuya debug 5 : Status: 0 Transid: 60 Dp: 1025 (0x04,0x01) Fn: 0 Data 1

17:09:48:180 Tuya debug 4 : Address 0xA4C138DD3EDCC321 Payload 003d0e04000102
17:09:48:181 Tuya debug 5 : Status: 0 Transid: 61 Dp: 1038 (0x04,0x0E) Fn: 0 Data 2
```

Can you just take a look in logs, press the test button and look for a line that is like the previous one ?

I don't know thoses 2 command but the type is enum, so can't be a battery (but can be a state)

Onshuis, 2021-09-17:
`19:08:18:871 Tuya debug 4 : Address 0xA4C138DD3EDCC321 Payload 004f0104000100
19:08:18:872 Tuya debug 5 : Status: 0 Transid: 79 Dp: 1025 (0x04,0x01) Fn: 0 Data 0
19:08:20:786 Tuya debug 4 : Address 0xA4C138DD3EDCC321 Payload 00500104000101
19:08:20:786 Tuya debug 5 : Status: 0 Transid: 80 Dp: 1025 (0x04,0x01) Fn: 0 Data 1
`

It says data 0 and data 1

Smanar, 2021-09-17:
Ok, have updated the code again
The field state/fire now react at request 0x04, 0x01 (if I m right, can be reversed)

Onshuis, 2021-09-17:
Correct, now pressing the test button, the detector beeps and i see the change from "off" to "on" and back to off state.

Smanar, 2021-09-17:
So I think the 
```
17:09:48:180 Tuya debug 4 : Address 0xA4C138DD3EDCC321 Payload 003d0e04000102
17:09:48:181 Tuya debug 5 : Status: 0 Transid: 61 Dp: 1038 (0x04,0x0E) Fn: 0 Data 2
```

is probably the battery state, with empty/low/full value

Onshuis, 2021-09-17:
Should we try right?

Smanar, 2021-09-17:
If you have empty batterie ^^ ?

Onshuis, 2021-09-17:
No im sorry misunderstood. I cant see if battery is low. There is no battery

Smanar, 2021-09-17:
Yep but to be sure, if you can use a low level battery in the detector to see if the value change in the request ?
Else I will make the PR without the battery ?

Onshuis, 2021-09-17:
Ok, there is a CR type of battery in it. I know i have one somewhere. I will let you know tomorrow.

Onshuis, 2021-09-18:
OK, i did not have a spare battery. So what i did was hook 2 wires on my lab supply and attach to the smoke detector.
I then got from 3V down with steps of 10mV to 2,10V.
But i really have no clue what to look for. I do not see a value changing. Debug was set to info. Perhaps another debug show a change in value, but i don't know that.

Debug: https://pastebin.com/dSq0TSgh

And if i have some spare time, which is very little, i now try to read up, how things are put together.
I use this: https://www.nxp.com/docs/en/user-guide/JN-UG-3077.pdf

Maybe you know of a better information source?

Smanar, 2021-09-18:
Yeah ^^, the bible https://zigbeealliance.org/wp-content/uploads/2019/12/07-5123-06-zigbee-cluster-library-specification.pdf and easier to understand.

BTW, you don't need to re-include the device.
Unfortunately there is always the same

>12:53:42:277 Tuya debug 5 : Status: 0 Transid: 4 Dp: 1038 (0x04,0x0E) Fn: 0 Data 2

I think on the json you have the field "lowbattery" ? I can set it to true if the value is different than 2 ?

Onshuis, 2021-09-19:
Yes that would be a good compromise.

Smanar, 2021-09-19:
Done, have updated the code again, if you are agree with it, I will make the PR.

Onshuis, 2021-09-19:
Thanks for the link to "the bible".
I have no idea how to do this with json.
I guess i need something to do with nodered or zigbee2mqtt. Maybe some time in the future when i have some time to figure that out. For now, it is fine by me. It works, go ahead make the pull request and thank you very much.