dresden-elektronik/deconz-rest-plugin #5149: Mi Motion Sensor not working as expected

leonardpitzu, 2021-07-21:
## Describe the bug
I have 5 motion sensors from Xiaomi (Mi Motion Sensor). Sensor #1 work as expected (see screenshot below) - it detects motion and reports temperature and light intensity as well. The next 4 sensors only report motion - temperature and light intensity are always 0. Also these 4 sensors become unavailable after some time and need to be re-paired.

## Steps to reproduce the behavior
open deconz/phoscon -> the sensors and their data should be visible and one can easily see what's missing.

## Expected behavior
all motion sensors should report the same data and stay connected (e.g. should not need re-pairing every now and then)

## Screenshots
first sensor is, the next are working strange
[image]
this is the good sensor
[image]
these are working bad
[image]

## Environment
 - Host system: Raspberry Pi
 - Running method: Raspbian
 - Firmware version: 26660700
 - deCONZ version: 2.12.02 (tested with beta and stable)
 - Device: ConBee II
 - Do you use an USB extension cable: no
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs

## Additional context

Mimiix, 2021-07-21:
Can you check the Basic clusters and provide the model ID and manufacturer ID?

leonardpitzu, 2021-07-21:
i can only read the Basic cluster for the first sensor, for the next 4 it's not working. I tried rebooting and doing the basic stuff. Any hints how to "get this going"?

SwoopX, 2021-07-21:
Neither the node descriptor nor the clusters are read correctly. Those devices are known to be shi**y. Try reset and re-pair until it works

Mimiix, 2021-07-21:
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Xiaomi:-Lost-devices-or-functional-issues

leonardpitzu, 2021-07-21:
@Mimiix - thanks for the link, i was able to read the damn thing.
[image]

Mimiix, 2021-07-21:
Can you also read the others?

leonardpitzu, 2021-07-21:
all other 4 sensors are the same as this one

[image]
[image]
[image]
[image]

Mimiix, 2021-07-21:
No, i mean the Basic clusters of the devices that show "correct:" info and whioch do not.

I want to compare the modelID and Manuf ID with all your working and non working motion sensors

leonardpitzu, 2021-07-21:
you already have it:
Storage Room Motion Sensor -> correct
Bedroom Motion Sensor -> bad

leonardpitzu, 2021-07-21:
good
[image]

bad
[image]

Mimiix, 2021-07-21:
Thats my point :)

Theres 2 types. The one with lux etc and without. They have different Model numbers on the bottom as well.

Obviously, the software cant report what it doesn't "sense" 😢.

leonardpitzu, 2021-07-21:
The are the same for me - same box, case, code/label, etc. I ordered 5 and i got 1 + 4. What can i say except for "crap"...

leonardpitzu, 2021-07-21:
yeap... long live aliexpress

Mimiix, 2021-07-21:
Sorry to hear that :(

Perhaps you can reach out to the seller.

Either way, closing this issue

closed 2021-07-21 by Mimiix