dresden-elektronik/deconz-rest-plugin #5150: iobroker sql.0 log läuft über

Satmedia, 2021-07-21:
## Describe the bug
Hallo,

bei meiner iobroker Installation (Ubuntu 20.4 vorher Raspi 4) läuft der log für sql logging von deconz objekte über.
Logging funktioniert, ist aber sehr nervig, weil Log Infos meiner Scripte "untergehen".
Logging getestet mit SQLite3 und MySQL Datenbank am gleichen Ubuntu Gerät.
Die Info-Meldungen im Log kommen nur von Deconz Objekten!

Log Beispiel siehe unten:

Deconz Version, Conbee II Firmware und Iobroker deconz Adapter waren alle auf neuester Stable Version installiert.
Zum Test wurde nun jeweils die letzte Beta installiert --> Keine Änderung

Hier meine Iobroker Config:

Iobroker deConz Adapter: 1.3.18

Platform: linux
os: linux
Architecture: x64
CPUs: 4
Speed: 1506 MHz
Model: Intel(R) Core(TM) i3-4010U CPU @ 1.70GHz
RAM: 3.7 GB
System uptime: 12 T. 00:29:50
Node.js: v12.22.2
Disk size: 27.2 GiB
Disk free: 17.5 GiB
adapters count: 348
NPM: v6.14.13
Uptime: 00:19:14
Active instances: 11

LOG:
sql.0 2021-07-20 15:43:55.934 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:43:41.531 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:43:26.733 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:43:00.526 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:42:45.631 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:42:21.533 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:42:07.138 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:41:45.531 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:41:30.638 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:41:15.934 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:40:58.933 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:40:39.028 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:40:17.272 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:39:56.788 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:39:37.027 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:39:14.880 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:38:53.582 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:38:33.979 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:38:14.642 info (162415) enabled logging of deconz.0.Sensors.17.temperature, Alias=false
sql.0 2021-07-20 15:38:12.482 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:37:52.189 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:37:11.279 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:36:51.378 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:36:29.982 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:36:08.375 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:35:53.474 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:35:32.276 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:35:18.039 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:34:56.413 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:34:41.378 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:34:12.389 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:33:57.481 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:33:36.326 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:33:21.878 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:32:59.379 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:32:44.980 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:32:19.875 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:32:04.887 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:31:49.880 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:31:35.879 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:31:21.877 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:31:00.375 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:30:45.385 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:30:23.276 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:30:08.480 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:29:47.281 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false
sql.0 2021-07-20 15:29:32.267 info (162415) enabled logging of deconz.0.Sensors.21.power, Alias=false

mfg Roland

## Steps to reproduce the behavior

## Expected behavior

## Screenshots

## Environment
 - Host system: Now PC, also tested Raspi 4
 - Running method: Ubuntu 20.4 / before Raspbian 
 - Firmware version: 266F0700
 - deCONZ version: 2.12.03 
 - Device:  ConBee II 
 - Do you use an USB extension cable: yes 
 - Is there any other USB or serial devices connected to the host system? no

## deCONZ Logs
Im Deconz Log konnte ich keine Hinweise finden

## Additional context

Mimiix, 2021-07-21:
Hi,

This repository is only for deCONZ related items. Iobroker is an 3rd party integration. This repository does not provide any support for that.

closing.

closed 2021-07-21 by Mimiix

Satmedia, 2021-07-21:
Do you know where I can report this problem?

Mimiix, 2021-07-21:
Perhaps here: https://forum.iobroker.net/

Satmedia, 2021-07-21:
thxs, I do